kellerza/pysma #1: Adding more data keys

badblocks, 2017-12-05:
```
KEY_CURRENT_POWER_W = '6100_40263F00' 
KEY_CURRENT_CONSUMPTION_W = '6100_00543100' 
KEY_TOTAL_YIELD_KWH = '6400_00260100' 
KEY_TOTAL_CONSUMPTION_KWH = '6400_00543A00' 
```
I want to add more data keys to pysma; like for example  'KEY_DAILY_YIELD_KWH' (getting the dailly yield of the SMA device).
In what documentation from SMA I can find the corresponding values (like 6100_40263F00)?

kellerza, 2017-12-05:
These keys came from spending some time with Firebug/Chrome dev tools. Open the SMA interface, navigate around and it should be clear from the requests in the network activity tab...
Unfortunately no easy answer!

At least you can add custom keys in yaml to test them, but would be great if you post your findings here/add it to the HomeAssistant docs

badblocks, 2017-12-07:
Thanks for the info. I will try to get more keys (if possible).

badblocks, 2017-12-27:
I failed to get more keys, so I using the modbus connection (must be enabled) to get all the data I want.

kellerza, 2018-02-05:
Strange, yesterday and today does not seem to be part of the requests.

These seem to be all the kwh values from ```/data/ObjectMetadata_Istl.json``` if it can help anyone

```
6400_00462500
6400_00462400
6400_00262200
6400_00260100 KEY_TOTAL_YIELD_KWH
6400_0046C300
6400_00543C00
6400_00543B00
6400_00543D00
6400_00543A00 KEY_TOTAL_CONSUMPTION_KWH
```

closed 2018-02-05 by kellerza