kellerza/pysma #2: Home Assistant Issue

kimi1983, 2017-12-18:
Hello,

im a trying to use your home assistant component but i am unable to connect to my SMA Tripower Interface.

My Error:

Error doing job: Task exception was never retrieved
09:02 components/sensor/sma.py (ERROR)

Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/srv/homeassistant/lib/python3.5/site-packages/pysma/__init__.py", line 45, in _fetch_json
    url, data=json.dumps(payload), headers=headers, params=params)
  File "/srv/homeassistant/lib/python3.5/site-packages/aiohttp/helpers.py", line 99, in __iter__
    ret = yield from self._coro
  File "/srv/homeassistant/lib/python3.5/site-packages/aiohttp/client.py", line 267, in _request
    conn = yield from self._connector.connect(req)
  File "/srv/homeassistant/lib/python3.5/site-packages/aiohttp/connector.py", line 402, in connect
    proto = yield from self._create_connection(req)
  File "/srv/homeassistant/lib/python3.5/site-packages/aiohttp/connector.py", line 748, in _create_connection
    _, proto = yield from self._create_direct_connection(req)
  File "/srv/homeassistant/lib/python3.5/site-packages/aiohttp/connector.py", line 831, in _create_direct_connection
    req=req, client_error=client_error)
  File "/srv/homeassistant/lib/python3.5/site-packages/aiohttp/connector.py", line 796, in _wrap_create_connection
    return (yield from self._loop.create_connection(*args, **kwargs))
  File "/usr/lib/python3.5/asyncio/base_events.py", line 762, in create_connection
    yield from self.sock_connect(sock, address)
  File "/usr/lib/python3.5/asyncio/selector_events.py", line 451, in sock_connect
    return (yield from fut)
  File "/usr/lib/python3.5/asyncio/futures.py", line 380, in __iter__
    yield self  # This tells Task to wait for completion.
  File "/usr/lib/python3.5/asyncio/tasks.py", line 304, in _wakeup
    future.result()
  File "/usr/lib/python3.5/asyncio/futures.py", line 285, in result
    raise CancelledError
concurrent.futures._base.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.5/asyncio/tasks.py", line 241, in _step
    result = coro.throw(exc)
  File "/srv/homeassistant/lib/python3.5/site-packages/homeassistant/components/sensor/sma.py", line 125, in async_sma
    values = yield from sma.read(keys_to_query)
  File "/srv/homeassistant/lib/python3.5/site-packages/pysma/__init__.py", line 79, in read
    yield from self.new_session()
  File "/srv/homeassistant/lib/python3.5/site-packages/pysma/__init__.py", line 52, in new_session
    URL_LOGIN.format(self._ip), self._new_session_data)
  File "/srv/homeassistant/lib/python3.5/site-packages/pysma/__init__.py", line 46, in _fetch_json
    return (yield from res.json())
  File "/srv/homeassistant/lib/python3.5/site-packages/async_timeout/__init__.py", line 35, in __exit__
    self._do_exit(exc_type)
  File "/srv/homeassistant/lib/python3.5/site-packages/async_timeout/__init__.py", line 80, in _do_exit
    raise asyncio.TimeoutError
concurrent.futures._base.TimeoutError

How can i troubleshoot this issue ?

Best regards and thanks for your help.

kimi1983, 2017-12-19:
Hi, and how can i use your example.py ?

What is the syntax for username an ip ?

best regards

kellerza, 2017-12-20:
Hi,

IP can be anything you pass in a URL. So IP, IP:port, hostname or hostname:port.

No username, I only have a choice between user or installer... example.py currently defaults to installer

See a screenshot of my Webconnect login below

kimi1983, 2017-12-20:
I am sorry, there is no Screenshot.

But i own a SMA Tripower Inverter which is connected to my Network and which is sending Data to sunnyportal.com, an i own a sunnyportal User Account to see online my data.

But your script is used to connect locally to my inverter and to pull data, right ?

what port do i need to enter ?

kellerza, 2017-12-20:
Maybe the screenshot does not survive the email posting...

Mine is simply port 80, you should be able to do connect from any browser on your LAN once you have the IP

kimi1983, 2017-12-20:
Ok so this seems not work on SMA Sunny Tripower 15000TL.
This on is not reachable via local IP on Port 80.

It is only contacting the sma Sunnyportal Webportal to push the data online.

best regards

closed 2018-02-05 by kellerza