kellerza/pysma #3: Doesn't work with https

cby016, 2018-01-19:
My inverter forces https. Is there any way to set a configuration option to change the protocol from http to https?

For now I have manually changed the URLs but it would be nice for this to be configurable.

PhilRW, 2018-02-05:
I'm confirming this observation. I manually changed the `http` to `https` and it now connects to my Sunny Boy 7.7 but obviously this is a temporary workaround.

kellerza, 2018-02-05:
Was that the only change `URL_LOGIN`, `URL_LOGOUT` and `URL_VALUES` to start with https?

PhilRW, 2018-02-05:
Yes, that's all that I changed.

Freshhat, 2018-02-11:
Also just tried to change the values from http to https but it seems that i have still some issues. Here is the trace which i get:
```
Traceback (most recent call last):
  File "/home/fri/hass_venv/lib64/python3.6/site-packages/pysma/__init__.py", line 46, in _fetch_json
    url, data=json.dumps(payload), headers=headers, params=params)
  File "/home/fri/hass_venv/lib64/python3.6/site-packages/aiohttp/helpers.py", line 99, in __iter__
    ret = yield from self._coro
  File "/home/fri/hass_venv/lib64/python3.6/site-packages/aiohttp/client.py", line 267, in _request
    conn = yield from self._connector.connect(req)
  File "/home/fri/hass_venv/lib64/python3.6/site-packages/aiohttp/connector.py", line 402, in connect
    proto = yield from self._create_connection(req)
  File "/home/fri/hass_venv/lib64/python3.6/site-packages/aiohttp/connector.py", line 749, in _create_connection
    _, proto = yield from self._create_direct_connection(req)
  File "/home/fri/hass_venv/lib64/python3.6/site-packages/aiohttp/connector.py", line 832, in _create_direct_connection
    req=req, client_error=client_error)
  File "/home/fri/hass_venv/lib64/python3.6/site-packages/aiohttp/connector.py", line 797, in _wrap_create_connection
    return (yield from self._loop.create_connection(*args, **kwargs))
  File "/usr/lib64/python3.6/asyncio/base_events.py", line 764, in create_connection
    yield from self.sock_connect(sock, address)
  File "/usr/lib64/python3.6/asyncio/selector_events.py", line 451, in sock_connect
    return (yield from fut)
  File "/usr/lib64/python3.6/asyncio/futures.py", line 332, in __iter__
    yield self  # This tells Task to wait for completion.
  File "/usr/lib64/python3.6/asyncio/tasks.py", line 250, in _wakeup
    future.result()
  File "/usr/lib64/python3.6/asyncio/futures.py", line 237, in result
    raise CancelledError
concurrent.futures._base.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib64/python3.6/asyncio/tasks.py", line 182, in _step
    result = coro.throw(exc)
  File "/home/fri/hass_venv/lib64/python3.6/site-packages/homeassistant/components/sensor/sma.py", line 125, in async_sma
    values = yield from sma.read(keys_to_query)
  File "/home/fri/hass_venv/lib64/python3.6/site-packages/pysma/__init__.py", line 80, in read
    yield from self.new_session()
  File "/home/fri/hass_venv/lib64/python3.6/site-packages/pysma/__init__.py", line 53, in new_session
    URL_LOGIN.format(self._ip), self._new_session_data)
  File "/home/fri/hass_venv/lib64/python3.6/site-packages/pysma/__init__.py", line 47, in _fetch_json
    return (yield from res.json())
  File "/home/fri/hass_venv/lib64/python3.6/site-packages/async_timeout/__init__.py", line 35, in __exit__
    self._do_exit(exc_type)
  File "/home/fri/hass_venv/lib64/python3.6/site-packages/async_timeout/__init__.py", line 80, in _do_exit
    raise asyncio.TimeoutError
concurrent.futures._base.TimeoutError
```

And i have this inverter: SMA Sunny Tripower STP 8000TL-20

Any idea about that?

kellerza, 2018-02-11:
@Freshhat it seems like a timeout

Are you sure the IP is configured correctly? Does the 8000 have a local webserver? (Did not see one in the installation manual)

I will do an update to change to https shortly... hopefully this helps as well

Freshhat, 2018-02-11:
Just checked the manual again from my inverter, it seems that you're right. So this model doesn't have a local webserver - damn :/

kellerza, 2018-02-12:
@Freshhat sorry about that!

Added an option to specify SSL and additional Timeout options [here](../commit/138fc7096de76e7cc6183b29ff9cfd6f62ee0931)

closed 2018-02-12 by kellerza

cby016, 2018-02-12:
@kellerza Hey thanks for updating this!

kellerza, 2018-02-13:
Comitted in Home Assiatnt `dev` branch, add `ssl: true` if you want to test it. Might still remove this option to autodetect SSL