kellerza/pysma #6: Using custom sensor keys in Home Assistant

MimbaMonkeyHouse, 2018-03-26:
I managed to get this working with Home Assistant using the predefined keys. I'm not having any luck with setting up custom values and am getting errors in the log. The docs are not very clear on how to set this up (I'm happy to update these once I understand).
@kellerza: Can you please have a quick look at the code below and let me know how to correct this?
```
- platform: sma
  host: "192.168.xx.xx"
  password: "xxxxxxxx"
  sensors:
    custom:
      battery_charge: 
        key: "6100_00295A00"
        unit: "%"
        factor: "1"
```

A few questions while I'm on it. 
1) Is the SMA component meant to be used with a User or Installer account?
2) Is the factor value needed-compulsory? What does it do?
3) Can we use any of the standard component settings (friendly_name, scan_interval, etc)?

If I can get this to work, I'd much rather use this than the Modbus integration. Seems safe not to use Modbus as this allows for settings to be changed remotely. It's also easier to set up with Google Chrome's Inspect function.

kellerza, 2018-03-27:
@MimbaMonkeyHouse it would help if you can post the errors (or output of `hass --script check_config`). I suspect you must just remove all the quotes.

> Is the SMA component meant to be used with a User or Installer account?

set with the `group` key

> Is the factor value needed-compulsory? What does it do?

Optional, default will be 1. Its for units like `W` or `kW` (1 or 1000)

> Can we use any of the standard component settings (friendly_name, scan_interval, etc)?

scan interval is supported, friendly name probably not, since this platform creates multiple sensors... You probably want to use customize.yaml

It would be great if you can contribute to the documentation!

kellerza, 2018-03-27:
Try:

```yaml
sensors:
      battery_charge:
custom:
      battery_charge: 
          key: "6100_00295A00"
          unit: "%"
```

MimbaMonkeyHouse, 2018-03-27:
These are some of the errors I was getting this morning:
```
Unexpected return: {} - '6100_00543100'
8:26 AM components/sensor/sma.py (ERROR)
Unexpected return: {} - '6100_00543100'
8:25 AM components/sensor/sma.py (ERROR)
Unexpected return: {} - '6100_00543100'
8:25 AM components/sensor/sma.py (ERROR)
Unexpected return: {} - '6100_00543100'
8:25 AM components/sensor/sma.py (ERROR)
Unexpected return: {} - '6100_00543100'
8:25 AM components/sensor/sma.py (ERROR)
Unexpected return: {} - '6100_00543100'
8:24 AM components/sensor/sma.py (ERROR)
```
This is strange as I was not trying to use this particular custom sensor key...

MimbaMonkeyHouse, 2018-03-27:
I added quote as these are required when unit is a percent sign. Will take these off for other values then...

MimbaMonkeyHouse, 2018-03-27:
OK, problem solved. It's now working with this code:
```
- platform: sma
  host: 192.168.xx.xxx
  group: user
  password: xxxxx
  sensors:
    current_consumption:
    current_power:
    sma_grid_voltage:
    sma_pv_power:
    sma_daily_yield:
    sma_total_yield:
  custom:
    sma_grid_voltage: 
      key: "6100_00464800"
      unit: V
    sma_pv_power:
      key: "6100_0046C200"
      unit: W
    sma_daily_yield:
      key: "6400_00262200"
      unit: Wh
    sma_total_yield:
      key: "6400_00260100"
      unit: Wh
- platform: sma
  host: 192.168.xx.xxx
  group: user
  password: xxxxx
  sensors:
    sma_si_battery_charge:
    sma_si_total_yield:
  custom:
    sma_si_battery_charge: 
      key: "6100_00295A00"
      unit: "%"
    sma_si_total_yield: 
      key: "6400_00462400"
      unit: Wh
      factor: 1000
```
The strange thing is the output from the current_power sensor returning a value of 5... Will look into this and other issues before updating the Home Assistant docs!

Thanks for the component, tips and... time.

closed 2018-03-27 by MimbaMonkeyHouse