kellerza/pysma #7: Allow keys with more than 13 characters

MimbaMonkeyHouse, 2018-03-27:
Some keys in Webconnect use more than 13 characters. This is the case, for instance, for inverters with 2 separate arrays/MPPTT. Here is what I found on my Sunnyboy 5.0:
```
  custom:
    sma_status: 
      key: "6180_08214800"
    sma_power_a:
      key: "6380_40251E00_0"
      unit: W
    sma_power_b:
      key: "6380_40251E00_1"
      unit: W
    sma_voltage_a:
      key: "6380_40451F00_0"
      unit: V
    sma_voltage_b:
      key: "6380_40451F00_1"
      unit: V
    sma_current_a:
      key: "6380_40452100_0"
      unit: A
    sma_current_b:
      key: "6380_40452100_1"
      unit: A
```
It can be interesting to be able to monitor several arrays separately rather than the entire output of an inverter. This would for example allow users to better determine the effect of shading, orientation and inclination on PV output.

Can the pysma code be corrected easily to account for these sensor keys with 15 characters?

MimbaMonkeyHouse, 2018-03-27:
Here is the error log caused by these 15 characters sensor keys:
```
Invalid config for [sensor.sma]: length of value must be at most 13 for dictionary value @ data['custom']['sma_current_a']['key']. Got '6380_40452100_0'
length of value must be at most 13 for dictionary value @ data['custom']['sma_current_b']['key']. Got '6380_40452100_1'
length of value must be at most 13 for dictionary value @ data['custom']['sma_power_a']['key']. Got '6380_40251E00_0'
length of value must be at most 13 for dictionary value @ data['custom']['sma_power_b']['key']. Got '6380_40251E00_1'
length of value must be at most 13 for dictionary value @ data['custom']['sma_voltage_a']['key']. Got '6380_40451F00_0'
length of value must be at most 13 for dictionary value @ data['custom']['sma_voltage_b']['key']. Got '6380_40451F00_1'
required key not provided @ data['custom']['sma_status']['unit']. Got None. (See ?, line ?). Please check the docs at https://home-assistant.io/components/sensor.sma/
```

kellerza, 2018-03-27:
You can try by changing [this](https://github.com/home-assistant/home-assistant/blob/dev/homeassistant/components/sensor/sma.py#L59) line. If it works, you can even submit a PR

MimbaMonkeyHouse, 2018-03-27:
Sounds good, but that's way beyond my noob abilities. Do I just create a pysma.py file and add it to the deps folder?

kellerza, 2018-03-27:
You can save the file as  `your_config_dir\custom_components\sensor\sma.py`. If it works I can do the change for you in the next iteration

MimbaMonkeyHouse, 2018-03-27:
Yup, this one seems to work. Can't fully test it as it's night here and sensor values are equal to none. Will test tomorrow...

MimbaMonkeyHouse, 2018-05-10:
Changed to:
```
            vol.Required('key'): vol.All(str, vol.Length(min=13, max=15)),
```
Error reads:
```
Unexpected return: {'6100_00543100': {'1': [{'val': 1616}]}, '6100_40263F00': {'1': [{'val': None}]}, '6380_40251E00': {'1': [{'val': None}, {'val': None}]}} - '6380_40251E00_0'
```

kellerza, 2018-05-18:
It seems like the `_0` and `_1`  could refer to the index of the array?

Full response: `'6380_40251E00': {'1': [{'val': None}, {'val': None}]}` you will see all the other keys just have a single value in the array.

At the moment your example shows `None` for both: `[{'val': None}, {'val': None}]`, maybe you can try grab the example when these sensors are active / i.e. during the day and just confirm?

MimbaMonkeyHouse, 2018-06-11:
Hey @kellerza, just want to give you an update that I am now unable to check these as I figured out that the Home Manager device I have in my setup closes the WebConnect connection.

commit 1980efe5 (2018-08-12): Fix #7
closed 2018-08-12 by kellerza