kellerza/pysma #8: Allow sensor keys with no unit defined

MimbaMonkeyHouse, 2018-03-27:
Some sensor keys have no unit attached. This can be seen with status messages or features being turned on/off.
Here is an example of a sensor key I would like to integrate:
```
  custom:
    sma_status: 
      key: "6180_08214800"
```
Here is the error logged:
```
Invalid config for [sensor.sma]: required key not provided @ data['custom']['sma_si_battery_activity']['unit']. Got None
required key not provided @ data['custom']['sma_si_battery_cycles']['unit']. Got None. (See ?, line ?). Please check the docs at https://home-assistant.io/components/sensor.sma/
```
Can this be changed without too much effort?

kellerza, 2018-03-27:
You can use "" for an empty string, or make the validator `vol.Optional('unit', default='')`

MimbaMonkeyHouse, 2018-03-27:
"" returns this error:
```
'popitem(): dictionary is empty'
8:48 PM components/sensor/sma.py (ERROR)
```
I'll figure option 2 out and commit...

Thanks again

kellerza, 2018-03-27:
Option 1 must probably be: `default=[""]`

kellerza, 2018-05-18:
It seems that `val` is a list and not an integer

Can you try wrapping [`res = {key: val // sensor_defs[key][2] for key, val in res.items()}`](https://github.com/home-assistant/home-assistant/blob/dev/homeassistant/components/sensor/sma.py#L136) in a try/catch to see what is going on?

```python
try:
    res = {key: val // sensor_defs[key][2] for key, val in res.items()}
except TypeError as err:
    print("All these values should be simple integers")
    print("TypeError", str(err), res)
    print("All units, third item in array [2], should be floats")
    print(sensor_defs)
```

kellerza, 2018-08-12:
@MimbaMonkeyHouse I made some changes to the library to allow more flexible extraction of parameters. It's not part of HASS yet, only in the library.

If you can try to `git clone` the repo and then run `python example.py 192.168.88.199 user your_user_pass` it would be great.

Added more default keys like the [status](https://github.com/kellerza/pysma/blob/master/pysma/__init__.py#L62) you suggested, but this returns a simple value (307) on my unit for OK

kellerza, 2019-01-03:
I'm assuming this was fixed in the previous release of the library & HA SMA sensor. If not you can reopen

closed 2019-01-03 by kellerza