kellerza/pysma #9: SMA 2500TLST-21: Could not connect to SMA at http://192.168.0.128

jghaanstra, 2018-04-14:
I'm currently using SBFspot to read out my SMA 2500TLST-21 inverter through speedwire connection. This is working fine, but due to some changes in my home automation stack I'd like to start using pysma.

I tried using the Home Assistant component and the library directly but both give me the error "Could not connect to SMA at http://192.168.0.128". I tried both the user and the installer password but to no avail.

Is my inverter supposed to be supported? If so, how can I troubleshoot this further.

kellerza, 2018-04-14:
Hi, can you try connect via https?

The HA component has an undocumented `ssl: true` config option for https, but you really have to be able to connect via a browser in some way for this to work

jghaanstra, 2018-04-14:
Makes no difference with https. I also updated the firmware of the inverter to the latest version but that does not make any difference as well. What am I supposed to see when accessing the inverter by it's IP? Perhaps its something only supported by newer versions.

jghaanstra, 2018-04-14:
OK, apparently some Sunny Boy inverters come with an integrated webserver with their own use interface  (e.g. Sunny Boy 1.5/2.5). My inverter does not (SB 2500TLST-21) have this feature. So older models are probably not supported with this library. I'll stick with SBFspot.

closed 2018-04-14 by jghaanstra