kellerza/pysma #10: int object is not iterable

mauritskorse, 2018-08-07:
I am getting the following error twice each poll:
`ERROR (MainThread) [pysma] 'int' object is not iterable`

I am connecting to our sunny boy 2.5 with Hassio SMA component

kellerza, 2018-08-07:
Any more detail if you click on the error?

mauritskorse, 2018-08-07:
No, unfortunately not. But I have been doing some debugging.

It seems that the error occurs on line 118:
`result[idx] = [value['val'] for value in result[idx]]`

I am now also getting the following error as well:
`ERROR (MainThread) [pysma] 'popitem(): dictionary is empty'`

kellerza, 2018-08-12:
@mauritskorse,  I made some changes to improve extracting values, but might still be that your unit returns values differently.

Can you `git clone` the repo and run `python example.py 192.168.88.199 user your_user_password`?

mauritskorse, 2018-08-24:
@kellerza, this is what the script returns:
```
DEBUG:asyncio:Using selector: EpollSelector
INFO:__main__:NEW SID: _I6k0S2PcTjPFQGN
DEBUG:pysma:{"result": {"012F-7309ACFB": {"6180_08214800": {"val": [{"tag": 307}]}, "6400_00262200": {"val": 9939}, "6100_00464800": {"val": 23832}, "6400_00543A00": {"val": 0}, "6100_00543100": {"val": null}, "6100_0046C200": {"val": 228}, "6100_40263F00": {"val": 228}, "6400_00260100": {"val": 6517358}}}}
DEBUG:pysma:{"result": {"012F-7309ACFB": {"6180_08214800": {"val": [{"tag": 307}]}, "6400_00262200": {"val": 9939}, "6100_00464800": {"val": 23822}, "6400_00543A00": {"val": 0}, "6100_00543100": {"val": null}, "6100_0046C200": {"val": 228}, "6100_40263F00": {"val": 228}, "6400_00260100": {"val": 6517358}}}}
DEBUG:pysma:{"result": {"012F-7309ACFB": {"6180_08214800": {"val": [{"tag": 307}]}, "6400_00262200": {"val": 9939}, "6100_00464800": {"val": 23825}, "6400_00543A00": {"val": 0}, "6100_00543100": {"val": null}, "6100_0046C200": {"val": 228}, "6100_40263F00": {"val": 228}, "6400_00260100": {"val": 6517358}}}}
DEBUG:pysma:{"result": {"012F-7309ACFB": {"6180_08214800": {"val": [{"tag": 307}]}, "6400_00262200": {"val": 9939}, "6100_00464800": {"val": 23826}, "6400_00543A00": {"val": 0}, "6100_00543100": {"val": null}, "6100_0046C200": {"val": 227}, "6100_40263F00": {"val": 227}, "6400_00260100": {"val": 6517358}}}}
DEBUG:pysma:{"result": {"012F-7309ACFB": {"6180_08214800": {"val": [{"tag": 307}]}, "6400_00262200": {"val": 9939}, "6100_00464800": {"val": 23819}, "6400_00543A00": {"val": 0}, "6100_00543100": {"val": null}, "6100_0046C200": {"val": 227}, "6100_40263F00": {"val": 227}, "6400_00260100": {"val": 6517358}}}}
```

closed 2019-01-28 by kellerza