kellerza/pysma #11: no session ID

danbates2, 2018-08-15:
Hi, any clues as to why I'm getting thrown this error?

```
pi@emonpi(rw):pysma$ python3 example.py 192.168.2.2 user 0000
DEBUG:asyncio:Using selector: EpollSelector
ERROR:pysma:Could not start session, Could not connect to SMA at http://192.168.2.2 (timeout), got {'err': 'Could not connect to SMA at http://192.168.2.2 (timeout)'}
INFO:__main__:No session ID
```

SBFSpot connects fine, but was hoping to get a simpler program to work, so I can send the data elsewhere.

kellerza, 2018-08-15:
Can you access that address through a web browser? Maybe you need `https://192.168.2.2`?

danbates2, 2018-08-15:
Browser does nothing. https:// hasn't changed things either.
I'm 100% sure it's the ip address of the inverter, which is a 6000tl-20, and the user password 0000 default works.

danbates2, 2018-08-15:
Does it impact if SBFSpot has recently connected to the inverter? Is there a conflict of session-ID maybe?

kellerza, 2018-08-15:
If you cannot access the local Web UI, this module will not work for you.
SBFSpot seems to use SpeedWire, which is a different protocol

danbates2, 2018-08-15:
understood, thanks.

closed 2018-08-15 by danbates2

RaihaX, 2019-04-02:
Hi,

I am having similar issue with my Sunny Boy 5.0 which does have a local web ui (WebConnect).

Currently I'm just scraping the data from sunny portal but ideally i would like to use this to get the info locally and more real time.

I did briefly attempt to scrape the local info myself but i didn't spend too much time on it and it seems to use angular (maybe?) and I was getting similar session issue, was going to try using one of the headless browser but i thought I might try this first.

kellerza, 2019-04-02:
Hi @RaihaX,
Did you try `example.py` from this repo, or the Homeassistant component?

RaihaX, 2019-04-02:
from this repo - I haven't tried the HASS component in a few months (last time i tried it didn't work). Is the HASS component more up to date then the one in this repo?

kellerza, 2019-04-02:
The HASS component uses this library, it's just another (easier?) way to set it up

RaihaX, 2019-04-02:
Fair enough - I find it easier to debug an issue using the python script as I am not too familiar in debugging the python script when is running through HASS.

Do you have any inkling as to why I am getting a No Session ID issue?

kellerza, 2019-04-02:
Login fails, so it could be:
- Wrong password. User is fixed user/installer
- Too many sessions,- you might already have a web browser open on this session
- Wrong IP/port/http/https setting. Check with browser first

kellerza, 2019-04-02:
Or the response from your WebConnect is different... The script has quite a lot of debugs, so should see if it fails/succeeds in any way

RaihaX, 2019-04-02:
Login seems to be ok - i tested the login for both user and installer.

IP and port seems correct and web connect only works with http (https fails via browser).

I did check the python script and it doesn't provide much debugging as it literally fails upon connection.

this is the error debug from the script anyway

DEBUG:asyncio:Using selector: EpollSelector
ERROR:pysma:Could not start session, Session ID expected [result.sid], got {'result': {'sid': None}}
INFO:__main__:No session ID

FYI this is the same issue i had when i try logging in using domcrawler with my own script - i just assume angular must have built in functionality to prevent website scraping checking user agent etc etc.