kellerza/pysma #12: Can't call example.py

JensAtGitHub, 2018-10-29:
Could you provide an example for the correct syntax to call example.py?

I have problems to handover the parameter (ip. user, passwd). 
________________
Pysma/pysma $ python3.6 example.py "192.168.2.116" "user" "0000"

DEBUG:asyncio:Using selector: EpollSelector
Traceback (most recent call last):
  File "example.py", line 73, in <module>
    main()
  File "example.py", line 70, in main
    loop, user=args.user, password=args.password, ip=args.ip))
  File "/usr/local/lib/python3.6/asyncio/base_events.py", line 467, in run_until_complete
    return future.result()
  File "example.py", line 27, in main_loop
    if VAR['sma'].sma_sid is None:
AttributeError: 'SMA' object has no attribute 'sma_sid'

kellerza, 2018-10-29:
Your syntax seems to be ok

I use this on Windows
```
python example.py http://192.168.88.199 user secret*
```

The IP only also works.

Can you check if you don't perhaps have an older version (<0.2.2) installed? `pip list | grep pysma`

JensAtGitHub, 2018-10-30:
...thank you for the hint - on another (Win) installation it gives me results ;-). 
I guess the reason for the above problem on the raspi is that I have SBFSPOT running there which is already using the webconnect interface. 

I am a bit  lost with the library usage. How would I query individual values e.g. pv_power or current_consumption from my own code (using the functionget_sensor() ?)? I see no chance to adapt example.py to my needs. Is there some other example or document? Thank you anyhow, Regards,

kellerza, 2018-10-30:
See known sensor list [here](https://github.com/kellerza/pysma/blob/master/pysma/__init__.py#L59).You can add your own / create a PR on here to add more.

```python
# Log in like in example.py
the_sma = pysma.sma(.....)
await the_sma.new session()

# Decide which sensors you want to read
scc = get_sensor('current_consumption')
scp = get_sensor('current_power')

# Read them, accepts a list of sensors, in example.py I simply read them all pysma.SENSORS
await the_sma.read( [scc, scp] )

# Use the values & units...
print(scc.value, scc.unit)
print(scp.value, scp.unit)
```

You can also see the PR in [HASS](https://github.com/home-assistant/home-assistant/pull/17988/files) using this pattern

closed 2018-10-31 by JensAtGitHub