kellerza/pysma #14: Pysma ignores comma as decimal separator

DavidMenting, 2018-11-19:
When fetching data, pysma passes on whatever it grabs from WebConnect.

My inverter is set to Dutch, which means the decimal separator is a comma instead of a dot. In Home Assistant, the comma is not shown and the values displayed are 100 times too high.

Somewhere in the process, the comma should be converted to a dot. I don't know whether this should be done in pysma or in Home Assistant. As early as possible (=pysma) seems logical.

kellerza, 2018-11-19:
It should be in pysma, as it converts the text to a number...

kellerza, 2019-01-01:
@DavidMenting can you post the output of `example.py` or give the specific key that gives this incorrect value?

I added some unit tests and would like to include your case, but the examples I have from my the inverter does not send a decimal separator

kellerza, 2019-01-14:
No worries. If it's just voltage the new release should have you covered [here](https://github.com/kellerza/pysma/blob/master/pysma/__init__.py#L101). It seems like some sensors have other "multipliers" and it'snot really a regional setting.

Feel free to open again if not solved in the next release of HASS (likely 0.86)

kellerza, 2019-01-14:
You probably got the `example.py` error since you are trying to log in twice with the same user

closed 2019-01-14 by kellerza

kellerza, 2019-01-17:
Your voltage and frequency is correct with the latest version