kellerza/pysma #17: Can’t execute example.py if there is a file named pysma.py

tmsch13, 2019-02-25:
I stumbled over a strange bug when trying to execute and test the example script for pysma.
If there is a file pysma.py in the same directory I get the following error:
`DEBUG:asyncio:Using selector: EpollSelector
Traceback (most recent call last):
  File "get-sma-vaules.py", line 84, in <module>
    main()
  File "get-sma-vaules.py", line 80, in main
    loop, user=args.user, password=args.password, ip=args.ip))
  File "/usr/lib/python3.5/asyncio/base_events.py", line 466, in run_until_complete
    return future.result()
  File "/usr/lib/python3.5/asyncio/futures.py", line 293, in result
    raise self._exception
  File "/usr/lib/python3.5/asyncio/tasks.py", line 239, in _step
    result = coro.send(None)
  File "get-sma-vaules.py", line 33, in main_loop
    VAR['sma'] = pysma.SMA(session, ip, password=password, group=user)
AttributeError: module 'pysma' has no attribute 'SMA‘`

I can reproduce that behavior by simply create an empty file with `touch pysma.py`

kellerza, 2019-02-25:
And if you delete the file? :-)

tmsch13, 2019-02-26:
Then it works of cause. 
I’m new to python so if this is a normal python behaviour please close this issue

kellerza, 2019-02-26:
When you create `pysma.py` it seems to have a higher import priority than the package `pysma` (which is a folder `pysma\__init__.py` or this repo).

So when `example.py` imports pysam with `import pysma` you import your file (likely blank) that does not contain any SMA/any other code.

The [official docs](https://docs.python.org/3/reference/import.html) explains imports and packages in more detail

closed 2019-02-26 by kellerza