kellerza/pysma #18: no successful value decoded yet - sma energy meter

knoxxa, 2019-03-11:
using the sma energy meter instead of the sma sunny boy.  getting "no successful value decoded yet", any ideas?

config:

sensor:
#  - platform: yr
  - platform: sma
    host: 192.168.1.6
    password: !secret my_password
    scan_interval: 5
    ssl: true
    verify_ssl: false
    group: installer
    sensors:
      voltage_l1:
      voltage_l2:
      voltage_l3:
    custom:
      voltage_l1:
        key: "6100_00464800"
        unit: V
        factor: 100
      voltage_l2:
        key: "6100_00464900"
        unit: V
        factor: 100
      voltage_l3:
        key: "6100_00464A00"
        unit: V
        factor: 100

logs:

2019-03-11 20:24:11 DEBUG (MainThread) [pysma] {"result": {"015D-715540E1": {"6100_00464900": {"65": [{"val": 22091}]}, "6100_00464800": {"65": [{"val": 22493}]}, "6100_00464A00": {"65": [{"val": 0}]}}}}
2019-03-11 20:24:11 DEBUG (MainThread) [pysma] Sensor voltage_l2: No successful value decoded yet: {'65': [{'val': 22091}]}
2019-03-11 20:24:11 DEBUG (MainThread) [pysma] Sensor voltage_l3: No successful value decoded yet: {'65': [{'val': 0}]}
2019-03-11 20:24:11 DEBUG (MainThread) [pysma] Sensor voltage_l1: No successful value decoded yet: {'65': [{'val': 22493}]}

kellerza, 2019-03-13:
How do you query the SMA energy meter? (I never tried, but was under the impression it broadcasts measurements in UDP only)

The '65' seems to be an issue. Any idea what that refers to? To date I've only seen '1'

I never added an option to supply a `path` (custome JMES expression) to HASS - you need `'"65"[0].val'` as the last param to the Sensor constructor - see [here](https://github.com/kellerza/pysma/blob/master/pysma/__init__.py#L123) - maybe you can edit `__init__.py` line 101-103 on you machine and try it (without the `custom:` config in hass)

Will have to add a `path` option to the custom part for a permanent fix

kellerza, 2019-07-04:
You should be able to add a [path:](https://github.com/home-assistant/home-assistant/pull/24558) in the current HA:

Try this:

```yaml
      voltage_l1:
        key: "6100_00464800"
        unit: V
        factor: 100
        path: '"65"[0].val'
      voltage_l2:
        key: "6100_00464900"
        unit: V
        factor: 100
        path: '"65"[0].val'

```

closed 2019-07-11 by kellerza

wica128, 2019-09-06:
Hi,

Just for anyone, who is looking at this for a SUNNY ISLAND 4.4M
add `path: '"7"[0].val'` to the HA config. Looks like most values work.

What I'm wondering is, what is the meaning of "7" ?

kellerza, 2019-09-06:
Thanks for the feedback @wica128 ! No idea what that means, must be some internal number for various products...