kellerza/pysma #19: Multiple inverters?

MYeager1967, 2019-05-20:
Any luck on resolving the multiple inverters not working issue? Sorry to clutter the thread here. I have no idea where else to ask the question and I just managed to get the component to work only to realize I can only read one inverter...

MYeager1967, 2019-05-21:
I found a workaround for this using custom. However, I have a rolling error in my log:

Traceback (most recent call last):
  File "/usr/src/app/homeassistant/components/sma/sensor.py", line 128, in async_sma
    values = await sma.read(used_sensors)
  File "/usr/local/lib/python3.7/site-packages/pysma/__init__.py", line 243, in read
    body = yield from self._fetch_json(URL_VALUES, payload=payload)
  File "/usr/local/lib/python3.7/site-packages/pysma/__init__.py", line 201, in _fetch_json
    self._url + url, **params)
  File "/usr/local/lib/python3.7/site-packages/aiohttp/client.py", line 497, in _request
    await resp.start(conn)
  File "/usr/local/lib/python3.7/site-packages/aiohttp/client_reqrep.py", line 844, in start
    message, payload = await self._protocol.read()  # type: ignore  # noqa
  File "/usr/local/lib/python3.7/site-packages/aiohttp/streams.py", line 588, in read
    await self._waiter
aiohttp.client_exceptions.ServerDisconnectedError: None
2019-05-21 13:45:39 ERROR (MainThread) [homeassistant.core] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/src/app/homeassistant/components/sma/sensor.py", line 128, in async_sma
    values = await sma.read(used_sensors)
  File "/usr/local/lib/python3.7/site-packages/pysma/__init__.py", line 243, in read
    body = yield from self._fetch_json(URL_VALUES, payload=payload)
  File "/usr/local/lib/python3.7/site-packages/pysma/__init__.py", line 201, in _fetch_json
    self._url + url, **params)
  File "/usr/local/lib/python3.7/site-packages/aiohttp/client.py", line 497, in _request
    await resp.start(conn)
  File "/usr/local/lib/python3.7/site-packages/aiohttp/client_reqrep.py", line 844, in start
    message, payload = await self._protocol.read()  # type: ignore  # noqa
  File "/usr/local/lib/python3.7/site-packages/aiohttp/streams.py", line 588, in read
    await self._waiter
aiohttp.client_exceptions.ServerDisconnectedError: None
2019-05-21 13:45:51 ERROR (MainThread) [homeassistant.core] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/src/app/homeassistant/components/sma/sensor.py", line 128, in async_sma
    values = await sma.read(used_sensors)
  File "/usr/local/lib/python3.7/site-packages/pysma/__init__.py", line 243, in read
    body = yield from self._fetch_json(URL_VALUES, payload=payload)
  File "/usr/local/lib/python3.7/site-packages/pysma/__init__.py", line 201, in _fetch_json
    self._url + url, **params)
  File "/usr/local/lib/python3.7/site-packages/aiohttp/client.py", line 497, in _request
    await resp.start(conn)
  File "/usr/local/lib/python3.7/site-packages/aiohttp/client_reqrep.py", line 844, in start
    message, payload = await self._protocol.read()  # type: ignore  # noqa
  File "/usr/local/lib/python3.7/site-packages/aiohttp/streams.py", line 588, in read
    await self._waiter
aiohttp.client_exceptions.ServerDisconnectedError: None
2019-05-21 13:45:57 ERROR (MainThread) [homeassistant.core] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/src/app/homeassistant/components/sma/sensor.py", line 128, in async_sma
    values = await sma.read(used_sensors)
  File "/usr/local/lib/python3.7/site-packages/pysma/__init__.py", line 243, in read
    body = yield from self._fetch_json(URL_VALUES, payload=payload)
  File "/usr/local/lib/python3.7/site-packages/pysma/__init__.py", line 201, in _fetch_json
    self._url + url, **params)
  File "/usr/local/lib/python3.7/site-packages/aiohttp/client.py", line 497, in _request
    await resp.start(conn)
  File "/usr/local/lib/python3.7/site-packages/aiohttp/client_reqrep.py", line 844, in start
    message, payload = await self._protocol.read()  # type: ignore  # noqa
  File "/usr/local/lib/python3.7/site-packages/aiohttp/streams.py", line 588, in read
    await self._waiter
aiohttp.client_exceptions.ServerDisconnectedError: None

Any idea how to correct?

kellerza, 2019-06-13:
Hi @MYeager1967, unfortunately the custom option that I proposed on HASS community forums is the only option today. Have not had time to modify this to allow multiple inverters yet.

I have no idea why you would get `aiohttp.client_exceptions.ServerDisconnectedError` - are you trying to log into this using your web browser at the same time? / is this over wired/wifi?

kellerza, 2019-09-09:
https://github.com/home-assistant/home-assistant/issues/20375 - custom workaround

kellerza, 2021-05-06:
Please try the configuring SMA via the UI... multiple inverters should now be supported - see https://www.home-assistant.io/blog/2021/05/05/release-20215/#breaking-changes