kellerza/pysma #20: SUNNY TRIPOWER 6.0 / Inputs [A] and [B] / Home Assistant

Hooch76, 2019-06-13:
Hi,

this is not really an issue but rather a question. Details see https://community.home-assistant.io/t/sma-energy-meter-in-home-assistant/47482/69. 
Seems as if pysma is only reading the value of input [A] of the inverter, but not [B]. Or is this just a wrong configuration on my end?

kellerza, 2019-06-13:
Please share your homeassistant config. Do you have entries ending with _0 and _1?

Hooch76, 2019-06-13:
This is the Home Assistant configuration / sensor setup:

- platform: sma
  host: <my-ip>
  ssl: true
  verify_ssl: false
  group: user
  password: <yesthereisone>
  scan_interval: 60
  sensors:
    pv_power:
    pv_cust_leistung:
    pv_cust_spannung:
    pv_cust_strom:
  custom:
    pv_cust_leistung:
      key: "6380_40251E00"
      unit: W
    pv_cust_spannung:
      key: "6380_40451F00"
      unit: V
      factor: 100
    pv_cust_strom:
      key: "6380_40452100"
      unit: A
      factor: 1000

And this is the result:

[image]

kellerza, 2019-06-13:
Try adding the "B" set (default is _0)

```
    pv_cust_leistungb:
      key: "6380_40251E00_1"
      unit: W
    pv_cust_spannungb:
      key: "6380_40451F00_1"
      unit: V
      factor: 100
    pv_cust_stromb:
      key: "6380_40452100_1"
      unit: A
```

Hooch76, 2019-06-13:
Excellent. That did the trick. Thanks!

closed 2019-06-13 by kellerza