kellerza/pysma #21: Could not start session, Session ID expected

dimitriy-k, 2019-06-19:
Sunny Boy 2.5
Home Assistant 0.94.3

```
Could not start session, Session ID expected [result.sid], got {'result': {'sid': None}}
9:47 PM components/sma/sensor.py (ERROR) - message first occured at 9:47 PM and shows up 4 times
```

configuration:
```
- platform: sma
  host: 192.xxx.xxx.xxx
  password: !secret password
  verify_ssl: false
  group: user
  sensors:
    daily_yield:
    status:
```

kellerza, 2019-06-20:
The component is unable to log in due to some reason. Most likely you have another session open...

dimitriy-k, 2019-06-21:
No, no connections from my computer (no one else can connect I hope)... tried with "installer" and "user", both does not work

kellerza, 2019-06-21:
Has this ever worked?

Can you connect/login via a browser?

dimitriy-k, 2019-06-22:
Never worked with HA. Via browser I am able to login as installer or user, using the same credentials as with HA

kellerza, 2019-06-22:
Might be something else then. Can you use chrome dev tools to capture the login request and send me the request and response?
(Use my handle at gmail . com)

Are you sure you send the correct password in HA?
What firmware version do you have on your SB2.5?

dimitriy-k, 2019-06-25:
yesterday upgraded firmware to 2.5.12.R, still the same issue.

[image]
[image]

dimitriy-k, 2019-06-26:
LoL... I found the problem: my password was 20 chars long, but I saw in the request that it was trimmed to 12. Looks like via the browser they trim it with javascript and send it, but with HA whole password was send. Updated HA password to the same 12 chars and it works now. Such a crap 🤦‍♂

closed 2019-06-26 by dimitriy-k

kellerza, 2019-06-26:
Thanks for letting me know! Was stumped, since your request seemed fine... really strange way to accept longer passwords