kellerza/pysma #22: Home assistant sensor stops updating

Hooch76, 2019-06-24:
The SMA sensor in Home Assistant works fine for some time (<24h) but then stops updating. Restarting Home Assistant usually fixes the issue . 

This is what I get in the logs then: 

```
2019-06-24 06:29:16 ERROR (MainThread) [homeassistant.core] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/opt/homeassistant/lib/python3.7/site-packages/homeassistant/components/sma/sensor.py", line 128, in async_sma
    values = await sma.read(used_sensors)
  File "/opt/homeassistant/lib/python3.7/site-packages/pysma/__init__.py", line 254, in read
    if sen.extract_value(body):
  File "/opt/homeassistant/lib/python3.7/site-packages/pysma/__init__.py", line 51, in extract_value
    res = next(iter(jmespath.search(JMESPATH_BASE, json_body)))
TypeError: 'NoneType' object is not iterable
``` 

Plus 

```
2019-06-24 06:30:11 ERROR (MainThread) [homeassistant.core] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/opt/homeassistant/lib/python3.7/site-packages/aiohttp/connector.py", line 924, in _wrap_create_connection
    await self._loop.create_connection(*args, **kwargs))
  File "/usr/local/lib/python3.7/asyncio/base_events.py", line 959, in create_connection
    raise exceptions[0]
  File "/usr/local/lib/python3.7/asyncio/base_events.py", line 946, in create_connection
    await self.sock_connect(sock, address)
  File "/usr/local/lib/python3.7/asyncio/selector_events.py", line 464, in sock_connect
    return await fut
  File "/usr/local/lib/python3.7/asyncio/selector_events.py", line 494, in _sock_connect_cb
    raise OSError(err, f'Connect call failed {address}')
OSError: [Errno 113] Connect call failed ('<SMA-IP>', 443)

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/opt/homeassistant/lib/python3.7/site-packages/homeassistant/components/sma/sensor.py", line 128, in async_sma
    values = await sma.read(used_sensors)
  File "/opt/homeassistant/lib/python3.7/site-packages/pysma/__init__.py", line 243, in read
    body = yield from self._fetch_json(URL_VALUES, payload=payload)
  File "/opt/homeassistant/lib/python3.7/site-packages/pysma/__init__.py", line 201, in _fetch_json
    self._url + url, **params)
  File "/opt/homeassistant/lib/python3.7/site-packages/aiohttp/client.py", line 476, in _request
    timeout=real_timeout
  File "/opt/homeassistant/lib/python3.7/site-packages/aiohttp/connector.py", line 522, in connect
    proto = await self._create_connection(req, traces, timeout)
  File "/opt/homeassistant/lib/python3.7/site-packages/aiohttp/connector.py", line 854, in _create_connection
    req, traces, timeout)
  File "/opt/homeassistant/lib/python3.7/site-packages/aiohttp/connector.py", line 992, in _create_direct_connection
    raise last_exc
  File "/opt/homeassistant/lib/python3.7/site-packages/aiohttp/connector.py", line 974, in _create_direct_connection
    req=req, client_error=client_error)
  File "/opt/homeassistant/lib/python3.7/site-packages/aiohttp/connector.py", line 931, in _wrap_create_connection
    raise client_error(req.connection_key, exc) from exc
aiohttp.client_exceptions.ClientConnectorError: Cannot connect to host <SMA-IP>:443 ssl:None [Connect call failed ('<SMA-IP>', 443)]
```

Hooch76, 2019-06-27:
Additional information: seems as if it might be related to WiFi connectivity of my Home Assistant box. 
Connection randomly drops for a little while (<1min) and it seems that pysma does not recover then. Is this possible? (All other components are not affected though. They throw an error when no network is available but will be updated once connection is re-established).

kellerza, 2019-06-27:
It can probably be improved by adding ClientConnectError or its base, [here](https://github.com/kellerza/pysma/blob/master/pysma/__init__.py#L203)

```python
            except asyncio.TimeoutError, aiohttp.client_exceptions.ClientConnectorError:
```

you have to also `import aiohttp` at the top

kellerza, 2019-07-11:
Let me know if you still experience issues after the new version

closed 2019-07-11 by kellerza

Hooch76, 2019-07-16:
pysma seems now to be able to recover from network hiccups, as seen from the home assistant debug log:

```
WARNING (MainThread) [pysma] No 'result' in reply from SMA, got: {'err': 'Could not connect to SMA at https://x.y.z.z (timeout)'}
```

There's another (unrelated) issue where pysma runs into the following error

```
WARNING (MainThread) [pysma] 401 error detected, closing session to force another login attempt
````
 The problem is that I cannot see any more tries of pysma connecting to the inverter once that occured (and data is not updated then).

kellerza, 2019-07-17:
No errors after that?

try wrapping [this](https://github.com/kellerza/pysma/blob/master/pysma/__init__.py#L230) line in try/finally

```
        try:
            yield from self._fetch_json(URL_LOGOUT, {})
        finally:
            self.sma_sid = None
```

but even that should have thrown another error after the one you shared

Hooch76, 2019-07-17:
Nothing at all from pysma after that ... I've updated it and let you know how it goes. Thanks

Hooch76, 2019-07-19:
This is how the logfile from home assistant looks like:

```
2019-07-19 06:00:47 WARNING (MainThread) [pysma] 401 error detected, closing session to force another login attempt
2019-07-19 06:01:48 WARNING (MainThread) [pysma] 401 error detected, closing session to force another login attempt
2019-07-19 06:02:58 WARNING (MainThread) [pysma] No 'result' in reply from SMA, got: {'err': 'Could not connect to SMA at https://x.x.x.x (timeout)'}
2019-07-19 06:09:01 WARNING (MainThread) [pysma] 401 error detected, closing session to force another login attempt
2019-07-19 06:39:22 WARNING (MainThread) [pysma] 401 error detected, closing session to force another login attempt
2019-07-19 07:40:21 WARNING (MainThread) [pysma] 401 error detected, closing session to force another login attempt
2019-07-19 08:41:17 WARNING (MainThread) [pysma] 401 error detected, closing session to force another login attempt
```
Also have a look at the timestamps. It is retried once an hour only. 
Any chance that "closing session" somehow does not work? Restarting home assistant immediately fixes the issue.

Human, 2019-08-04:
I'm not sure why exactly this is closed. I have the same issue.

kellerza, 2019-08-04:
@Human, the original or the new issue?

Human, 2019-08-04:
@kellerza I'm having the titular "Home assistant sensor stops updating" issue with the last-mentioned error logs:

`2019-07-30 06:56:16 WARNING (MainThread) [pysma] 401 error detected, closing session to force another login attempt`

JoooostB, 2019-08-07:
@Human It has been closed because the changes have been merged to the PySMA 0.3.2 release which is already available in home-assistant:dev

Human, 2019-08-10:
Thanks. I still have the same bug, as the title describes. Would it be best for me to a file a new bug report?

JoooostB, 2019-08-10:
@Human Yes, please do so! I just checked and I'm experiencing it again too...

Hooch76, 2019-08-19:
I've created a new issue, see https://github.com/kellerza/pysma/issues/25