kellerza/pysma #23: SMA forced SSL

sirfalo, 2019-07-03:
Hi

I was trying to get some support on HA forum, but so far, no luck.
I'm trying to connect SMA TriPower inverter, but no luck.
Thins like password, connection I checked dozen of times.
I'm using module integrated into latest HA 0.94.3.
As default, SMA module should NOT use SSL, but even disabled in configuration, it constantly have this error.

> aiohttp.client_exceptions.ClientConnectorCertificateError: Cannot connect to host 192.168.1.250:443 ssl:True [SSLCertVerificationError: (1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1056)')]

Any ideas??

sirfalo, 2019-07-03:
One more, I disables "veryfi_ssl" and"ssl" (set to FALSE) and now I have:

> 2019-07-03 23:55:16 ERROR (MainThread) [pysma] Could not start session, Could not connect to SMA at http://192.168.1.250 (timeout), got {'err': 'Could not connect to SMA at http://192.168.1.250 (timeout)'}

This is that it's 100% accessible...

kellerza, 2019-07-04:
What if you ping this from your HA machine?

sirfalo, 2019-07-06:
Hi

Well, no issue, pings fast and easy...

kellerza, 2019-07-07:
Reading your first comment again...

ssl: true
verify_ssl: false

sirfalo, 2019-07-07:
Unfortunatelly....

> Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/sma/sensor.py", line 130, in async_sma
    values = await sma.read(used_sensors)
  File "/usr/local/lib/python3.7/site-packages/pysma/__init__.py", line 243, in read
    body = yield from self._fetch_json(URL_VALUES, payload=payload)
  File "/usr/local/lib/python3.7/site-packages/pysma/__init__.py", line 201, in _fetch_json
    self._url + url, **params)
  File "/usr/local/lib/python3.7/site-packages/aiohttp/client.py", line 507, in _request
    raise ClientOSError(*exc.args) from exc
aiohttp.client_exceptions.ClientOSError: Cannot write to closing transport
2019-07-07 22:10:54 WARNING (MainThread) [pysma] Sensor 6100_00543100: Not found in {'6400_00260100': {'1': [{'val': 1374907}]}, '6100_0046C200': {'1': [{'val': None}]}}
2019-07-07 22:11:00 WARNING (MainThread) [pysma] Sensor 6100_00543100: Not found in {'6400_00260100': {'1': [{'val': 1374907}]}, '6100_0046C200': {'1': [{'val': None}]}}
2019-07-07 22:11:06 WARNING (MainThread) [pysma] Sensor 6100_00543100: Not found in {'6400_00260100': {'1': [{'val': 1374907}]}, '6100_0046C200': {'1': [{'val': None}]}}
2019-07-07 22:11:12 WARNING (MainThread) [pysma] Sensor 6100_00543100: Not found in {'6400_00260100': {'1': [{'val': 1374907}]}, '6100_0046C200': {'1': [{'val': None}]}}
2019-07-07 22:11:18 WARNING (MainThread) [pysma] Sensor 6100_00543100: Not found in {'6400_00260100': {'1': [{'val': 1374907}]}, '6100_0046C200': {'1': [{'val': None}]}}
2019-07-07 22:11:24 WARNING (MainThread) [pysma] Sensor 6100_00543100: Not found in {'6400_00260100': {'1': [{'val': 1374907}]}, '6100_0046C200': {'1': [{'val': None}]}}
2019-07-07 22:11:30 WARNING (MainThread) [pysma] Sensor 6100_00543100: Not found in {'6400_00260100': {'1': [{'val': 1374907}]}, '6100_0046C200': {'1': [{'val': None}]}}
2019-07-07 22:11:36 WARNING (MainThread) [pysma] Sensor 6100_00543100: Not found in {'6400_00260100': {'1': [{'val': 1374907}]}, '6100_0046C200': {'1': [{'val': None}]}}
2019-07-07 22:11:42 WARNING (MainThread) [pysma] Sensor 6100_00543100: Not found in {'6400_00260100': {'1': [{'val': 1374907}]}, '6100_0046C200': {'1': [{'val': None}]}}
2019-07-07 22:11:48 WARNING (MainThread) [pysma] Sensor 6100_00543100: Not found in {'6400_00260100': {'1': [{'val': 1374907}]}, '6100_0046C200': {'1': [{'val': None}]}}

kellerza, 2019-07-08:
Much better, just don’t use sensors your device does not support!

See the list of sensor name to code mappings in the source...

sirfalo, 2019-07-08:
Hi

Thanks for feedback!
Well, this was missing in HA documentation about SMA devices - if something is not supported, none will work. Now it's better :)
However, I setup:
PV Power
Pv Gen meter
total yield
daily yield

I got values from Gen meter and total yield.
Daily and PV Power is empty (PV Power I understand as now it's night, but daily?).

Nothing in logs.

kellerza, 2019-07-11:
The new version should improve things.

You are welcome to improve the docs

closed 2019-07-11 by kellerza

sirfalo, 2019-07-13:
Great, thank you!

pybouvy, 2020-04-05:
Hello and help. I have the same problem:
pi@raspberrypi:~/Documents $ python3 SMAretrieve.py https://192.168.1.160 installer 1234
DEBUG:asyncio:Using selector: EpollSelector
ERROR:asyncio:SSL handshake failed on verifying the certificate
protocol: <asyncio.sslproto.SSLProtocol object at 0x75c21c90>
transport: <_SelectorSocketTransport fd=6 read=polling write=<idle, bufsize=0>>
Traceback (most recent call last):
  File "/usr/lib/python3.7/asyncio/sslproto.py", line 625, in _on_handshake_complete
    raise handshake_exc
  File "/usr/lib/python3.7/asyncio/sslproto.py", line 189, in feed_ssldata
    self._sslobj.do_handshake()
  File "/usr/lib/python3.7/ssl.py", line 763, in do_handshake
    self._sslobj.do_handshake()
ssl.SSLCertVerificationError: [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1056)
ERROR:asyncio:SSL error in data received
protocol: <asyncio.sslproto.SSLProtocol object at 0x75c21c90>
transport: <_SelectorSocketTransport closing fd=6 read=idle write=<idle, bufsize=0>>
Traceback (most recent call last):
  File "/usr/lib/python3.7/asyncio/sslproto.py", line 526, in data_received
    ssldata, appdata = self._sslpipe.feed_ssldata(data)
  File "/usr/lib/python3.7/asyncio/sslproto.py", line 189, in feed_ssldata
    self._sslobj.do_handshake()
  File "/usr/lib/python3.7/ssl.py", line 763, in do_handshake
    self._sslobj.do_handshake()
ssl.SSLCertVerificationError: [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1056)
ERROR:asyncio:SSL handshake failed on verifying the certificate

kellerza, 2020-04-05:
Hi @pybouvy The SMAretrieve.py example does not set SSL verification to false and you need this for https when the server does not have a valid cert.

The two options is to either modify this example code and add this, or if you use homeassistant set `verify_ssl: false` in the config

pybouvy, 2020-04-06:
Hello. Many thanks for the answer. It doesn't seem possible to change the ssl setting on the router so I need to modify the example code. I'm not confident enough with these  kind of request. Could you help me and indicate what I precisely should change or add in te code and where? If you don't mind of course?
Thanks by advance...
Regards.