kellerza/pysma #25: [pysma] 401 error detected, closing session to force another login attempt

Hooch76, 2019-08-19:
Follow-up to this issue: https://github.com/kellerza/pysma/issues/22. 
I am also experiencing 

```
[pysma] 401 error detected, closing session to force another login attempt
```
This error does not recover itself. Home Assistant has to be restarted in order to get data again.

kellerza, 2019-09-03:
Is this solved in 0.98?

JoooostB, 2019-09-03:
It seems to have broken something else: (running pysma==0.3.4 on home-assistant:dev)
```
2019-09-03 07:49:11 ERROR (MainThread) [homeassistant.core] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/sma/sensor.py", line 182, in async_sma
    values = await sma.read(used_sensors)
  File "/usr/local/lib/python3.7/site-packages/pysma/__init__.py", line 246, in read
    payload = {"destDev": [], "keys": list(set([s.key for s in sensors]))}
  File "/usr/local/lib/python3.7/site-packages/pysma/__init__.py", line 246, in <listcomp>
    payload = {"destDev": [], "keys": list(set([s.key for s in sensors]))}
AttributeError: 'str' object has no attribute 'key'
```

kellerza, 2019-09-03:
That was fixed a while ago, so maybe you should update?

JoooostB, 2019-09-03:
That's weird, I was running a very recent version of the home-assistant:dev (one that already contained the new pysma) container but a new pull did fix the issue. Will update on here if I encounter a 401 later today.

kellerza, 2019-09-03:
That error was not related to the pysma library, but the way the hass component called it.... So you'll need the new hass 😉

Human, 2019-09-03:
In 0.98.1, I still get disconnected without reconnecting:
```
2019-09-03 15:08:41 WARNING (MainThread) [pysma] 401 error detected, closing session to force another login attempt
```

Human, 2019-09-03:
FWIW, it didn't look to me that the HASS PR for this fix made it into 0.98.x. Maybe 0.99?

Hooch76, 2019-09-04:
I am on 0.98.2 and also still get 
```
WARNING (MainThread) [pysma] 401 error detected, closing session to force another login attempt
```

Pjanssen, 2019-09-19:
I'm running hass 0.99, and I'm also getting these errors.

Hooch76, 2019-09-24:
Upgraded to Home Assistant 0.99.2 yesterday and still 
```
[pysma] 401 error detected, closing session to force another login attempt
```
which requires a restart of the Home Assistant service to recover from that situation.

Human, 2019-09-24:
Also on 0.99.2. It seems to go longer than before, but I still get permanent connection failures after a time, and lots of these in my logs:

```
WARNING (MainThread) [pysma] 401 error detected, closing session to force another login attempt
```

Human, 2019-09-28:
FWIW, pysma is still able to see 2 of my 4 inverters after about a week of Home Assistant uptime. The other 2 last connected 4 days ago and do not appear to be able to recover. Is there a more verbose logging option I could enable to see the nature of the permanent failure?

johansmitsnl, 2019-10-29:
I also still have this issue, after +/- 3 days I lose connection to HA with the same error: `[pysma] 401 error detected, closing session to force another login attempt`

Also see issue: home-assistant/home-assistant#27154

johansmitsnl, 2019-12-07:
It can be reproduced when the connection to the SMA is lost and then recovers. Is the session torn down correctly or does it needs to wait a little longer?

If you can suggest a change I can test it now.

Human, 2019-12-12:
I have a fix for this locally, but it isn't optimal because Home Assistant expects to read sensor values that aren't there and logs a ton of warnings while the pysma code keeps trying to log back in without hitting a timeout.

In the absence of an optimal fix, I'll submit it as a PR this weekend. Maybe someone else can optimize it.

Human, 2019-12-12:
@kellerza Please assign this to me.

johansmitsnl, 2019-12-12:
@Human I can take a look at it, looking forward to the PR.

Human, 2019-12-14:
@johansmitsnl See https://github.com/kellerza/pysma/pull/34

johansmitsnl, 2019-12-15:
@Human I did the reboot test and it worked for me. I will test this fix for the next 4 days and let you know if it solved it permanently for me :+1: .

The PR looks logic to me, not sure if we should reconnect on all errors. To me the 401 should be the one to reconnect but I don't know the SMA interface that well to know if there are other moments where a reconnect is needed.

Human, 2019-12-18:
@johansmitsnl I can see both sides of the argument to only connect on 401s vs all errors, but I would lean toward reconnecting after all errors (maybe with an added timeout, though) because of the issue you cited regarding the unknown inner workings of the server. Plus, since nobody was able to stay connected for more than 3-4 days in a row, maybe this also avoids some other error that pops up every 15 days. :man_shrugging: The new verbose logging will make it clear if this gets better or worse.

Given that _this specific issue_ is due to 401s popping up after a few days, it may be fine to just look for 401s. I'll leave it to @kellerza to decide if he wants to check for more than 401s.

johansmitsnl, 2019-12-18:
@Human for me it is working until today. Will give feedback this weekend or sooner if it fails but for me it solved the issue 👍

JoooostB, 2019-12-19:
Would love to see this getting merged! 😃

closed 2020-01-14 by kellerza

johansmitsnl, 2020-01-18:
What is the next step to get this fix in HA so that we can close the issue home-assistant/home-assistant#27154 ?

johansmitsnl, 2020-01-26:
Wat is holding a release with this fix back?

PR #34 (2020-01-14): on any connection error, reconnect
This fixes #25 , but it may not be optimal. I would have preferred to provide a unit test that fails before this patch and passes after it, but I am not familiar enough with async in python to do it.