kellerza/pysma #26: Script not working on Sunny Boy 4.0 AV-41

bccrew, 2019-09-02:
Hi,

Im trying to run your script but I get a timeout error.

root@OPENHABTEST:/home/user# python3 example.py 192.168.1.33 installer Password123
DEBUG:asyncio:Using selector: EpollSelector
ERROR:pysma:Could not start session, Could not connect to SMA at http://192.168.1.33 (timeout), got {'err': 'Could not connect to SMA at http://192.168.1.33 (timeout)'}
INFO:__main__:No session ID

Device type	Sunny Boy 4.0 AV-41
Software package	1.1.32.R	
Software version	1.4.63.R

There seems to be a redirect to https by default when you try to visit the webpage.

Anything I can try?

kellerza, 2019-09-03:
You need https

Instead of just the IP parameter, use` https://IP`

You will likely also have to pass allow_insecure to the session - https://github.com/kellerza/pysma/blob/4fe39d3191ac1ecad74d1319f4ac43a5371e8eb2/example.py#L32 since the unit will have a self signed cert

In the homeassistant sensor’s setup method you will also find an example of these params

bccrew, 2019-09-03:
Hi @kellerza,

Thanks for your reply. I'm using Openhab so I was just trying out the python example to see the output. How can I pass this allow_insecure. Is it a parameter of aiohttp?

My knowledge of python is very poor I'm afraid.

kellerza, 2019-09-03:
Try `verify_ssl`...

kellerza, 2019-09-03:
It might also be `ssl` only since aiohttp 3.0 - https://docs.aiohttp.org/en/stable/client_reference.html

bccrew, 2019-09-03:
I think it should be used with ClientSession.get

Parameters: 
verify_ssl (bool) – perform SSL certificate validation for HTTPS requests (enabled by default). 
May be disabled to skip validation for sites with invalid certificates.  

Deprecated since version 2.3: Pass verify_ssl to ClientSession.get() etc.

I cannot seem to construct it the right way. I'm very new to programming :)

   async with aiohttp.ClientSession.get(loop=loop,verify_ssl=false) as session:

bccrew, 2019-09-03:
I'm trying various options, described here:

http://docs.aiohttp.org/en/stable/client_advanced.html#ssl-control-for-tcp-sockets

It seems like a pain in the ass with self signed certificates :)

bccrew, 2019-09-03:
DEBUG:asyncio:Using selector: EpollSelector
Traceback (most recent call last):
  File "example.py", line 81, in <module>
    main()
  File "example.py", line 76, in main
    main_loop(loop, user=args.user, password=args.password, ip=args.ip)
  File "/usr/lib/python3.6/asyncio/base_events.py", line 484, in run_until_complete
    return future.result()
  File "example.py", line 32, in main_loop
    async with aiohttp.ClientSession.get(loop=loop,ssl=False) as session:
TypeError: get() missing 2 required positional arguments: 'self' and 'url'

bccrew, 2019-09-03:
I have setup a test system with Home Assistant, I'm able to connect from there as well but I get these errors:

```
2019-09-03 13:39:23 INFO (MainThread) [homeassistant.core] Starting Home Assistant
2019-09-03 13:39:23 INFO (MainThread) [homeassistant.core] Timer:starting
2019-09-03 13:39:29 WARNING (MainThread) [pysma] Sensor 6100_00543100: Not found in {'6400_00260100': {'1': [{'val': 76717}]}, '6100_0046C200': {'1': [{'val': 2356}]}}
2019-09-03 13:39:29 WARNING (MainThread) [pysma] Sensor 6400_00543A00: Not found in {'6400_00260100': {'1': [{'val': 76717}]}, '6100_0046C200': {'1': [{'val': 2356}]}}
2019-09-03 13:39:34 WARNING (MainThread) [pysma] Sensor 6100_00543100: Not found in {'6400_00260100': {'1': [{'val': 76720}]}, '6100_0046C200': {'1': [{'val': 2270}]}}
2019-09-03 13:39:34 WARNING (MainThread) [pysma] Sensor 6400_00543A00: Not found in {'6400_00260100': {'1': [{'val': 76720}]}, '6100_0046C200': {'1': [{'val': 2270}]}}
2019-09-03 13:39:40 WARNING (MainThread) [pysma] Sensor 6100_00543100: Not found in {'6400_00260100': {'1': [{'val': 76724}]}, '6100_0046C200': {'1': [{'val': 2236}]}}
2019-09-03 13:39:40 WARNING (MainThread) [pysma] Sensor 6400_00543A00: Not found in {'6400_00260100': {'1': [{'val': 76724}]}, '6100_0046C200': {'1': [{'val': 2236}]}}
2019-09-03 13:39:46 WARNING (MainThread) [pysma] Sensor 6100_00543100: Not found in {'6400_00260100': {'1': [{'val': 76728}]}, '6100_0046C200': {'1': [{'val': 2253}]}}
2019-09-03 13:39:46 WARNING (MainThread) [pysma] Sensor 6400_00543A00: Not found in {'6400_00260100': {'1': [{'val': 76728}]}, '6100_0046C200': {'1': [{'val': 2253}]}}
2019-09-03 13:39:51 WARNING (MainThread) [pysma] Sensor 6100_00543100: Not found in {'6400_00260100': {'1': [{'val': 76731}]}, '6100_0046C200': {'1': [{'val': 2336}]}}
2019-09-03 13:39:51 WARNING (MainThread) [pysma] Sensor 6400_00543A00: Not found in {'6400_00260100': {'1': [{'val': 76731}]}, '6100_0046C200': {'1': [{'val': 2336}]}}
2019-09-03 13:39:54 ERROR (MainThread) [aiohttp.server] Unhandled exception
Traceback (most recent call last):
  File "/home/gert/homeassistant/lib/python3.6/site-packages/aiohttp/web_protocol.py", line 447, in start
    await resp.prepare(request)
  File "/home/gert/homeassistant/lib/python3.6/site-packages/aiohttp/web_response.py", line 353, in prepare
    return await self._start(request)
  File "/home/gert/homeassistant/lib/python3.6/site-packages/aiohttp/web_response.py", line 667, in _start
    return await super()._start(request)
  File "/home/gert/homeassistant/lib/python3.6/site-packages/aiohttp/web_response.py", line 410, in _start
    await writer.write_headers(status_line, headers)
  File "/home/gert/homeassistant/lib/python3.6/site-packages/aiohttp/http_writer.py", line 112, in write_headers
    self._write(buf)
  File "/home/gert/homeassistant/lib/python3.6/site-packages/aiohttp/http_writer.py", line 67, in _write
    raise ConnectionResetError('Cannot write to closing transport')
ConnectionResetError: Cannot write to closing transport
2019-09-03 13:39:57 WARNING (MainThread) [pysma] Sensor 6100_00543100: Not found in {'6400_00260100': {'1': [{'val': 76736}]}, '6100_0046C200': {'1': [{'val': 3146}]}}
2019-09-03 13:39:57 WARNING (MainThread) [pysma] Sensor 6400_00543A00: Not found in {'6400_00260100': {'1': [{'val': 76736}]}, '6100_0046C200': {'1': [{'val': 3146}]}}
2019-09-03 13:40:02 WARNING (MainThread) [pysma] Sensor 6100_00543100: Not found in {'6400_00260100': {'1': [{'val': 76740}]}, '6100_0046C200': {'1': [{'val': 2251}]}}
2019-09-03 13:40:02 WARNING (MainThread) [pysma] Sensor 6400_00543A00: Not found in {'6400_00260100': {'1': [{'val': 76740}]}, '6100_0046C200': {'1': [{'val': 2251}]}}
2019-09-03 13:40:08 WARNING (MainThread) [pysma] Sensor 6100_00543100: Not found in {'6400_00260100': {'1': [{'val': 76743}]}, '6100_0046C200': {'1': [{'val': 2156}]}}
2019-09-03 13:40:08 WARNING (MainThread) [pysma] Sensor 6400_00543A00: Not found in {'6400_00260100': {'1': [{'val': 76743}]}, '6100_0046C200': {'1': [{'val': 2156}]}}
2019-09-03 13:40:14 WARNING (MainThread) [pysma] Sensor 6100_00543100: Not found in {'6400_00260100': {'1': [{'val': 76747}]}, '6100_0046C200': {'1': [{'val': 2130}]}}
2019-09-03 13:40:14 WARNING (MainThread) [pysma] Sensor 6400_00543A00: Not found in {'6400_00260100': {'1': [{'val': 76747}]}, '6100_0046C200': {'1': [{'val': 2130}]}}
2019-09-03 13:40:19 WARNING (MainThread) [pysma] Sensor 6100_00543100: Not found in {'6400_00260100': {'1': [{'val': 76750}]}, '6100_0046C200': {'1': [{'val': 2121}]}}
2019-09-03 13:40:19 WARNING (MainThread) [pysma] Sensor 6400_00543A00: Not found in {'6400_00260100': {'1': [{'val': 76750}]}, '6100_0046C200': {'1': [{'val': 2121}]}}
2019-09-03 13:40:24 WARNING (MainThread) [pysma] Sensor 6100_00543100: Not found in {'6400_00260100': {'1': [{'val': 76753}]}, '6100_0046C200': {'1': [{'val': 2104}]}}
2019-09-03 13:40:24 WARNING (MainThread) [pysma] Sensor 6400_00543A00: Not found in {'6400_00260100': {'1': [{'val': 76753}]}, '6100_0046C200': {'1': [{'val': 2104}]}}
2019-09-03 13:40:30 WARNING (MainThread) [pysma] Sensor 6100_00543100: Not found in {'6400_00260100': {'1': [{'val': 76756}]}, '6100_0046C200': {'1': [{'val': 2096}]}}
2019-09-03 13:40:30 WARNING (MainThread) [pysma] Sensor 6400_00543A00: Not found in {'6400_00260100': {'1': [{'val': 76756}]}, '6100_0046C200': {'1': [{'val': 2096}]}}
[... cut ...]
2019-09-03 13:40:54 WARNING (MainThread) [pysma] Sensor 6100_00543100: Not found in {'6400_00260100': {'1': [{'val': 76771}]}, '6100_0046C200': {'1': [{'val': 2556}]}}
2019-09-03 13:40:54 WARNING (MainThread) [pysma] Sensor 6400_00543A00: Not found in {'6400_00260100': {'1': [{'val': 76771}]}, '6100_0046C200': {'1': [{'val': 2556}]}}
```

My configuration.yaml:

sensor:
  - platform: sma
    host: 192.168.1.33
    ssl: true
    verify_ssl: false
    password: Password123
    sensors:
      - current_consumption
      - total_consumption
      - pv_power
      - total_yield

```

kellerza, 2019-09-03:
It seems you can only read `pv_power` and `total_yield`.

`current_consumption` and `total_consumption` is not available.

See [here](https://github.com/kellerza/pysma/blob/master/pysma/__init__.py#L103) for the sensor definitions that links these codes to the names. You can also try make your own `custom_sensors` (see Home Assistant docs)

Try without any `sensors:` defined and all standard sensors will be included. It ight give you some guidance on what is supported by your SMA... Another option is to use Chrome Dev Tools / Firebug to see what is happening on the SMA webpage and maybe you can create some custom sensors

kellerza, 2019-09-03:
On the earlier Python question:

```python
# Currently there is no ssl option in example.py
async with aiohttp.ClientSession(loop=loop) as session:

# Add opion to ignore SSL
async with aiohttp.ClientSession(loop=loop, ssl=False) as session:

# You had this (which has an extra get - and the get needs a URL)
async with aiohttp.ClientSession.get(loop=loop,ssl=False) as session:

```

kellerza, 2019-09-03:
If you have some working custom sensors, please include them here for future reference / create a PR on the Readme.md file in the repo.... thanks!

bccrew, 2019-09-04:
Ok, got it. I will make a list of all things I can find.

bccrew, 2019-09-04:
Hi @kellerza, I have created a list of all possible values. I have not found the time for a PR yet but I can do it later this week. Please find the list attached. 
[SMA-HA.xlsx](https://github.com/kellerza/pysma/files/3576055/SMA-HA.xlsx)

bccrew, 2019-09-04:
Using this line still does not work too:

`async with aiohttp.ClientSession(loop=loop, ssl=False) as session:`

```
user@OPENHABTEST:~$ python3 example.py https://192.168.1.33 user Password123
DEBUG:asyncio:Using selector: EpollSelector
Traceback (most recent call last):
  File "example.py", line 81, in <module>
    main()
  File "example.py", line 76, in main
    main_loop(loop, user=args.user, password=args.password, ip=args.ip)
  File "/usr/lib/python3.6/asyncio/base_events.py", line 484, in run_until_complete
    return future.result()
  File "example.py", line 32, in main_loop
    async with aiohttp.ClientSession(loop=loop, ssl=False) as session:
TypeError: __init__() got an unexpected keyword argument 'ssl'

```

I'm not sure what we're doing wrong regarding iohttp syntax. Their documentation is not always that transparant it seems...?

kellerza, 2019-09-09:
@bccrew using a [TCPConnector](https://docs.aiohttp.org/en/stable/client_reference.html#tcpconnector) on the [session](https://docs.aiohttp.org/en/stable/client_reference.html#aiohttp.ClientSession) you should be able to skip ssl verification

This line

https://github.com/kellerza/pysma/blob/4fe39d3191ac1ecad74d1319f4ac43a5371e8eb2/example.py#L32

should probably be:

```python
async with aiohttp.ClientSession(loop=loop, connector=TCPConnector(ssl=False)) as session: 
```

closed 2019-09-09 by kellerza

bccrew, 2019-09-10:
Hi, sorry but still not working..

```
DEBUG:asyncio:Using selector: EpollSelector
Traceback (most recent call last):
  File "example.py", line 81, in <module>
    main()
  File "example.py", line 76, in main
    main_loop(loop, user=args.user, password=args.password, ip=args.ip)
  File "/usr/lib/python3.6/asyncio/base_events.py", line 484, in run_until_complete
    return future.result()
  File "example.py", line 32, in main_loop
    async with aiohttp.ClientSession(loop=loop, connector=TCPConnector(ssl=False)) as session:
NameError: name 'TCPConnector' is not defined
```