kellerza/pysma #29: Support for today yield and yesterday yield on Sunny Boy AV-41 models

bccrew, 2019-09-20:
Hi @kellerza 

My invertor does not have a instantaneous value for the Today Yield and Yesterday Yield. It would come in handy to store these in an influxdb. 

Would you be able to add some code that handles this as well? The code on the homepage on the invertor looks like this:

<tbody> 

<tr ng-repeat="row in tableContent" class="ng-scope"> 
  <td translate="lYieldToday" class="ng-scope">Today:</td> 
  <td style="text-align: right"> 
     
	
	<span ng-if="row.calcValKey" class="ng-binding ng-scope"> 12.91 kWh </span>
	 
  </td> 
</tr>
<tr ng-repeat="row in tableContent" class="ng-scope"> 
   <td translate="lYieldYesterday" class="ng-scope">Yesterday:</td> 
   <td style="text-align: right"> 
    
   
   <span ng-if="row.calcValKey" class="ng-binding ng-scope"> 20.77 kWh </span>
    
  </td>
</tr>
<tr ng-repeat="row in tableContent" class="ng-scope">
   <td translate="lYieldTotal" class="ng-scope">Total:</td> 
   <td style="text-align: right">  
   
   <span ng-if="row.calcValKey" class="ng-binding ng-scope"> 394.3 kWh </span>
   
  </td> 
</tr> 
</tbody> 

I will add the page itself + screenshot too.

[yield.zip](https://github.com/kellerza/pysma/files/3635733/yield.zip)

[image]

kellerza, 2019-09-20:
In HomeAssistant I use a “utility meter” on the total yield field, which works quite well. This can also be written to influxdb in hass as well.

These values you see are not directly available from the api & calculated in some way by the sunnyboy page (they do multiple calls to a different history api endpoint) - have not looked into this & no current plans.

With the utility meter we calculate it using the difference of the totals at day’s end, maybe this could work for you?

bccrew, 2019-09-23:
Yes, it could! Thanks!

closed 2019-09-23 by bccrew

kellerza, 2019-09-23:
I’ll put my config on the homepage for future reference..