kellerza/pysma #31: Wrong id used for retrieving PV power

eelcohn, 2019-09-28:
I've found that the key used to retrieve the `pv_power` value should be `6380_40251E00` instead of `6100_0046C200`. The `6100_0046C200` key indicates the power generated by the converter, whereas the `6380_40251E00` key indicates the power generated by the solar panels.
Since converters are never 100% efficient, there's a slight difference between the two values.

kellerza, 2019-10-05:
If 6100_0046C200 is usable power, that seems more correct than some measurement before it is converted, or not?

How are these two affected when you limit the setup not to feed any power into the grid?

eelcohn, 2019-10-06:
Hi, thanks for responding.

> If 6100_0046C200 is usable power, that seems more correct than some measurement before it is converted, or not?

`6100_0046C200` is indeed the usable power returned to the grid, but that value is always the same as `6100_40263F00`, which makes it pretty useless IMHO to have two variables which have the same value (at least in the setups that I've seen).

The `pv_` prefix in `pv_power` would suggest that it's value is about the solar panels, not about the converter or the power grid. If one would need the usable power, you could use the `grid_power` variable, or not?

On a sidenote, in the SMA management console there are actually six(!) power values available:
`6380_40251E00` which is the power generated by the solar panels
`6100_40263F00` which is the total power returned to the grid
`6100_40464000` which is the power returned to the grid via phase L1
`6100_40464100` which is the power returned to the grid via phase L2
`6100_40464200` which is the power returned to the grid via phase L3
`6100_0046C200` which is the power generated by the converter (which is always the same as `6100_40263F00`)

> How are these two affected when you limit the setup not to feed any power into the grid?

I'd have to do some tests on that one, I'll get back to you.

eelcohn, 2019-10-20:
> How are these two affected when you limit the setup not to feed any power into the grid?

I've done some testing and all values return `0` when the converter is switched off grid.

kellerza, 2021-03-24:
Removed `pv_power` from the code as `pv_power_a` already uses the correct key today. Also added some notes to the sensors. Thanks!

commit 5afdf9de (2021-03-24): Fix #31

#31 - `pv_power_a` uses the correct key
closed 2021-03-24 by kellerza