kellerza/pysma #32: SMA Sensor on Home Assistant

ekkesa, 2019-10-03:
Hi,

This looks very awesome - I have been using webscaper up until now, but this will be much cleaner. Unfortunately, I get errors connection errors (See below). The IP address it tries to connect to is correct, I also verified the user password on the Sunny Explorer software.

`2019-10-04 00:25:29 ERROR (MainThread) [pysma] Could not start session, Could not connect to SMA at http://192.168.0.3 (timeout), got {'err': 'Could not connect to SMA at http://192.168.0.3 (timeout)'}`

I have enabled all external communication on the module with their default values - Please see snippet of the settings - https://imgur.com/a/7he7joj

Your assistance will be much appreciated!

System info
Sunnyboy SB-5000TL-21 with Webconnect Module

bccrew, 2019-10-04:
Can you do a portscan (zenmap - all tcp ports) on the IP of the inverter? Can you see port 80 or 443 is open?

ekkesa, 2019-10-04:
Hi,

I did scans against it, and it seems the ports are filtered. 

Port 502 shows open with an mbap service running.

I checked the Sunny Explorer software again, but do not see any option to enable such services. The only option I see is the once in the image I posted earlier.

Any idea why the ports will seem filtered.

Thank you for getting back to me...

`me@server:~ $ nmap -Pn -p80 192.168.0.3

Starting Nmap 7.40 ( https://nmap.org ) at 2019-10-05 00:06 SAST
Nmap scan report for 192.168.0.3
Host is up.
PORT   STATE    SERVICE
80/tcp filtered http

Nmap done: 1 IP address (1 host up) scanned in 2.25 seconds
me@server:~ $ nmap -Pn -p443 192.168.0.3

Starting Nmap 7.40 ( https://nmap.org ) at 2019-10-05 00:06 SAST
Nmap scan report for 192.168.0.3
Host is up.
PORT    STATE    SERVICE
443/tcp filtered https

Nmap done: 1 IP address (1 host up) scanned in 2.23 seconds

me@server:~ $ nmap -Pn -p502 192.168.0.3

Starting Nmap 7.40 ( https://nmap.org ) at 2019-10-05 00:08 SAST
Nmap scan report for 192.168.0.3
Host is up (0.00071s latency).
PORT    STATE SERVICE
502/tcp open  mbap
`

kellerza, 2019-10-05:
What url do you point your webscraper to?

ekkesa, 2019-10-07:
The URL is: https://www.sunnyportal.com/Templates/PublicPage.aspx?page=<<UNIQUE ID>>

kellerza, 2019-10-08:
you need to access it on your local network using a web browser.

This library does not use sunnyportal.com

ekkesa, 2019-10-08:
Hi,

Thank you for the quick reply - I gathered that much, but it doesn't seem to offer a web service running on the local device. Are there any special settings on should enable on the SMA to get this up? The unit was installed end 2016.

kellerza, 2019-10-08:
Sorry, no idea what options/configurations you get on these SB’s!

You could try the homeassistant forum/see if SMA some customer support

closed 2019-10-08 by kellerza