kellerza/pysma #33: Anonymous access

pniedzwiedzinski, 2019-12-03:
Is there an option for not passing the password? I would like only to have access to public data

kellerza, 2019-12-14:
Not today. Honestly did not know there was any public data - do you have some examples?

pniedzwiedzinski, 2019-12-14:
I wanted to create a PR for that, but I couldn’t get it work nicely with pysma so I ended with writing my own https://github.com/pniedzwiedzinski/beSMArt

It’s a simple wrapper for one endpoint by now but there’s also endpoint for time series data - as soon as I have time I want to add it.

It would be nice if we could merge it into pysma but I found it a little confusing. Maybe you have idea how to do it.

And by examples you mean what data can I get? Then, there is total yield, current voltage and historic yield.

kellerza, 2019-12-15:
Very nice!

Some things making pysma a bit more complex:

- It is async (to fit in with homeassistant)
- it extracts the inverter ID - SOME_CONST_I_DONT_KNOW_THE_MEANING 😃 )
- I allows extracting data deeper in the result (using jmespath)

Will try see if i get a response from https://{ip}/dyn/getDashValues.json on my inverter, it should be fairly easy to read this

What model SMA do you have?
What is your use case to collect this data?

pniedzwiedzinski, 2019-12-15:
- STP6.0-3AV-40
- I wanted to connect it as well to the homeassistant

PR #46 (2021-03-21): Get data without login
resolve #33
closed 2021-03-21 by kellerza