kellerza/pysma #35: Unable to get integration to work with HA

dweston, 2020-01-15:
Hi @kellerza I have a SMA Sunny Boy 4200TL solar Inverter that I'm keen to connect to my Home Assistant server. I'm running hass.io v0.103.6.

I've added the following to my configuration.yaml file:
`sensor: !include sensors.yaml
`
And the following to the sensors.yaml file:
`# sma Inverter sensors
  - platform: sma
    host: !secret sma_IP
    password: !secret sma_password
# (Option) If no sensors are specified below all available sensors will be retrieved
    sensors:
      - current_consumption
      - total_consumption
      - pv_power
      - total_yield
`
I know I have the IP of the Inverter correctly in place, and the password is the same one I use for the [SBFspot](https://github.com/SBFspot/SBFspot/) configuration I currently use successfully to retrieve all Inverter data into a sqlite database.

I've run the yaml parsing checker in Home Assistant which throws no flags. I've restarted HA but no sma sensors appear to have been created.

May I presume this issue is the consequence of this particular Inverter being a pre-2017 model that doesn't provide a web interface?

sheevz0r, 2020-01-15:
I have the same problem. it seems like the connection is ok but it doesn't pull data.
sensors are created but unknown state.
my model is Sunny Boy 5000TL-21. sunnyportal is running fine. 
Also looking forward to get this up and running in hassio.

kellerza, 2020-01-15:
@dweston what URL do you use to access sunny portal? Local/on the web?

kellerza, 2020-01-15:
You need to have a local web interface for this to work. i.e. http(s)://192.168.0.77

sheevz0r, 2020-01-15:
you know if this is dependent on the webconnect module ?

kellerza, 2020-01-15:
It seems it could be called WebConnect - https://www.sma.de/en/integrated-webconnect.html

My inverter only states it has an integrated Web UI (local web server), but some datasheets mentions it supports the WebConnect protocol

kellerza, 2020-01-15:
https://www.sma.de/en/integrated-webconnect.html

dweston, 2020-01-15:
> You need to have a local web interface for this to work. i.e. http(s)://192.168.0.77

I'm only able to use a WAN web interface. If I poll the internal LAN IP address for the Inverter I get a time out. I presume SBFspot works as it emulates 

I've checked out the "integrated webconnect" link you provided and note that my Inverter should have this feature capability, though I note the caveat that this capability only "applies as of production date October 1, 2014". Maybe my Inverter was manufactured pre-dating that.

Later today I'll give SMA Australia support a call and check out if its a firmware upgrade issue or there is any way around that production cut-off.

dweston, 2020-01-16:
I've made contact with SMA. The long and the short of it is that for inverters prior to the current model, which is fitted with WiFi, the only way to communicate with the inverter was via special Windows application called `Sunny Explorer`. However post October 2014, if the inverter is fitted with an Ethernet card it became possible to Web connect to the inverter via their online `Sunny Portal` service.

Hence it looks like the inverter version I have does not have an inbuilt Web interface.

I wonder whether the SBFspot code could be ported to enable Home Assistant access?

kellerza, 2020-01-16:
Thanks for the update. If SBFspot is working for you, my recommendation would be to keep using that, but publish it's readings via MQTT to Home Assistant - see [here](https://github.com/SBFspot/SBFspot/wiki/MQTT). You don't need to store anything in SBFspot, it should simply read Speedwire and publish it on MQTT. HomeAssistant has several methods to read MQTT values. 

You could start a discussion in the Home assistant forums to work through this idea as well... even tag me there and I can follow it

closed 2020-01-16 by kellerza

dweston, 2020-01-17:
OK - will do. Unfortunately I haven't managed to get the mqtt version of SBFspot to work as yet. I'm missing something in the local build. That's actually why I was opting to try the direct HA method.