kellerza/pysma #39: No data retrieved from sma

Tokn59, 2020-09-26:
Hi, thank you for sharing this with us !

However I cannot get it to work. I have tried several settings and options provided by other users but still get "Unknown" as values from sma. I have the correct user password and IP address of the sma and I can connect to it using Sunnyboy Explorer.

My configuration.yaml (one of the versions I tried): 

- platform: sma
  host: 192.168.1.118
  password: !secret smaPsw
#  ssl: true
#  verify_ssl: false
#  group: user
#  scan_interval: 120
#  sensors:
#    - current_consumption
#    - total_consumption
#    - pv_power
#    - total_yield 
#    - status
#    - pv_gen_meter

No matter what options here I comment out or not .. always get the same "Unknown" values.

The HA log shows :

Logger: pysma
Source: /usr/local/lib/python3.8/site-packages/pysma/__init__.py:234
First occurred: 17:03:08 (2 occurrences)
Last logged: 17:03:13

Could not start session, Could not connect to SMA at http://192.168.1.118 (timeout), got {}

Anything I should do/change or check ?

Big thanks for your help !

Tony

Tokn59, 2020-09-26:
btw .. I see the configuration.yaml copy/paste is messed up in the above message. It is OK in the real file. Screenshot : 

[image]

Remigiusz2020, 2020-09-27:
> Hi, thank you for sharing this with us !
>
> However I cannot get it to work. I have tried several settings and options provided by other users but still get "Unknown" as values from sma. I have the [...]

HI.......

same problem

kellerza, 2020-09-29:
Hi @Tokn59 

This line indicates that pysma could not connect to the inverter for some reason:

```
Could not start session, Could not connect to SMA at http://192.168.1.118 (timeout), got {}
```
especially the **``` (timeout), got {}```**

Haven't used Sunny Explorer (believe this uses bluetooth), but might support wifi/Ethernet

Possible issues:
1. Inverter does not support the Webconnect feature
1. Incorrect IP, incorrect http/https `ssl:true` or `ssl:false`. Forgot `verify_ssl: false` with `ssl: true`
1. Multiple sessions (@Remigiusz2020 's issue)

Best way to solve this is to connect with your laptop/browser (rules out # 1).
If this does not work (you should get a login page) your inverter might not support Webconnect and this will not work for you

Once you connect you should be able to fine-tune the parameters for # 2.

If you still get the error you might have multiples sessions to your inverter (# 3). Webconnect supports only one. Disconnect your browser and try again

Tokn59, 2020-12-01:
You need to use the modbus integration and allow TCP and UDP ports open on the SMA.

kellerza, 2020-12-02:
@ramoneich you do get a reply from the inverter, but your sensor values all have that '7' in the result,- waht invertr is this?

So looking at the reply, you cannot use the default sensors, only custom ones. You can create your own custom sensors.. see [here](https://github.com/kellerza/pysma/blob/master/pysma/__init__.py#L108) for some guidance on the default sensors and the ID's you will likely have to recreate...

This is the reply:
```
{'6100_40465300': {'7': [{'val': 0}]}, 
'6100_00465700': {'7': [{'val': 4997}]}, 
```

This custom sensor will give you a frequency sensor with value 49.97 Hz

```yaml
  custom:
      frequency:
        key: 6100_00465700
        unit: Hz
        factor: 100
        path: '"7"[0].val'
```

The important part here is the "path": `"7"[0].val`
- `"7"`: look in key 7
- `[0]` first element in the array
- `.val` the val key

See jmespath.org for more examples on this syntax

Good luck & please post your result here if successful

ramoneich, 2020-12-02:
Success! Thank you for the good explanation. I have done this for a friend who swear that it was working before?! I have assumed that it has to do with a firmware-upgrade on sma side. I will collect the information and share it with (all of) you! As soon as we have our code working I will share it as well! Thanks again for showing me the right way!

ramoneich, 2020-12-05:
All-right - here are my infos:
The problem was I was only given the IP Address of the **Storage** device (SBS3.7-10). That is where the values come from. 
As soon as I have entered the IP Adress of the **Inverter** (SUNNY TRIPOWER 6.0) in the configuration.yaml it gives me almost all the data listed in [HOME ASSISTANT Integration documentation](https://www.home-assistant.io/integrations/sma/). 

[image]

As I am also interested in the battery information (so that I know when it is time to charge the car or start the dryer) I used your code (and some code I found in the forum) in order to get the other information from the SBS3.7-10:

`sensor:
  - platform: sma
    host: 192.xx.xx.xx
    ssl: true
    verify_ssl: false
    group: installer
    password: !secret sma_storage
    sensors:
      sma_si_battery_voltage:
      sma_si_battery_charge:
      sma_si_battery_activity:
      sma_si_battery_cycles:
      sma_si_total_yield:
      sma_si_grid_frequency:
      sma_si_grid_voltage:
      sma_si_current_consumption:
      sma_si_current_grid_consumption:
      sma_si_current_pv_output:
      sma_si_total_pv_energy:
      sma_si_daily_pv_energy:
      sma_si_daily_total_energy:
    custom:
      sma_si_battery_voltage:
        key: "6100_00495C00"
        unit: V
        factor: 100
        path: '"7"[0].val'
      sma_si_battery_charge:
        key: "6100_00295A00"
        unit: "%"
        factor: 1
        path: '"7"[0].val'
      sma_si_battery_activity:
        key: "6180_08495E00"
        unit: ""
        # path: '"tag"[0].val' NOT WORKING
        path: '"7"[1].val'
      sma_si_battery_cycles:
        key: "6402_00497900"
        unit: "#"
        factor: 1
        path: '"7"[0].val'
      sma_si_total_yield:
        key: "6400_00462400"
        unit: Wh
        factor: 1
        path: '"7"[0].val'
      sma_si_grid_frequency:
        key: "6100_00465700"
        unit: Hz
        factor: 100
        path: '"7"[0].val'
      sma_si_grid_voltage:
        key: "6100_00464800"
        unit: V
        factor: 100
        path: '"7"[0].val'
      sma_si_current_consumption:
        key: "6100_40463900"
        unit: W
        factor: 1
        path: '"7"[0].val'
      sma_si_current_grid_consumption:
        key: "6100_40463700"
        unit: W
        factor: 1
        path: '"7"[0].val'
      sma_si_current_pv_output:
        key: "6100_00452200"
        unit: W
        factor: 1
        path: '"7"[0].val'
      sma_si_total_pv_energy:
        key: "6400_00454B00"
        unit: Wh
        factor: 1
        path: '"7"[0].val'
      sma_si_daily_pv_energy:
        key: "6400_00454C00"
        unit: Wh
        factor: 1  
        path: '"7"[0].val'
      sma_si_daily_total_energy:
        key: "6400_00461F00"
        unit: kWh
        factor: 1000
        path: '"7"[0].val'
`

The output looks like that:
[image]

And the Logg is telling me the following:
`Logger: pysma
Source: /usr/local/lib/python3.8/site-packages/pysma/__init__.py:296 
First occurred: 15:32:03 (426 occurrences) 
Last logged: 16:07:28

No values for sensors: sma_si_daily_pv_energy [6400_00454C00],sma_si_total_pv_energy [6400_00454B00],sma_si_battery_cycles [6402_00497900],sma_si_current_consumption [6100_40463900],sma_si_current_pv_output [6100_00452200]. Response from inverter: {'6100_00465700': {'7': [{'val': 5003}]}, '6180_08495E00': {'7': [{'val': [{'tag': 2291}]}]}, '6400_00461F00': {'7': [{'val': 2028039}]}, '6100_00464800': {'7': [{'val': 22249}]}, '6100_00295A00': {'7': [{'val': 0}]}, '6100_40463700': {'7': [{'val': 1561}]}, '6100_00495C00': {'7': [{'val': 19}, {'val': None}, {'val': None}]}, '6400_00462400': {'7': [{'val': 3471007}]}}
No values for sensors: sma_si_daily_pv_energy [6400_00454C00],sma_si_total_pv_energy [6400_00454B00],sma_si_battery_cycles [6402_00497900],sma_si_current_consumption [6100_40463900],sma_si_current_pv_output [6100_00452200]. Response from inverter: {'6100_00465700': {'7': [{'val': 5004}]}, '6180_08495E00': {'7': [{'val': [{'tag': 2291}]}]}, '6400_00461F00': {'7': [{'val': 2028039}]}, '6100_00464800': {'7': [{'val': 22298}]}, '6100_00295A00': {'7': [{'val': 0}]}, '6100_40463700': {'7': [{'val': 1557}]}, '6100_00495C00': {'7': [{'val': 19}, {'val': None}, {'val': None}]}, '6400_00462400': {'7': [{'val': 3471007}]}}`

My apologies for not knowing that there are so many different sma devices in the network. My friend explained me in detail that there are in total at least three different devices that have a web interface accessible: Inverter (SUNNY TRIPOWER 6.0), Storage(SBS3.7-10), SMA ENERGY METER

If you need more information I will be happy to share them with you

Thanks for the help @kellerza

hunkyn, 2020-12-08:
I am having the same issue. I do not have access to installer group but I setup using user group and I do get some values but most of them are returning as "unknown"

kellerza, 2021-01-07:
@ramoneich thanks for sharing your setup, hopefully other find it useful!

@hunkyn if you get some values then there is not much we can do. You can try to decipher more keys, but my suggetsion is just to get the total kWh from the grid & utility and then use the HASS utility sensor on them to get daily and monthly averages - there is an example on the main Readme.md of this repo. You will see even @ramoneich has a bunch of "Unknown" sensors. Some inverters support more than others

closed 2021-01-07 by kellerza