kellerza/pysma #45: total_consumption sensor creates error in home assistant: Error doing job: Task exception was never retrieved

quizzical86, 2020-12-27:
Home assistant logs are showing an error for the total_consumption sensor:
`Logger: homeassistant
Source: helpers/entity.py:289
First occurred: 22:25:48 (1 occurrences)
Last logged: 22:25:48

Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/sma/sensor.py", line 167, in async_sma
    sensor.async_update_values()
  File "/usr/src/homeassistant/homeassistant/components/sma/sensor.py", line 225, in async_update_values
    self.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 289, in async_write_ha_state
    raise RuntimeError(f"Attribute hass is None for {self}")
RuntimeError: Attribute hass is None for <Entity total_consumption: 0.0>`
It is a sensor that I'm not using, so not a major issue for me, but logs without errors are allways nicer :)

kellerza, 2021-01-07:
Strange, this is the first time that is see this error for a single sensor `RuntimeError: Attribute hass is None for <Entity total_consumption: 0.0>` and always though it was for all sensors.

This might mean that this sensor is not supported and initialized incorrectly in the setup phase

What might work for you is to simply exclude this sensor (That might mean including everything except this one)

quizzical86, 2021-01-08:
Dear @kellerza ,
Untill now I did not specify any sensors in the configuration.yaml file and just took in all the sensors.
Based on your suggestion I have now instead listed the sensors that I need so that I no longer have the total_consumption sensor.
This has solved (or rather hidden) the issue and the error is no longer appearing in the log.

I leave it for you to judge whether this 'closes' the issue or not :)

Thanks for your help (and for the integration with SMA devices!)!

kellerza, 2021-05-06:
I guess this can be closed, @rklomp recently added UI configuration for SMA and with this (I hope) we fixed this bug,- Unfortunately I was never able to replicate it on my setup

If it still persists even with config entries, maybe open a fresh issue here?

closed 2021-05-06 by kellerza

quizzical86, 2021-05-06:
Thank you @kellerza, the integration did not work as I got the following error:
ValueError: Unique id '1930064096-6380_40251E00_0' is already in use by 'sensor.pv_voltage
However removing the integration and adding it again solved that issue.
So fine to close this issue.
Many thanks!