kellerza/pysma #54: Key for "current_consumption" and "total_consumption" not working

cpohli, 2021-03-25:
Sorry, if this is not the right place for this, but i don't know, where to leave it.

I set up the SMA in HA and get all the standard readings correct, except for the
total_consumption and current consumption - sensors. These are giving me a
`No values for sensors: total_consumption [6400_00543A00],current_consumption [6100_00543100]`
in the logs. The remaining sensors provided are working as expected.

I dove into my converter and found the sensor to be
6400_00462500 - for total_consumption.
The current_consumption i am not able to find in my web-interface, but would very much like to provide, if given directions what to look for.

I also tried to set these up as custom sensors, but i have to admit, that i'm not clever enough to figure out, where exactly to put the custom: in my config.yaml :-( The SMA help page in HA states the how, but not very clearly the where...

Thanks and sorry for bothering you.

cpohli, 2021-03-26:
For what it's worth, i just found that the values given by current_consumtion are off (at least in my setup) anyway.
I constantly get a much lower reading than the data delivered by the HomeMager 2.0.
So i assume these readings are not take all consumption into account. Or i am just beeing wrong here what to expect from this.

kellerza, 2021-03-27:
Hi, the key you suggest for `total_consumption` seems to be `grid_total_absorbed` - see [here](https://github.com/kellerza/pysma/blob/master/pysma/__init__.py#L136).

Maybe you can try `grid_total_yield` as well and let us know what you find

cpohli, 2021-03-29:
hmm...
i tried your` grid_total_yield` (that's 6400_00462400) and that's giving me the same reading as 6400_00469100.
So that would be like a grid_total_feed?

I'm sorry for the missing naming, but i can only see the numbers of the keys in the local "webpage" of the inverter.

If i can be of any further assistance, i will provide anything needed (and poosible for me :-)).

closed 2021-07-07 by rklomp