kellerza/pysma #56: Too many sessions?

cpohli, 2021-04-09:
Hi,

i sucessfully used the SMA Integration for some weeks now and greatly appreciate all your efforts on this.

Today my sensors went away and i got the following error in my logs:

`Logger: pysma
Source: /usr/local/lib/python3.8/site-packages/pysma/__init__.py:232
First occurred: 05:50:42 (4 occurrences)
Last logged: 05:50:56

Could not start session, Max amount of sessions reached, got {}
`

I can login into the local webUI with no problems and see, that the inverter is up and running.
Has anyone any thoughts on this?

Thank you very much.

closed 2021-04-09 by cpohli

kellerza, 2021-04-09:
SMA only allows a single login, so if you logged in via the webpage/your Home Assistant restarted without closing the session to SMA it might take a bit for the session to timeout

cpohli, 2021-04-09:
Hmm... On my Tripower i have an constantly open webpage via UI (locally) **AND** the working SMA integration.
Does this count as one? It's working fine.
I suspect i'm getting this error for my second SMA integration with my Sunny Island (trying to get the batterie-% from there, as the Tripower doesn't deliver this).

cpohli, 2021-04-09:
Oh, thanks @kellerza !! :-)

You brought me some delight today:
I have two SMA devices (Tripower for solar and Sunny Island for battery).
Using your SMA integration for the Tripower worked fine. The SI never worked (i suspected 2 sma integrations were not possible).

Now i've done the folowing:
- two integrations
- logged out of both webUIs
- added `path: '"7"[0].val'` to the custom sensors for the SI
- restarted HA - sensors are working on both devices! :-)
- logged in into both webUIs - working as well and in parallel to the sensors
- adds up to sensational!