kellerza/pysma #58: Battery inverter data: some working keys & issue

fitool, 2021-05-05:
Hello @kellerza ,

thanks for your work on this! I'm using the HA integration to fetch data from a Sunny Boy 5.0 . I tried the same for a Sunny Boy Storage 3.7 but it seems the data is not read in correctly, I think it might have something to do with the devices class ("1" for solar inverter and "7" for battery inverter). In the feedback from the logs below, for the battery-specific parameters, 3 values are shown because the inverter has possibility to connect up to 3 batteries (only 1 connected in this case). I assume 180 = 18°C, 15921 = 159.21 V and 1569 = 1.569 A, which fits with AC power of 233 W (DC-AC conversion efficiency of 93%)

- battery_soc `response from inverter: '6100_00295A00': {'7': [{'val': 87}]}`
- battery_temp `response from inverter: '6100_40495B00': {'7': [{'val': 180}, {'val': None}, {'val': None}]}`
- battery_voltage:  `response from inverter: '6100_00495C00': {'7': [{'val': 15921}, {'val': None}, {'val': None}]}` 
- battery_current:  `response from inverter: '6100_40495D00': {'7': [{'val': 1569}, {'val': None}, {'val': None}]}`
- battery_inverter_power:  `response from inverter: '6100_40263F00': {'7': [{'val': 233}]}}`

Would it be possible to adapt the code to be compatible with battery inverters as well? 

Thx
Kr,

fitool

kellerza, 2021-05-06:
Have you tried custom sensors? They can all use the same key. `"7"[0].val`, which means: The 7 key, then the first value in the array, then the val key

```yaml
custom:
  battery_soc:
   key: 6100_00295A00
   unit: '%"
   path: '"7"[0].val'
```

@rklomp started adding a more generic approach in #57 maybe we should add this "7" if these are always batteries?

Or we should just include all battery values (they are a bit different than solar inverters after all)?

@fitool can you see if you can add these battery sensors as default sensors? - hint, look [here](https://github.com/kellerza/pysma/blob/master/pysma/__init__.py#L162) the parameters for Sensor are key, name, unit, multiplier, path.... use the path from the yaml above, so example for SOC

```python
Sensor("6100_00295A00", "battery_soc", "%", 1, '"7"[0].val')
```

rklomp, 2021-05-06:
Have you every seen multiple values at the same time in that path? Maybe we could do a wildcard `'"*"[0].val'`

kellerza, 2021-05-06:
Never seen multiple values. I like the idea and made a suggestion on your PR

rklomp, 2021-05-06:
@fitool 

I am building a mapping between device class and sensors.
Do you have any other sensors available besides these battery sensors?

Would you be able to provide me some extra information?

Login to the webinterface of your device and go to the "Home" page
Open developer tools, go to network tab. Click clear in the top left and make sure recording is on. Then reload the page (F5).

You should now see already a lot of requests.

While keeping the recording running:
Go to page "Instantaneous values" (a request to getAllOnlValues.json should appear)
Go to page "Device parameters" (a request to getAllParamValues.json should appear)

Save these requests as a HAR file and share that file.

This document might help:
https://support.zendesk.com/hc/en-us/articles/204410413-Generating-a-HAR-file-for-troubleshooting

fitool, 2021-05-06:
@kellerza yes, I tried with custom sensor; but I didn't know that 'path' was a possible parameter, adding it solves the issue, thanks!

@rklomp aha, this is the way to get keys :-) nothing personal but I prefer not to share a full dump in case any sensitive information would be included; I can do the check for all relevant parameters; I already started with the ones below for the battery inverter SBS3.7. For  SB5.0, pysma already gives most of the relevant info, I just added two custom sensors for the power per string input on the inverter. Which additional information are you looking for? I found a lot of info in this file in SBFspot repository (https://github.com/SBFspot/SBFspot/blob/master/SBFspot/TagListEN-US.txt) device classes are present there as well starting line 4998. Also, if you take last 6 characters from a key and convert from hex to decimal, you can find a description using this file:
e.g. 6180_08214800 --> 214800 (hex) = 2181120 (dec) --> line 5814 in txt file ("10110=Operation.Health\2181120\Condition")

**SB5.0**
    custom:
      pv_power_string_a:
         key: "6380_40251E00_0"
         unit: W
      pv_power_string_b:
         key: "6380_40251E00_1"
         unit: W

**SBS3.7**
Raw feedback via web browser page of inverter:
{"6400_00496700":{"7":[{"val":56269}]},"6400_00496800":{"7":[{"val":36854}]},"6100_00295A00":{"7":[{"val":82}]},"6180_08495E00":{"7":[{"val":[{"tag":2293}]}]},"6100_00496900":{"7":[{"val":0}]},"6100_00496A00":{"7":[{"val":267}]},"6100_00696E00":{"7":[{"val":100}]},"6100_40263F00":{"7":[{"val":267}]},"6800_00832A00":{"7":[{"low":3680,"high":3680,"val":3680}]},"6180_08214800":{"7":[{"val":[{"tag":307}]}]},"6180_08414900":{"7":[{"val":[{"tag":886}]}]},"6400_00462500":{"7":[{"val":41880}]},"6400_00462400":{"7":[{"val":541940}]},"6100_40463700":{"7":[{"val":4}]},"6100_40463600":{"7":[{"val":0}]},"6800_008AA200":{"7":[{"low":0,"high":null,"val":3009876189}]},"6400_00260100":{"7":[{"val":36868}]},"6800_08811F00":{"7":[{"validVals":[1129,1130],"val":[{"tag":1129}]}]},"6400_00462E00":{"7":[{"val":1163122}]}}

Corresponding custom sensors (part 1):

    custom:
      battery_soc: 
         key: "6100_00295A00"
         unit: '%'
         path: '"7"[0].val'
      battery_temp: 
         key: "6100_40495B00"
         unit: '°C'
         factor: 10
         path: '"7"[0].val'
      battery_voltage: 
         key: "6100_00495C00"
         unit: 'V'
         factor: 100
         path: '"7"[0].val'
      battery_current: 
         key: "6100_40495D00"
         unit: 'A'
         factor: 1000
         path: '"7"[0].val'
      battery_inverter_power: 
         key: "6100_40263F00"
         unit: 'W'
         path: '"7"[0].val'
      battery_energy_charge_total:
         key: "6400_00496700"
         unit: 'kWh'
         factor: 1000
         path: '"7"[0].val'
      battery_energy_discharge_total:
         key: "6400_00496800"
         unit: 'kWh'
         factor: 1000
         path: '"7"[0].val'
      battery_operating_status:  # gives back a code: 2292 = Charge battery, 2293 = Discharge battery
         key: "6180_08495E00"
         path: '"7"[0].val[0].tag'
         unit: ''
      battery_power_charging: 
         key: "6100_00496900"
         unit: 'W'
         path: '"7"[0].val'
      battery_power_discharging: 
         key: "6100_00496A00"
         unit: 'W'
         path: '"7"[0].val'
      battery_capacity: 
         key: "6100_00696E00"
         unit: '%'
         path: '"7"[0].val'
      battery_inverter_power_limit: 
         key: "6800_00832A00"
         unit: 'W'
         path: '"7"[0].val'
      battery_condition:  # gives back a code: 307 = Ok
         key: "6180_08214800"
         path: '"7"[0].val[0].tag'
         unit: ''

fitool, 2021-05-06:
> @fitool can you see if you can add these battery sensors as default sensors? - hint, look [here](https://github.com/kellerza/pysma/blob/master/pysma/__init__.py#L162) the parameters for Sensor are key, name, unit, multiplier, path.... use the path from the yaml above, so example for SOC

honestly I don't have much coding experience so I wouldn't know where to start...
however, I see that some keys are used both by solar inverters (device class 1) and battery inverters (device class 7), for example 6100_40263F00,  so probably this needs to be taken into account when adding it as a default sensor ?

rklomp, 2021-05-07:
@fitool 
How does this work? Does the inverter have a seperate web interface or is that incorporated into the Sunny Boy webinterface and are you getting values from te inverter in both devices class 1 and 7?

Your custom sensors `pv_power_string_a` and `pv_power_string_b` are included in pysma since the latest update and called `pv_power_a` and `pv_power_b`

fitool, 2021-05-08:
@rklomp 
The battery storage has a separate inverter (Sunny Boy Storage) with its own IP address and web interface. Some keys are used by both devices and give a different value (e.g. AC power production of the inverters), while others give the same value (e.g. data from energy meter / Sunny Home Manager 2.0 installed at grid connection point). I use Sunny Home Manager 2.0 at the grid connection point, it has an IP address as well, but I can't access a web interface, I guess it's data is only accessible via the inverters
         
```
# Data from web interface battery inverter: Sunny Boy Storage 3.7:
"6100_40263F00":{"7":[{"val":140}]} # Battery inverter power 
"6100_40463700":{"7":[{"val":9}]}   # Power at grid connection point (data from Sunny Home Manager 2.0)
# Data from web interface PV inverter Sunny Boy 5.0:
"6100_40263F00":{"1":[{"val":944}]} # Solar inverter power 
"6100_40463700":{"1":[{"val":9}]}   # Power at grid connection point (data from Sunny Home Manager 2.0)
```

rklomp, 2021-05-08:
In my `dev` branch I have started working on a mapping between different device classes and applicable sensors
https://github.com/rklomp/pysma/blob/dev/pysma/const.py#L300-L312

Could you have a look and complete the list of sensors required for `DEVCLASS_BATTERY`?

fitool, 2021-05-08:
@rklomp OK, I'm checking and testing the relevant sensors for battery inverter and will update above file afterwards; indeed power absorbed or injected by battery should be included for sure; there are several keys for this, also a key which indicates whether the battery is in charge or discharge mode; i keep you informed!

rklomp, 2021-05-08:
Thanks! For sensors like `battery_condition` and `battery_operating_status` you can set `SL10N: True`. It will than try to translate the returned code to a string using the language file of the device located at /data/l10n/en-US.json

rklomp, 2021-05-08:
This might be a useful list as well: https://github.com/rklomp/pysma/wiki/Sensor-Overview

rklomp, 2021-05-09:
I have also updated the example script to easily test this. It should now retrieve the sensor list from `DEVICE_MAP` based on the device class.

https://github.com/rklomp/pysma/blob/dev/example.py

fitool, 2021-05-09:
@rklomp I edited the const.py file via and sent you the proposed modification via pull request; hope this is the right way to do it :-)

closed 2021-07-07 by rklomp

rklomp, 2022-09-22:
@fitool 

I am creating a [file with test data](https://github.com/rklomp/pysma/blob/dynamic-sensors/tests/__init__.py#L24) for testing the code.
Would you be able to share the response of your inverter for POST requests to `/dyn/getAllOnlValues.json` and `/dyn/getAllParamValues.json`?

You can get those via the developer console of your browser when you visit the Instantaneous values and Device parameters pages. The `getAllParamValues.json` includes WiFi network names, MAC address and IP addresses if you want you can change these to something else. But please keep it as valid data ;)

If you need any help let me know. Thanks in advance!

fitool, 2022-10-08:
@rklomp 

please find the requested info (for Sunny Boy Storage 3.7) below. I replaced characters of serial codes by '9' and MAC addresses by 'A'
[SMA_SB37_getAllOnValues.txt](https://github.com/kellerza/pysma/files/9739318/SMA_SB37_getAllOnValues.txt)
[SMA_SB37_getAllParamValues.txt](https://github.com/kellerza/pysma/files/9739321/SMA_SB37_getAllParamValues.txt)

don't hesitate to contact me if you need any other information & my apologies for the delay!