kellerza/pysma #59: Incoming W per Phase from grid

clasheleander, 2021-05-06:
My custom sensors regarding this in homeassistant no longer work since update as I suspect it is not supported to run this custom sensors anymore? Can these sensors be added? Or is there any possibility to still run custom sensors?
Iam using below sensors to monitor current incoming W on each Phase from grid.
Sensors are from SUNNY TRIPOWER 10.0 (STP10.0-3AV-40 010) with SMA energy meter installed in the house.

      incoming_l1:
          key: '6100_0046EB00'
          unit: W
       incoming_l2:
          key: '6100_0046EC00'
          unit: W
       incoming_l3:
          key: '6100_0046ED00'
          unit: W

Thanks.

kellerza, 2021-05-06:
Custom sensors should still work, see the homeassistant [docs](https://github.com/kellerza/pysma/blob/master/pysma/__init__.py#L138)

Ideally you should add these to pysma [here](https://github.com/kellerza/pysma/blob/master/pysma/__init__.py#L138) maybe as power_l1/power_l2/power_l3?

clasheleander, 2021-05-06:
Thanks for your reply!
Sorry if I ask but am new to this.
I have put a pull request on changes, is this the right way?
Can I test this in any way? as mentioned earlier, these no longer work as custom sensors, this broke during the last update. no custom sensors are loaded for the SMA integration from configuration.yaml
Warning in logfile, "Loading SMA via platform setup is deprecated. Please remove it from your configuration"

Thanks!

rklomp, 2021-05-07:
@clasheleander While deprecated, customs sensors using yaml should still work. Could you post your full yaml for SMA so I could test this?

clasheleander, 2021-05-07:
@rklomp 

Thanks!

Stopped to work after update to 2021.5

Here are yaml, (everything was uncomment before update and working)

sensor:
  - platform: sma
    host: 192.168.1.89
    ssl: true
    verify_ssl: false
    password: xxx
    group: installer
    sensors:
#      - pv_power
#      - pv_current
#      - pv_voltage
#      - pv_gen_meter
#      - total_yield
#      - daily_yield
#      - status
#      - current_l1
#      - current_l2
#      - current_l3
#      - grid_power
#      - grid_power_absorbed
#      - grid_power_supplied
    custom:
#       dc_current:
#          key: '6380_40251E00'
#          unit: W
#       dc_currentb:
#          key: '6380_40251E00_1'
#          unit: W
#       pv_currentb:
#          key: '6380_40452100_1'
#          unit: A
#          factor: 1000
#       pv_voltageb:
#          key: '6380_40451F00_1'
#          unit: V
#          factor: 100
#       active_power_l1:
#          key: '6100_40464000'
#          unit: W
#       active_power_l2:
#          key: '6100_40464100'
#          unit: W
#       active_power_l3:
#          key: '6100_40464200'
#          unit: W
       inkommande_l1:
          key: '6100_0046EB00'
          unit: W
       inkommande_l2:
          key: '6100_0046EC00'
          unit: W
       inkommande_l3:
          key: '6100_0046ED00'
          unit: W

clasheleander, 2021-05-08:
@rklomp 
SMA integration is installed using the gui. Can this be the cause?
After meter are in installed the meter and inverter auto connects and start feeding readings to inverter and can then be seen in the inverter.
The energy meter also have its own gui very similar as the inverter

rklomp, 2021-05-08:
Ah that would explain. The custom custom sensor parameters are only loaded when initially configured via yaml.

clasheleander, 2021-05-08:
[image]
[image]
[image]

rklomp, 2021-05-08:
Thanks. Would you be able to provide an export of requests to `getAllOnlValues.json` and `getAllParamValues.json`? That would help locating the correct keys to query to determine if the energy meter is connected.

Login to the webinterface of your device and go to the "Home" page
Open developer tools, go to network tab. Click clear in the top left and make sure recording is on. 

While keeping the recording running:
Go to page "Instantaneous values" (a request to getAllOnlValues.json should appear)
Go to page "Device parameters" (a request to getAllParamValues.json should appear)

Save these requests as a HAR file and share that file.

This document might help:
https://support.zendesk.com/hc/en-us/articles/204410413-Generating-a-HAR-file-for-troubleshooting

I am trying to create some logic to only load the Energy Meter sensors if the energy meter is actually connected, instead of having those sensors loaded for all users.
So if I understand correctly you would also be able to add the energy meter as a separate device to HA? Could you create the same export of `getAllOnlValues.json` and `getAllParamValues.json` on the the energy meter GUI?

clasheleander, 2021-05-08:
@rklomp 

Of cource!
Will do this asap. Childrens party here at the moment 😀.

clasheleander, 2021-05-08:
What is the correct way to install this manualy? Uninstall from gui, Download to custom_components and start using yaml again with new sensorsnames from pysma file?

Thanks.

clasheleander, 2021-05-08:
@rklomp 

Childrens party is over. :)

Attached.
[tripower_10_instantaneous_values.har.txt](https://github.com/kellerza/pysma/files/6445915/tripower_10_instantaneous_values.har.txt)
[tripower_10_device_parameters.har.txt](https://github.com/kellerza/pysma/files/6445916/tripower_10_device_parameters.har.txt)
[sma_energy_meter_instantaneous_values.har.txt](https://github.com/kellerza/pysma/files/6445917/sma_energy_meter_instantaneous_values.har.txt)
[sma_energy_meter_device_parameters.har.txt](https://github.com/kellerza/pysma/files/6445918/sma_energy_meter_device_parameters.har.txt)

clasheleander, 2021-05-08:
I solved the problem with custom sensors!

Anyway, It would have been good if the sensors were included in the core so you do not have to keep track of new functionality manually. :)

Thanks!

rklomp, 2021-05-08:
> I solved the problem with custom sensors!

Good to hear! It will only work if initially loaded from yaml. If configured from the GUI initially any new yaml will be ignored.

> @rklomp
>
> Childrens party is over. :)
>
> Attached.
> [tripower_10_instantaneous_values.har.txt](https://github.com/kellerza/pysma/files/6445915/tripower_10_instantaneous_values.har.txt) [...]

Thanks. I found the serial number in  `Energy_Meter_Add: 6800_008AA200` and `Energy_Meter: 6800_008AA300`. No idea what the difference is between these two.
```
6800_008AA200: {1: [{low: 0, high: null, val: 1901427721}]}
6800_008AA300: {1: [{low: 0, high: null, val: 1901427721}]}
```

In my sunnyboy this will return
```
6800_008AA200: {1: [{low: 0, high: null, val: 0}]}
6800_008AA300: {1: [{low: 0, high: null, val: null}]}
```

We have two options:
1. Add both devices to HA and let the Tripower return values specific to solar panels only and Energy Meter only values specific to energy meter
2. Add only the Tripower to HA en detect if Energy Meter is linked and if so load both the solar and the energy meter sensors.

What do you think? Which one would be best? What would you prefer?

rklomp, 2021-05-08:
@itinside. You are also using energy meter. Do you have an inverter as well?

clasheleander, 2021-05-08:
hmm.. both would work well, see pros and cons with both options. there is not much for readings in energy meter self so this alone does not give me much personally, would be in installations where you only have a energy meter without an inverter. 
[image]

inverter uses these metrics for a lot of improved graphs and info. perhaps option 2 is easiest and best?
While Option 1 provides readings for installations without inverter.
Option 3 is to do both. Not many readings in energy meter to maintain. :)

[image]
[image]
[image]

clasheleander, 2021-05-08:
on a second thought, the readings for Phase currents (Ampere per phase) is not listed in the inverter i think and could be useful to read in HA.
Inverter only shows how much inverter provides in A per phase, not at grid connection point. :)

kellerza, 2021-05-08:
The energymeter by itself will not work with `pysma` - it sends multicast packets that an inverter etc and does not have a Webconnect interface itself.

Even if you use Webconnect you can get energy meter readings in HomeAssistant using this add-on - https://github.com/kellerza/hassio-sma-em

rklomp, 2021-05-08:
Looking at what @itinside and @clasheleander provide it seems it does have a webinterface and if queried directly returns values using device class 65.

clasheleander, 2021-05-08:
@kellerza 

Nice! Will try this! 👍

clasheleander, 2021-05-08:
Looks like the meter have its own webservice. I can access meter the same way i access inverter. From a browser.

kellerza, 2021-05-08:
I can also access mine through the web interface, but it is only for setup and no way to get any readings (firmware 1.02.04.R)

If I try to access any of the known URLs (without a SID) I get a 404 and redirected to the setup

kellerza, 2021-05-08:
[image]

clasheleander, 2021-05-08:
This is me logged in to meter ip from browser:
[image]

Firmware 2.0.18.R

kellerza, 2021-05-08:
Maybe I should try an update!

clasheleander, 2021-05-08:
I think there are 2 versions of meter? Mine is called "emeter-20"
Installed a year ago. Do you have an older one maybe?

clasheleander, 2021-05-08:
[image]

kellerza, 2021-05-08:
Seems I have the emeter-10, but from the website unclear if this is just new firmware or some new hardware https://www.sma.de/en/products/monitoring-control/sma-energy-meter.html#tab-content-243745-2
I've mailed their support & asked

rklomp, 2021-05-08:
In my `dev` branch I have started working on a mapping between different device classes and applicable sensors
https://github.com/rklomp/pysma/blob/dev/pysma/const.py#L269

@clasheleander, Looking at your har files it looks like the keys from the Energy Meter are different from the Keys of the same data received via the Inverter.

Could you have a look and compose a list of sensors you would like to get from either the inverter or the energy meter directly.
I think the `SENSOR_GRID_ACTIVE_POWER_CONSUMED_L1`,  `_L2` and  `_L3` added by you are only applicable to installations that have a Energy Meter?

My idea is that if you connect to an Energy Meter directly you will get all sensors listed under `DEVCLASS_ENERGY_METER`. If you connect to an inverter without Energy Meter attached you will get `DEVCLASS_INVERTER` and if the energy meter is attached you will get `DEVCLASS_INVERTER` + `ENERGY_METER_VIA_INVERTER`.

Maybe some other sensors from the `DEVCLASS_INVERTER` list could be moved to  `ENERGY_METER_VIA_INVERTER` as well, as I get a few that do not return any value. For example: `SENSOR_GRID_POWER_SUPPLIED`, `SENSOR_GRID_POWER_ABSORBED` , `SENSOR_GRID_TOTAL_YIELD` and `SENSOR_GRID_TOTAL_ABSORBED`

rklomp, 2021-05-09:
I have also updated the example script to easily test this. It should now retrieve the sensor list from `DEVICE_MAP` based on the device class. Sensors from `ENERGY_METER_VIA_INVERTER` will be added only if a serial number is found under `SENSOR_ENERGY_METER`.

https://github.com/rklomp/pysma/blob/dev/example.py

clasheleander, 2021-05-10:
@rklomp 
Nice work! :) I have tried example.py, works perfecly! Below i have also added a few more sensors (all from sma energy meter that inverter lists)

These two sensors dont give me any readings:
**No values for sensors: current_consumption [6100_00543100],total_consumption [6400_00543A00]** i have disable them in current version in HA (i cant find them in your dev const.py, have you removed them?)

new sensors to be added in const.py:

SENSOR_GRID_PHASE_CURRENTS_L1 = {
    SKEY: "6100_40466500",
    SNAME: "grid_phase_currents_l1",
    SUNIT: "A",
    SFACTOR: 1000,
}
SENSOR_GRID_PHASE_CURRENTS_L2 = {
    SKEY: "6100_40466600",
    SNAME: "grid_phase_currents_l2",
    SUNIT: "A",
    SFACTOR: 1000,
}
SENSOR_GRID_PHASE_CURRENTS_L3 = {
    SKEY: "6100_40466B00",
    SNAME: "grid_phase_currents_l3",
    SUNIT: "A",
    SFACTOR: 1000,
}
SENSOR_GRID_SUPPLIED_POWER = {
    SKEY: "6100_40463600",
    SNAME: "grid_supplied_power",
    SUNIT: "W",
}

    ENERGY_METER_VIA_INVERTER: [
        SENSOR_GRID_ACTIVE_POWER_CONSUMED_L1,
        SENSOR_GRID_ACTIVE_POWER_CONSUMED_L2,
        SENSOR_GRID_ACTIVE_POWER_CONSUMED_L3,
	**SENSOR_GRID_PHASE_CURRENTS_L1,
	SENSOR_GRID_PHASE_CURRENTS_L2,
	SENSOR_GRID_PHASE_CURRENTS_L3,
	SENSOR_GRID_SUPPLIED_POWER,**

Log from example.py

DEBUG:asyncio:Using proactor: IocpProactor
INFO:__main__:NEW SID: xtpqgFBQukfQAFB-
DEBUG:pysma:Found device class 1
DEBUG:pysma:Sensor status: Will be decoded with "1"[0].val[0].tag from {'1': [{'val': [{'tag': 307}]}]}
DEBUG:pysma:Loading sensors for device class 1
DEBUG:pysma:Sensor energy_meter: Will be decoded with "1"[0].val from {'1': [{'low': 0, 'high': None, 'val': 1901427721}]}
DEBUG:pysma:Energy Meter with serial 1901427721 detected. Adding extra sensors.
WARNING:pysma:Duplicate SMA sensor key 6100_40463600 (idx: 0)
DEBUG:pysma:Sensor serial_number: Will be decoded with "1"[0].val from {'1': [{'low': 0, 'high': None, 'val': 3006927010}]}
DEBUG:pysma:Sensor device_name: Will be decoded with "1"[0].val from {'1': [{'val': 'STP10.0-3AV-40 010'}]}
DEBUG:pysma:Sensor device_type: Will be decoded with "1"[0].val[0].tag from {'1': [{'validVals': [9347, 9348], 'val': [{'tag': 9348}]}]}
DEBUG:pysma:Sensor device_manufacturer: Will be decoded with "1"[0].val[0].tag from {'1': [{'validVals': [461], 'val': [{'tag': 461}]}]}
         serial               3006927010
           name       STP10.0-3AV-40 010
           type      SUNNY TRIPOWER 10.0
   manufacturer                      SMA
DEBUG:pysma:Sensor grid_power: Will be decoded with "1"[0].val from {'1': [{'val': 8137}]}
DEBUG:pysma:Sensor power_l1: Will be decoded with "1"[0].val from {'1': [{'val': 2712}]}
DEBUG:pysma:Sensor power_l2: Will be decoded with "1"[0].val from {'1': [{'val': 2702}]}
DEBUG:pysma:Sensor power_l3: Will be decoded with "1"[0].val from {'1': [{'val': 2723}]}
DEBUG:pysma:Sensor frequency: Will be decoded with "1"[0].val from {'1': [{'val': 5003}]}
DEBUG:pysma:Sensor voltage_l1: Will be decoded with "1"[0].val from {'1': [{'val': 24111}]}
DEBUG:pysma:Sensor voltage_l2: Will be decoded with "1"[0].val from {'1': [{'val': 24095}]}
DEBUG:pysma:Sensor voltage_l3: Will be decoded with "1"[0].val from {'1': [{'val': 24228}]}
DEBUG:pysma:Sensor current_l1: Will be decoded with "1"[0].val from {'1': [{'val': 11263}]}
DEBUG:pysma:Sensor current_l2: Will be decoded with "1"[0].val from {'1': [{'val': 11242}]}
DEBUG:pysma:Sensor current_l3: Will be decoded with "1"[0].val from {'1': [{'val': 11269}]}
[... 28 lines cut ...]
               voltage_l3         242.28 V
               current_l1         11.263 A
               current_l2         11.242 A
               current_l3         11.269 A
               pv_power_a           6197 W
               pv_power_b           2249 W
             pv_voltage_a         409.22 V
             pv_voltage_b         383.55 V
             pv_current_a         15.145 A
             pv_current_b          5.864 A
             pv_gen_meter       8640.494 kWh
              total_yield       8624.741 kWh
              daily_yield          15215 Wh
      grid_power_supplied           7544 W
      grid_power_absorbed              0 W
         grid_total_yield       4493.548 kWh
      grid_total_absorbed       6360.166 kWh
                   status             Ok
grid_active_power_consumed_l1              0 W
grid_active_power_consumed_l2              0 W
grid_active_power_consumed_l3              0 W
   grid_phase_currents_l1          9.776 A
   grid_phase_currents_l2         10.687 A
   grid_phase_currents_l3         10.928 A
      grid_supplied_power           7544 W

rklomp, 2021-05-10:
`current_consumption` and `total_consumption` have been removed as it looks like they dont work for anyone.

`SENSOR_GRID_SUPPLIED_POWER` already exists as `SENSOR_GRID_POWER_SUPPLIED`

I think `SENSOR_GRID_POWER_ABSORBED` , `SENSOR_GRID_TOTAL_YIELD` and `SENSOR_GRID_TOTAL_ABSORBED` could be moved to `ENERGY_METER_VIA_INVERTER` as well?

These are not showing any values at my setup
[image]

clasheleander, 2021-05-10:
You are right, SENSOR_GRID_SUPPLIED_POWER are same as SENSOR_GRID_POWER_SUPPLIED.
Would the be happy if you only add these to meter section then:

SENSOR_GRID_PHASE_CURRENTS_L1 = {
SKEY: "6100_40466500",
SNAME: "grid_phase_currents_l1",
SUNIT: "A",
SFACTOR: 1000,
}
SENSOR_GRID_PHASE_CURRENTS_L2 = {
SKEY: "6100_40466600",
SNAME: "grid_phase_currents_l2",
SUNIT: "A",
SFACTOR: 1000,
}
SENSOR_GRID_PHASE_CURRENTS_L3 = {
SKEY: "6100_40466B00",
SNAME: "grid_phase_currents_l3",
SUNIT: "A",
SFACTOR: 1000,
}

I also agree on SENSOR_GRID_POWER_ABSORBED, SENSOR_GRID_TOTAL_YIELD, SENSOR_GRID_TOTAL_ABSORBED. these should be moved to meter aswell.

rklomp, 2021-05-10:
What is the difference between these new sensors and the existing current sensors?
```
SENSOR_CURRENT_L1 = {
    SKEY: "6100_40465300",
    SNAME: "current_l1",
    SUNIT: "A",
    SFACTOR: 1000,
}
SENSOR_CURRENT_L2 = {
    SKEY: "6100_40465400",
    SNAME: "current_l2",
    SUNIT: "A",
    SFACTOR: 1000,
}
SENSOR_CURRENT_L3 = {
    SKEY: "6100_40465500",
    SNAME: "current_l3",
    SUNIT: "A",
    SFACTOR: 1000,
}
```

rklomp, 2021-05-10:
Would `metering_current_l1`,  `metering_current_l2` and  `metering_current_l3`  be a more applicable name for your sensors?

clasheleander, 2021-05-10:
@rklomp "SENSOR_GRID_POWER_SUPPLIED", how are your readings on this one without energy meter? This is for me power to grid, after home cunsumption.
Should this be moved to Meter aswell?

rklomp, 2021-05-10:
`SENSOR_GRID_POWER_SUPPLIED` is showing the power supplied to the grid. The value is always the same as under PV Generation (key `6100_0046C200`).
Are these two also always the same for you?

[image]
[image]
* showing slight difference due to time interval between screenshots...

rklomp, 2021-05-10:
Ah I mentioned the wrong name `SENSOR_GRID_POWER` is equal to the value shown under PV Generation.  `SENSOR_GRID_POWER_SUPPLIED` is not showing anything. I guess this can be moved to Meter as well

clasheleander, 2021-05-10:
Seems 6100_40463600 and 6100_40468F00 give same readings for me.
Is your sensor "grid_power" and "grid_power_supplied" same on your installation without energy meter?

[image]

clasheleander, 2021-05-10:
Ok, i have data on all under Measured Values, this is what you are "missing" :)

[image]

rklomp, 2021-05-10:
I have updated the list https://github.com/rklomp/pysma/blob/dev/pysma/const.py
Ordered the sensors in the file based on the headers in the GUI. Could you verify all sensors in the `ENERGY_METER_VIA_INVERTER` list? Do you require sensors for Apparent Power and Reactive Power?

rklomp, 2021-05-10:
I see `6100_40468F00` actually is the Apparent power under Grid measurements.

clasheleander, 2021-05-10:
Nice, looking good! thanks!

I personally do not look on values for Apparent and Reactive Power.

DEBUG:asyncio:Using proactor: IocpProactor
INFO:__main__:NEW SID: yNnQMLx3IXl3ZKBV
DEBUG:pysma:Found device class 1
DEBUG:pysma:Sensor status: Will be decoded with "1"[0].val[0].tag from {'1': [{'val': [{'tag': 307}]}]}
DEBUG:pysma:Loading sensors for device class 1
DEBUG:pysma:Sensor energy_meter: Will be decoded with "1"[0].val from {'1': [{'low': 0, 'high': None, 'val': 1901427721}]}
DEBUG:pysma:Energy Meter with serial 1901427721 detected. Adding extra sensors.
DEBUG:pysma:Sensor serial_number: Will be decoded with "1"[0].val from {'1': [{'low': 0, 'high': None, 'val': 3006927010}]}
DEBUG:pysma:Sensor device_name: Will be decoded with "1"[0].val from {'1': [{'val': 'STP10.0-3AV-40 010'}]}
DEBUG:pysma:Sensor device_type: Will be decoded with "1"[0].val[0].tag from {'1': [{'validVals': [9347, 9348], 'val': [{'tag': 9348}]}]}
DEBUG:pysma:Sensor device_manufacturer: Will be decoded with "1"[0].val[0].tag from {'1': [{'validVals': [461], 'val': [{'tag': 461}]}]}
         serial               3006927010
           name       STP10.0-3AV-40 010
           type      SUNNY TRIPOWER 10.0
   manufacturer                      SMA
DEBUG:pysma:Sensor status: Will be decoded with "1"[0].val[0].tag from {'1': [{'val': [{'tag': 307}]}]}
DEBUG:pysma:Sensor pv_power_a: Will be decoded with "1"[0].val from {'1': [{'val': 1315}, {'val': 557}]}
DEBUG:pysma:Sensor pv_power_b: Will be decoded with "1"[1].val from {'1': [{'val': 1315}, {'val': 557}]}
DEBUG:pysma:Sensor pv_voltage_a: Will be decoded with "1"[0].val from {'1': [{'val': 41757}, {'val': 38629}]}
DEBUG:pysma:Sensor pv_voltage_b: Will be decoded with "1"[1].val from {'1': [{'val': 41757}, {'val': 38629}]}
DEBUG:pysma:Sensor pv_current_a: Will be decoded with "1"[0].val from {'1': [{'val': 3153}, {'val': 1442}]}
DEBUG:pysma:Sensor pv_current_b: Will be decoded with "1"[1].val from {'1': [{'val': 3153}, {'val': 1442}]}
DEBUG:pysma:Sensor grid_power: Will be decoded with "1"[0].val from {'1': [{'val': 1816}]}
DEBUG:pysma:Sensor frequency: Will be decoded with "1"[0].val from {'1': [{'val': 4998}]}
DEBUG:pysma:Sensor current_l1: Will be decoded with "1"[0].val from {'1': [{'val': 2578}]}
DEBUG:pysma:Sensor current_l2: Will be decoded with "1"[0].val from {'1': [{'val': 2572}]}
DEBUG:pysma:Sensor current_l3: Will be decoded with "1"[0].val from {'1': [{'val': 2571}]}
DEBUG:pysma:Sensor voltage_l1: Will be decoded with "1"[0].val from {'1': [{'val': 23806}]}
DEBUG:pysma:Sensor voltage_l2: Will be decoded with "1"[0].val from {'1': [{'val': 23571}]}
DEBUG:pysma:Sensor voltage_l3: Will be decoded with "1"[0].val from {'1': [{'val': 23737}]}
DEBUG:pysma:Sensor power_l1: Will be decoded with "1"[0].val from {'1': [{'val': 609}]}
DEBUG:pysma:Sensor power_l2: Will be decoded with "1"[0].val from {'1': [{'val': 601}]}
DEBUG:pysma:Sensor power_l3: Will be decoded with "1"[0].val from {'1': [{'val': 606}]}
DEBUG:pysma:Sensor total_yield: Will be decoded with "1"[0].val from {'1': [{'val': 8652554}]}
DEBUG:pysma:Sensor daily_yield: Will be decoded with "1"[0].val from {'1': [{'val': 43028}]}
DEBUG:pysma:Sensor pv_gen_meter: Will be decoded with "1"[0].val from {'1': [{'val': 8668307}]}
DEBUG:pysma:Sensor metering_power_supplied: Will be decoded with "1"[0].val from {'1': [{'val': 1303}]}
DEBUG:pysma:Sensor metering_power_absorbed: Will be decoded with "1"[0].val from {'1': [{'val': 0}]}
DEBUG:pysma:Sensor metering_total_yield: Will be decoded with "1"[0].val from {'1': [{'val': 4518425}]}
DEBUG:pysma:Sensor metering_total_absorbed: Will be decoded with "1"[0].val from {'1': [{'val': 6360167}]}
DEBUG:pysma:Sensor metering_current_l1: Will be decoded with "1"[0].val from {'1': [{'val': 1441}]}
DEBUG:pysma:Sensor metering_current_l2: Will be decoded with "1"[0].val from {'1': [{'val': 2487}]}
DEBUG:pysma:Sensor metering_current_l3: Will be decoded with "1"[0].val from {'1': [{'val': 2308}]}
DEBUG:pysma:Sensor metering_voltage_l1: Will be decoded with "1"[0].val from {'1': [{'val': 23910}]}
DEBUG:pysma:Sensor metering_voltage_l2: Will be decoded with "1"[0].val from {'1': [{'val': 23651}]}
DEBUG:pysma:Sensor metering_voltage_l3: Will be decoded with "1"[0].val from {'1': [{'val': 23750}]}
DEBUG:pysma:Sensor metering_active_power_l1: Will be decoded with "1"[0].val from {'1': [{'val': 235}]}
DEBUG:pysma:Sensor metering_active_power_l2: Will be decoded with "1"[0].val from {'1': [{'val': 568}]}
DEBUG:pysma:Sensor metering_active_power_l3: Will be decoded with "1"[0].val from {'1': [{'val': 500}]}
DEBUG:pysma:Sensor metering_active_power_consumed_l1: Will be decoded with "1"[0].val from {'1': [{'val': 0}]}
DEBUG:pysma:Sensor metering_active_power_consumed_l2: Will be decoded with "1"[0].val from {'1': [{'val': 0}]}
DEBUG:pysma:Sensor metering_active_power_consumed_l3: Will be decoded with "1"[0].val from {'1': [{'val': 0}]}
                   status             Ok
               pv_power_a           1315 W
               pv_power_b            557 W
             pv_voltage_a         417.57 V
             pv_voltage_b         386.29 V
             pv_current_a          3.153 A
             pv_current_b          1.442 A
               grid_power           1816 W
                frequency          49.98 Hz
               current_l1          2.578 A
               current_l2          2.572 A
               current_l3          2.571 A
               voltage_l1         238.06 V
[... cut ...]
      metering_current_l1          1.441 A
      metering_current_l2          2.487 A
      metering_current_l3          2.308 A
      metering_voltage_l1          239.1 V
      metering_voltage_l2         236.51 V
      metering_voltage_l3          237.5 V
 metering_active_power_l1            235 W
 metering_active_power_l2            568 W
 metering_active_power_l3            500 W
metering_active_power_consumed_l1              0 W
metering_active_power_consumed_l2              0 W
metering_active_power_consumed_l3              0 W

rklomp, 2021-05-10:
Awesome! I think we completely covered the Energy Meter via Inverter case now 😃 I just added the `metering_frequency` as well. If anyone wants to have Apparent Power and Reactive Power we could easily add that later. 

Now about querying the Energy Meter directly. This will load the `DEVCLASS_ENERGY_METER` sensors. Does that cover everything or are there more sensors that could be added?

clasheleander, 2021-05-10:
Tested directly against the Energy meter below, i think you got exactly all info covered! :)
Good job!

DEBUG:asyncio:Using proactor: IocpProactor
INFO:__main__:NEW SID: M4zDSxGOftcO1xWy
DEBUG:pysma:Found device class 65
DEBUG:pysma:Sensor status: Will be decoded with "65"[0].val[0].tag from {'65': [{'val': [{'tag': 307}]}]}
DEBUG:pysma:Loading sensors for device class 65
DEBUG:pysma:Sensor serial_number: Will be decoded with "65"[0].val from {'65': [{'low': 0, 'high': 0, 'val': 1901427721}]}
DEBUG:pysma:Sensor device_name: Will be decoded with "65"[0].val from {'65': [{'val': 'SMA Energy Meter'}]}
DEBUG:pysma:Sensor device_type: Will be decoded with "65"[0].val[0].tag from {'65': [{'validVals': [9327], 'val': [{'tag': 9327}]}]}
DEBUG:pysma:Sensor device_manufacturer: Will be decoded with "65"[0].val[0].tag from {'65': [{'validVals': [461], 'val': [{'tag': 461}]}]}
         serial               1901427721
           name         SMA Energy Meter
           type         SMA Energy Meter
   manufacturer                      SMA
DEBUG:pysma:Sensor status: Will be decoded with "65"[0].val[0].tag from {'65': [{'val': [{'tag': 307}]}]}
DEBUG:pysma:Sensor grid_power: Will be decoded with "65"[0].val from {'65': [{'val': 1308}]}
DEBUG:pysma:Sensor frequency: Will be decoded with "65"[0].val from {'65': [{'val': 4998}]}
DEBUG:pysma:Sensor current_l1: Will be decoded with "65"[0].val from {'65': [{'val': 1471}]}
DEBUG:pysma:Sensor current_l2: Will be decoded with "65"[0].val from {'65': [{'val': 2533}]}
DEBUG:pysma:Sensor current_l3: Will be decoded with "65"[0].val from {'65': [{'val': 2184}]}
DEBUG:pysma:Sensor voltage_l1: Will be decoded with "65"[0].val from {'65': [{'val': 23722}]}
DEBUG:pysma:Sensor voltage_l2: Will be decoded with "65"[0].val from {'65': [{'val': 23557}]}
DEBUG:pysma:Sensor voltage_l3: Will be decoded with "65"[0].val from {'65': [{'val': 23687}]}
DEBUG:pysma:Sensor power_l1: Will be decoded with "65"[0].val from {'65': [{'val': 253}]}
DEBUG:pysma:Sensor power_l2: Will be decoded with "65"[0].val from {'65': [{'val': 578}]}
DEBUG:pysma:Sensor power_l3: Will be decoded with "65"[0].val from {'65': [{'val': 475}]}
                   status             Ok
               grid_power           1308 W
                frequency          49.98 Hz
               current_l1          1.471 A
               current_l2          2.533 A
               current_l3          2.184 A
               voltage_l1         237.22 V
               voltage_l2         235.57 V
               voltage_l3         236.87 V
                 power_l1            253 W
                 power_l2            578 W
                 power_l3            475 W

[image]

rklomp, 2021-05-10:
Thanks! Looking good. 🥳 

I am waiting on https://github.com/rklomp/pysma/pull/1 from #58 to support SMA battery device classes as well.
After that is merged I will merge everything into `master` via #61 and release a new version of `pysma`. In the mean time I will start working on a change for the Home Assistant integration to support the new way of loading sensors.

Thanks for your help getting this all figured out.

clasheleander, 2021-05-10:
It's me to thank! :) good job!

closed 2021-07-07 by rklomp