kellerza/pysma #62: Tigo optimizers still need the custom sensor

popoviciri, 2021-05-09:
I use Tigo optimizers logged by my Sunny Tripower via a Tigo Access Point (TAP). The following custom sensors are logging Power and Temperature of each optimizer (only first two shown as example):
```
- platform: sma
  custom:
    power_tigo_1:
      key: '6100_40652A00'
      unit: W
      factor: 1
      path: '"1"[0].val'
    power_tigo_2:
      key: '6100_40652A00'
      unit: W
      factor: 1
      path: '"1"[1].val'
    temperature_tigo_1:
      key: '6100_40652B00'
      unit: °C
      factor: 10
      path: '"1"[0].val'
    temperature_tigo_2:
      key: '6100_40652B00'
      unit: °C
      factor: 10
      path: '"1"[1].val'
```
To add this in the main integration may be complicated since requires additional inputs regarding the number of panels monitored, if any.
Please consider keeping the yaml custom sensor alongside the GUI integration, unless this is a hard requirement from the Homeassistant folks.
Will keep using a custom integration for now. Thanks for your work!
Cheers!

rklomp, 2021-05-09:
Hi @popoviciri 

I am currently working on new logic to easily provide different sensors based on the device type and connected extra's
Do you know if there is a key to determine if (and how many) are connected?

Would you be able to provide an export of requests to `getAllOnlValues.json` and `getAllParamValues.json`? That would help locating the correct keys to query and what data to expect.

Login to the webinterface of your device and go to the "Home" page
Open developer tools, go to network tab. Click clear in the top left and make sure recording is on.

While keeping the recording running:
Go to page "Instantaneous values" (a request to getAllOnlValues.json should appear)
Go to page "Device parameters" (a request to getAllParamValues.json should appear)

Save these requests as a HAR file and share that file.

This document might help:
https://support.zendesk.com/hc/en-us/articles/204410413-Generating-a-HAR-file-for-troubleshooting

rklomp, 2021-05-09:
After a quick search I found these:

```
            DcMs_PanLevMdul_Vol: {
                obj: "6100",
                lri: "40652800"
            },
            DcMs_PanLevMdul_Amp: {
                obj: "6100",
                lri: "40652900"
            },
            DcMs_PanLevMdul_W: {
                obj: "6100",
                lri: "40652A00"
            },
            DcMs_PanLevMdul_Tmp: {
                obj: "6100",
                lri: "40652B00"
            },
            DcMs_PanLevMdul_SigPwr: {
                obj: "6100",
                lri: "00652C00"
            },
            Operation_PanLevMdul_Health: {
                obj: "6180",
                lri: "08652400"
            },

            DcCfg_PanLevMdul_Lbl: {
                obj: "6800",
                lri: "10852500"
            },
            DcCfg_PanLevMdul_SerNum: {
                obj: "6800",
                lri: "10852600"
            },
            DcCfg_PanLevMdul_Model: {
                obj: "6800",
                lri: "08852800"
            },
            DcCfg_PanLevMdul_SwRev: {
                obj: "6800",
                lri: "10853300"
            },
```

Maybe the `6800_10852600` could be used. If multiple serials are provided I could count them and add sensors as needed.

rklomp, 2021-05-09:
Could you try to run the `example.py` from my `dev` branch against your inverter to see if it returns the optimizers now?
https://github.com/rklomp/pysma/tree/dev

popoviciri, 2021-05-09:
WOW. I did not expect this when I asked for the custom sensor back :). Awesome work!
"The sun is getting low" :wink: and there is no power generated at the moment but inverter is still up. Here's the output from your `example.py`:
```
DEBUG:pysma:Sensor grid_power_supplied: No successful value decoded yet: {'1': [{'val': None}]}
DEBUG:pysma:Sensor grid_power_absorbed: No successful value decoded yet: {'1': [{'val': None}]}
DEBUG:pysma:Sensor grid_total_yield: No successful value decoded yet: {'1': [{'val': None}]}
DEBUG:pysma:Sensor grid_total_absorbed: No successful value decoded yet: {'1': [{'val': None}]}
               grid_power              0 W
                 power_l1              0 W
                 power_l2              0 W
                 power_l3              0 W
                frequency          50.02 Hz
               voltage_l1         229.18 V
               voltage_l2         223.29 V
               voltage_l3         229.42 V
               current_l1            0.0 A
               current_l2            0.0 A
               current_l3            0.0 A
               pv_power_a              0 W
               pv_power_b              0 W
             pv_voltage_a         273.78 V
             pv_voltage_b            0.0 V
             pv_current_a            0.0 A
             pv_current_b            0.0 A
             pv_gen_meter       2817.638 kWh
              total_yield       2817.638 kWh
              daily_yield          10126 Wh
      grid_power_supplied
      grid_power_absorbed
         grid_total_yield
      grid_total_absorbed
                   status             Ok
       optimizer_serial_004C05B400014CXXX
        optimizer_power_0              0 W
      optimizer_current_0            0.0 A
      optimizer_voltage_0           34.2 V
         optimizer_temp_0          342.0 C
       optimizer_serial_104C05B400014CXXX
        optimizer_power_1              0 W
      optimizer_current_1            0.0 A
      optimizer_voltage_1          34.15 V
         optimizer_temp_1          341.5 C
       optimizer_serial_204C05B400014CXXX
        optimizer_power_2              1 W
      optimizer_current_2           0.04 A
      optimizer_voltage_2           34.2 V
         optimizer_temp_2          342.0 C
       optimizer_serial_304C05B400014CXXX
        optimizer_power_3              0 W
      optimizer_current_3            0.0 A
      optimizer_voltage_3          34.35 V
         optimizer_temp_3          343.5 C
       optimizer_serial_404C05B400014CXXX
        optimizer_power_4              1 W
      optimizer_current_4          0.045 A
      optimizer_voltage_4          33.85 V
         optimizer_temp_4          338.5 C
       optimizer_serial_504C05B400014CXXX
        optimizer_power_5              1 W
      optimizer_current_5           0.04 A
      optimizer_voltage_5          33.79 V
         optimizer_temp_5          337.9 C
       optimizer_serial_604C05B400014CXXX
        optimizer_power_6              0 W
      optimizer_current_6            0.0 A
      optimizer_voltage_6           33.9 V
         optimizer_temp_6          339.0 C
       optimizer_serial_704C05B400014CXXX
        optimizer_power_7              0 W
      optimizer_current_7            0.0 A
      optimizer_voltage_7           34.0 V
         optimizer_temp_7          340.0 C
INFO:__main__:Closing Session...
```
I can try tomorrow again, when the sun will hopefully shine again, but even with this data I can say it works! Temperature has to be divided by 10 (I do not know why SMA did that. I apply a factor 10 in my custom sensors as well). I do not have a meter so the first lines are throwing the error.
You were right to use `6800_10852600`. There are other options as well, but serial numbers seems solid! 
I am not comfortable sharing entire HAR files and editing those is a pain, but I got the json response extracts from firefox: [dumps.zip](https://github.com/kellerza/pysma/files/6448144/dumps.zip)
The attached files mirrors this:
[image]
You probably don't need them anymore but maybe you find other useful stuff in there.
Thank you very much once again for your work!
Cheers
[dumps.zip](https://github.com/kellerza/pysma/files/6448144/dumps.zip)

rklomp, 2021-05-10:
I just fixed the `optimizer_temp`. It was querying the wrong key (same key as voltage). I should now return correct values.

popoviciri, 2021-05-10:
Right!
```
                   status             Ok
               pv_power_a            328 W
               pv_power_b              0 W
             pv_voltage_a         264.29 V
             pv_voltage_b            0.0 V
             pv_current_a          1.246 A
             pv_current_b            0.0 A
               grid_power            302 W
                frequency          50.02 Hz
               current_l1          0.462 A
               current_l2          0.466 A
               current_l3          0.453 A
               voltage_l1         229.24 V
               voltage_l2          229.1 V
               voltage_l3         225.59 V
                 power_l1            101 W
                 power_l2            103 W
                 power_l3             98 W
              total_yield       2822.126 kWh
              daily_yield           4460 Wh
             pv_gen_meter       2822.126 kWh
        optimizer_power_0             42 W
      optimizer_current_0          1.285 A
      optimizer_voltage_0           33.1 V
         optimizer_temp_0           25.6 C
        optimizer_power_1             42 W
      optimizer_current_1           1.29 A
      optimizer_voltage_1          32.95 V
         optimizer_temp_1           24.8 C
        optimizer_power_2             42 W
      optimizer_current_2          1.285 A
      optimizer_voltage_2          33.15 V
         optimizer_temp_2           23.1 C
        optimizer_power_3             43 W
      optimizer_current_3           1.29 A
      optimizer_voltage_3          33.45 V
         optimizer_temp_3           22.4 C
        optimizer_power_4             40 W
      optimizer_current_4           1.29 A
      optimizer_voltage_4           31.4 V
         optimizer_temp_4           24.5 C
        optimizer_power_5             40 W
      optimizer_current_5            1.3 A
      optimizer_voltage_5          31.45 V
         optimizer_temp_5           24.0 C
        optimizer_power_6             40 W
      optimizer_current_6           1.29 A
      optimizer_voltage_6          31.75 V
         optimizer_temp_6           26.2 C
        optimizer_power_7             40 W
      optimizer_current_7           1.28 A
      optimizer_voltage_7           32.0 V
         optimizer_temp_7           24.9 C
```
If there is anything else you wanna try on my setup, let me know. I have a Tripower with one string and a Sunnyboy with two strings, both with with TS4-O optimizers.
Cheers!

closed 2021-07-07 by rklomp

rklomp, 2022-09-22:
@popoviciri 

I am creating a file with test data for testing the code.
Would you be able to share the response of your inverter for POST requests to `/dyn/getAllOnlValues.json` and ``/dyn/getAllParamValues.json`?

You can get those via the developer console of your browser when you visit the Instantaneous values and Device parameters pages.

If you need any help let me know. Thanks in advance!

popoviciri, 2022-09-22:
Hi @rklomp,
I had these exported one year ago when you first updated this integration: [dumps.zip](https://github.com/kellerza/pysma/files/6448144/dumps.zip). The optimizer bit reflects this screenshot:
[image]
If you need something else, let me know.
Cheers!

rklomp, 2022-09-22:
Thanks! I did not spot it while scrolling trough this issue. 
Added your data as a test case: https://github.com/kellerza/pysma/pull/109/commits/e71b25e6b5a5e7205aae47c50ad62ded87e5c3b7