kellerza/pysma #64: Error fetching SMA data

roybosch, 2021-05-18:
Since yesterday afternoon, my SMA integration doesn't work properly anymore. The sensors aren't giving any output.

Logbook:

`Logger: homeassistant.components.sma
Source: helpers/update_coordinator.py:212
Integration: SMA Solar (documentation, issues)
First occurred: 10:44:42 (2 occurrences)
Last logged: 10:50:45

Error fetching sma data:`

When I reload the integration, the power output sensor gets a value for a second. After that the value returns to 0 and stays that way.

Also my utilitymeters of my weekly and monthly numbers don't work anymore.

rklomp, 2021-05-18:
Dit you update HA or made any other changes?

roybosch, 2021-05-18:
I installed two new addons; Wireguard and Rhasspy. Beside this, I haven't changed anything.

rklomp, 2021-05-18:
Do you have more logs? Does it show any other errors or exceptions?

roybosch, 2021-05-18:
I'm afraid not. If I look in the logbook itself, it's just this one line:

`2021-05-18 11:49:48 ERROR (MainThread) [homeassistant.components.sma] Error fetching sma data:`

Is there any way to log more?

rklomp, 2021-05-18:
https://www.home-assistant.io/integrations/logger/

For example:
```yaml
logger:
  default: warning
  logs:
    homeassistant.components.sma: debug
```

roybosch, 2021-05-18:
I added the component to the logger. It already was on "warning" as default. After another restart, I got this message:

`Logger: pysma
Source: /usr/local/lib/python3.8/site-packages/pysma/__init__.py:383
First occurred: 14:35:32 (2 occurrences)
Last logged: 14:35:37

No values for sensors: voltage_l1 [6100_00464800],pv_power_a [6380_40251E00],pv_gen_meter [6400_0046C300],total_yield [6400_00260100],sma_dc_voltage [v6380_40451F00]. Response from inverter: {'6400_00262200': {'1': [{'val': 11600}]}, '6400_00543A00': {'1': [{'val': 0}]}, '6100_00543100': {'1': [{'val': None}]}, '6180_08214800': {'1': [{'val': [{'tag': 307}]}]}}
No values for sensors: voltage_l1 [6100_00464800],pv_power_a [6380_40251E00],pv_gen_meter [6400_0046C300],total_yield [6400_00260100],sma_dc_voltage [v6380_40451F00]. Response from inverter: {'6400_00262200': {'1': [{'val': 11602}]}, '6400_00543A00': {'1': [{'val': 0}]}, '6100_00543100': {'1': [{'val': None}]}, '6180_08214800': {'1': [{'val': [{'tag': 307}]}]}}`

After a refresh this warning repeats 33 times and still counting.

At this moment, the power output of the sensor isn't 0 anymore, but it seems to be stuck at 1911 Watts. The latest sensor update was 4 minutes ago. My daily yield [kWh] sensor seems to be working again, but the utilitymeters don't react on this new values of the daily yield.

roybosch, 2021-05-18:
> When did you upgrade your HA to 2021.05.x? Has it worked using that version?

The same day it came out for me. That's weeks ago I believe? It stopped working since yesterday, so it shouldn't have anything to do with the new HA version. 

At this moment, the sensor seems to be working again. It's getting updates every few seconds. I still get the warnings:

> 2021-05-18 16:22:08 DEBUG (MainThread) [homeassistant.components.sma] Finished fetching sma data in 0.133 seconds
2021-05-18 16:22:13 WARNING (MainThread) [pysma] No values for sensors: pv_gen_meter [6400_0046C300],total_yield [6400_00260100],daily_yield [6400_00262200],total_consumption [6400_00543A00],sma_dc_voltage [v6380_40451F00]. Response from inverter: {'6380_40251E00': {'1': [{'val': 1226}]}, '6100_00543100': {'1': [{'val': None}]}, '6180_08214800': {'1': [{'val': [{'tag': 307}]}]}, '6100_00464800': {'1': [{'val': 23496}]}}
2021-05-18 16:22:13 DEBUG (MainThread) [homeassistant.components.sma] Finished fetching sma data in 0.122 seconds
2021-05-18 16:22:18 WARNING (MainThread) [pysma] No values for sensors: pv_gen_meter [6400_0046C300],total_yield [6400_00260100],daily_yield [6400_00262200],total_consumption [6400_00543A00],sma_dc_voltage [v6380_40451F00]. Response from inverter: {'6380_40251E00': {'1': [{'val': 1214}]}, '6100_00543100': {'1': [{'val': None}]}, '6180_08214800': {'1': [{'val': [{'tag': 307}]}]}, '6100_00464800': {'1': [{'val': 23447}]}}
2021-05-18 16:22:18 DEBUG (MainThread) [homeassistant.components.sma] Finished fetching sma data in 0.129 seconds
2021-05-18 16:22:23 WARNING (MainThread) [pysma] No values for sensors: pv_gen_meter [6400_0046C300],total_yield [6400_00260100],daily_yield [6400_00262200],total_consumption [6400_00543A00],sma_dc_voltage [v6380_40451F00]. Response from inverter: {'6380_40251E00': {'1': [{'val': 1204}]}, '6100_00543100': {'1': [{'val': None}]}, '6180_08214800': {'1': [{'val': [{'tag': 307}]}]}, '6100_00464800': {'1': [{'val': 23401}]}}
2021-05-18 16:22:23 DEBUG (MainThread) [homeassistant.components.sma] Finished fetching sma data in 0.137 seconds
2021-05-18 16:22:28 WARNING (MainThread) [pysma] No values for sensors: pv_gen_meter [6400_0046C300],total_yield [6400_00260100],daily_yield [6400_00262200],total_consumption [6400_00543A00],sma_dc_voltage [v6380_40451F00]. Response from inverter: {'6380_40251E00': {'1': [{'val': 1214}]}, '6100_00543100': {'1': [{'val': None}]}, '6180_08214800': {'1': [{'val': [{'tag': 307}]}]}, '6100_00464800': {'1': [{'val': 23473}]}}

rklomp, 2021-05-18:
To get rid of the warning you can disable the related sensors in HA.

roybosch, 2021-05-18:
> To get rid of the warning you can disable the related sensors in HA. (pv_gen_meter, total_yield, daily_yield, total_consumption, sma_dc_voltage)

Those sensors are the sensors I'm using. All other sensors, which don't give any output, are disabled already.

closed 2021-07-07 by rklomp