kellerza/pysma #65: Fetching data from 2 (or more) devices with overlapping sensor-ids?

cpohli, 2021-05-26:
Hi there,
first of all a big thank you for your efforts on this, it makes my day a lot more informative (multiple times :-)

Ok, so here is what i actually have (preventing me from update HA to post 21.4)
- SMA Tripower (for the solar panels)
- SMA Sunny Island (for the battery)
I successfully configured both devices via YAML (getting some "sensor-id already exists, will be ignored") with some standard (pysma)sensors and some (about 7) custom sensors.

This i do have to do, because for instance the battery specific sensors are only given by the Sunny Insland, some others only given by the TriPower, some are given by both (identical readouts, but different ids or even identical ids).

I do need sensors from both devices for quite a lot of template sensors to get the bigger picture of the whole installation.

If i am right, after updating post 21.5.x i can only configure ONE device via integrations and cannot use the custom sensors anymore, right?

So i am basically stuck with 21.4.x (i am relying on my personal HA solar dashboard and some calculations that are done there)?

I would very much like to contribute to the pysma library to get the custom sensor data included (how would i do that (new to github stuff, sorry)) and foremost to work out an integration of more than one device. Is that even planned/thought of?)

Thnak you very much.

cpohli, 2021-05-26:
Oh, thank you for clarification.
I updated my original post for some of my custom sensors, if that helps a bit.
I will look at the file. If i can help with any data, please ask for it, i will deliver to the best of my knowledge.

cpohli, 2021-05-26:
> EDIT:
> I see your 4 custom sensors are in the definitions file 🥳

So, basicly it should work?

cpohli, 2021-05-26:
> 
> 
> As soon as [home-assistant/core#51098](https://github.com/home-assistant/core/pull/51098) has been merged it will be available in the next available release (hopefully 2021.6.0)

This is sounding good and promising :-)
I will wait patiently 👍

rklomp, 2021-05-30:
Unfortunately I expect this wont be merged in the upcoming 2021.6.0 release.
What kind of HA setup are you running? If you run docker you can easily switch to the dev branch until 2021.7.0 is released.

cpohli, 2021-05-31:
Oh, that's a pity. Are there any specific reasons for the delay?
I'm running HA as a "normal" installation on a Raspberry Pi4.
If i use the guide in your kindly provided link, will there be caveats? I know, it's DEV, but as i do run a fairly big HA installation ...

And, if i do this now: will the not yet merged pysma ids talked about here be included in the dev branch right now?
Can i go back to stable, once 2021.7.0 is released or am i stuck with dev?

Thank you very much!

rklomp, 2021-05-31:
The beta for 2021.6.0 was released the day before these changes were merged and not all changes in dev are added to the beta/new release.

De dev branch is basically beta 2021.6.0 including some extra changes scheduled for 2021.7.0. 
You can view the beta release notes of 2021.6.0 here to see if there are any breaking changes for your setup: 
https://rc.home-assistant.io/blog/2021/05/26/release-20216/

You should be able to switch back to stable channel with `ha supervisor options --channel=stable`

cpohli, 2021-05-31:
Ok, thank you again :-)
I i find an hour of time, i will try this out. Backup for sure. :-)

You don't happen to know when one could expect 2021.7.0 though?
I like running this thing on a fairly stable basis (using more than 15 integrations and about 100ish sensors...)

rklomp, 2021-05-31:
Great! Let me know once you have been able to test it. Any feedback will be appreciated :)

rklomp, 2021-06-30:
The beta for 2021.7.0 is available now :)

cpohli, 2021-06-30:
> 
> 
> The beta for 2021.7.0 is available now :)

Woohoo, then i think it's time to backup my installation and take a test drive. 👍 
Thanks for letting me know, i'll report back.

cpohli, 2021-07-02:
Hi, i'm trying out the beta for 2021.7.0 and the multi-device-fetching is working fine.
At least in my case with a SMA Tripower 8.0 and a SMA Sunny Island 4.4.

There are a lot of entities created that are "unavailable" or "zero"  though and it is a lot work to get rid of them (mostly the *_l1,2,3 and *_a,b,c things).
And i'm trying to find out what's up with the naming -  the actual total yield is named "grid_power_2" and things like that (it's not having to do something with the grid? - shouldn't it called pv_power as before?
But those are different things i suspect.
One other question (please advise, if this doesn't belong here): the soc of the battery (SI 4.4) does no longer have a history. my old custom key 6100_00295A00 did act as a normal entity with history and all, the new soc does not behave this way?

rklomp, 2021-07-02:
How do you get 150 entities? Thats a lot.
Can you share some screenshots of your current devices and entities?
I want to make less used entities disabled by default.

rklomp, 2021-07-04:
> There are a lot of entities created that are "unavailable" or "zero" though and it is a lot work to get rid of them, if you don't actually need to read them (mostly the *_l1,2,3 and *_a,b,c things) and one is interested in the "totals".

@cpohli I have create a new pull request for `pysma` to disable 30 sensors by default.
Can you have a look if these are sensors you mean and if there are other sensors you think are not needed by default?
https://github.com/kellerza/pysma/pull/78/files

cpohli, 2021-07-04:
@rklomp I will of course take a look at this. It may take a day or two (not longer), because i'm in my holidays and the family is having demands ;-)

cpohli, 2021-07-05:
> 
> Thanks for testing!
> 
> How do you get 150 entities? Thats a lot.
> Can you share some screenshots of your current devices and entities?
> I want to make less used entities disabled by default.

I may have aexaggerated a bit (as in "feels like" :-)), but in fact i do have 102 entities created for my two devices.
Here are two screenshots for both.
[image]
[image]

cpohli, 2021-07-05:
> "grid_power" is the power supplied to the grid by the inverter. "pv_power_a/b" is the power supplied by the solar panels.

Hmmm, the `grid_power_2` is (at least in my case) the actual pv power of my whole pv strings a+b (not the one delivered to the grid). The power delivered to the grid is `metering_power_supplied.`

rklomp, 2021-07-05:
> Hmmm, the `grid_power_2` is (at least in my case) the actual pv power of my whole pv strings a+b (not the one delivered to the grid). The power delivered to the grid is `metering_power_supplied.`

The `grid_power_2` is the power supplied by the solar inverter on the AC side _of the device_. `pv_power_a` and `pv_power_b` are power of the separate strings on the DC side. `metering_power_supplied` is measured at your SMA energy meter.

cpohli, 2021-07-05:
> The `grid_power_2` is the power supplied by the solar inverter on the AC side _of the device_. `pv_power_a` and `pv_power_b` are power of the separate strings on the DC side. `metering_power_supplied` is measured at your SMA energy meter.

Ok, maybe i misunderstood something here, as i see the sum of `pv_power_a` and `pv_power_b` equals `grid_power_2`.
And `grid_power_2` is not always delivered to the grid (leaving my house) but sometimes used to charge the battery. So only `metering_power_supplied` is really delivered to the grid. 
In that case "grid" was a bit misinterpreted by me as "leaving the house".

closed 2021-07-16 by rklomp