kellerza/pysma #83: Statistic entities home assistant

clasheleander, 2021-08-07:
Any plans for adding support for entities in the new Energy thing in homeassistant?
Thanks!

https://www.home-assistant.io/more-info/statistics/

rklomp, 2021-08-07:
Should be supported already. I am using it without issues.

clasheleander, 2021-08-08:
Ok, Strange, i disabled all yaml config for SMA, uninstalled the integration and installed it again. the Energy integration still not showing correct entities from drop down.

[image]
[image]
[image]
[image]

clasheleander, 2021-08-08:
I have also deleted sma and pysma folder from custom_components to make sure im not using any outdated version.

clasheleander, 2021-08-08:
https://developers.home-assistant.io/docs/core/entity/sensor/#long-term-statistics

clasheleander, 2021-08-08:
Is there any difference between below entities and the others? (with icon "lightning bolt") lightning bolt ones are the only listed in the Energy configuration drop down.
[image]

kellerza, 2021-08-08:
The lightning ones are only ones that satisfy:

- device_class: energy
- State_class: measurement
- Last_reset: set to some date/time

(you can see this in the dev tools as well)

It is only energy devices (kWh unit) so all seems ok from your screenshots

clasheleander, 2021-08-08:
Ok.
How can i then add the right entitie, for example "grid_power" to the production config/make it selectable?

Thanks.

clasheleander, 2021-08-08:
or is it just me who is stupid and has not understood what the new energy function does? :)

kellerza, 2021-08-08:
Its only energy (kWh) and not power (W / kW) 😉

Or see it as the power units you buy from the electricity grid, which they sell as 1 hour of 1000W

once you have it working you nicely soo how much energy/units you consumed per hour

clasheleander, 2021-08-08:
I feel so stupid!!! :)
Sorry i took your time!

kellerza, 2021-08-08:
No worries 👍

closed 2021-08-08 by kellerza