kellerza/pysma #90: Sunny Tripower 10

flopp999, 2021-12-09:
I can get lots of data with your code, very nice tool :)

But I do not get all of the values, some is None.
Is this possible to fix?
I also have SMA Energy Meter.

I am missing metering_current_consumption, metering_total_consumption

```
Sensor(key='6180_08214800', name='status', unit='', factor=None, path=('"{}"[{}].val[0].tag', 'val[0].tag'), enabled=True, l10n_translate=True, value=None)
Sensor(key='6380_40251E00', name='pv_power_a', unit='W', factor=None, path=None, enabled=True, l10n_translate=False, value=None)
Sensor(key='6380_40251E00', name='pv_power_b', unit='W', factor=None, path=None, enabled=True, l10n_translate=False, value=None)
Sensor(key='6380_40251E00', name='pv_power_c', unit='W', factor=None, path=None, enabled=False, l10n_translate=False, value=None)
Sensor(key='6380_40451F00', name='pv_voltage_a', unit='V', factor=100, path=None, enabled=False, l10n_translate=False, value=None)
Sensor(key='6380_40451F00', name='pv_voltage_b', unit='V', factor=100, path=None, enabled=False, l10n_translate=False, value=None)
Sensor(key='6380_40451F00', name='pv_voltage_c', unit='V', factor=100, path=None, enabled=False, l10n_translate=False, value=None)
Sensor(key='6380_40452100', name='pv_current_a', unit='A', factor=1000, path=None, enabled=True, l10n_translate=False, value=None)
Sensor(key='6380_40452100', name='pv_current_b', unit='A', factor=1000, path=None, enabled=True, l10n_translate=False, value=None)
Sensor(key='6380_40452100', name='pv_current_c', unit='A', factor=1000, path=None, enabled=False, l10n_translate=False, value=None)
Sensor(key='6100_40263F00', name='grid_power', unit='W', factor=None, path=None, enabled=True, l10n_translate=False, value=None)
Sensor(key='6100_00465700', name='frequency', unit='Hz', factor=100, path=None, enabled=False, l10n_translate=False, value=None)
Sensor(key='6100_40465300', name='current_l1', unit='A', factor=1000, path=None, enabled=False, l10n_translate=False, value=None)
Sensor(key='6100_40465400', name='current_l2', unit='A', factor=1000, path=None, enabled=False, l10n_translate=False, value=None)
Sensor(key='6100_40465500', name='current_l3', unit='A', factor=1000, path=None, enabled=False, l10n_translate=False, value=None)
Sensor(key='6100_00464800', name='voltage_l1', unit='V', factor=100, path=None, enabled=False, l10n_translate=False, value=None)
Sensor(key='6100_00464900', name='voltage_l2', unit='V', factor=100, path=None, enabled=False, l10n_translate=False, value=None)
Sensor(key='6100_00464A00', name='voltage_l3', unit='V', factor=100, path=None, enabled=False, l10n_translate=False, value=None)
Sensor(key='6100_40464000', name='power_l1', unit='W', factor=None, path=None, enabled=False, l10n_translate=False, value=None)
Sensor(key='6100_40464100', name='power_l2', unit='W', factor=None, path=None, enabled=False, l10n_translate=False, value=None)
Sensor(key='6100_40464200', name='power_l3', unit='W', factor=None, path=None, enabled=False, l10n_translate=False, value=None)
Sensor(key='6400_00260100', name='total_yield', unit='kWh', factor=1000, path=None, enabled=True, l10n_translate=False, value=None)
Sensor(key='6400_00262200', name='daily_yield', unit='Wh', factor=None, path=None, enabled=True, l10n_translate=False, value=None)
Sensor(key='6400_0046C300', name='pv_gen_meter', unit='kWh', factor=1000, path=None, enabled=True, l10n_translate=False, value=None)
Sensor(key='6100_40463600', name='metering_power_supplied', unit='W', factor=None, path=None, enabled=True, l10n_translate=False, value=None)
Sensor(key='6100_40463700', name='metering_power_absorbed', unit='W', factor=None, path=None, enabled=True, l10n_translate=False, value=None)
Sensor(key='6100_00468100', name='metering_frequency', unit='Hz', factor=100, path=None, enabled=True, l10n_translate=False, value=None)
Sensor(key='6400_00462400', name='metering_total_yield', unit='kWh', factor=1000, path=None, enabled=True, l10n_translate=False, value=None)
Sensor(key='6400_00462500', name='metering_total_absorbed', unit='kWh', factor=1000, path=None, enabled=True, l10n_translate=False, value=None)
Sensor(key='6100_40466500', name='metering_current_l1', unit='A', factor=1000, path=None, enabled=True, l10n_translate=False, value=None)
Sensor(key='6100_40466600', name='metering_current_l2', unit='A', factor=1000, path=None, enabled=True, l10n_translate=False, value=None)
Sensor(key='6100_40466B00', name='metering_current_l3', unit='A', factor=1000, path=None, enabled=True, l10n_translate=False, value=None)
Sensor(key='6100_0046E500', name='metering_voltage_l1', unit='V', factor=100, path=None, enabled=False, l10n_translate=False, value=None)
Sensor(key='6100_0046E600', name='metering_voltage_l2', unit='V', factor=100, path=None, enabled=False, l10n_translate=False, value=None)
Sensor(key='6100_0046E700', name='metering_voltage_l3', unit='V', factor=100, path=None, enabled=False, l10n_translate=False, value=None)
Sensor(key='6100_0046E800', name='metering_active_power_feed_l1', unit='W', factor=None, path=None, enabled=True, l10n_translate=False, value=None)
Sensor(key='6100_0046E900', name='metering_active_power_feed_l2', unit='W', factor=None, path=None, enabled=True, l10n_translate=False, value=None)
Sensor(key='6100_0046EA00', name='metering_active_power_feed_l3', unit='W', factor=None, path=None, enabled=True, l10n_translate=False, value=None)
Sensor(key='6100_0046EB00', name='metering_active_power_draw_l1', unit='W', factor=None, path=None, enabled=True, l10n_translate=False, value=None)
Sensor(key='6100_0046EC00', name='metering_active_power_draw_l2', unit='W', factor=None, path=None, enabled=True, l10n_translate=False, value=None)
Sensor(key='6100_0046ED00', name='metering_active_power_draw_l3', unit='W', factor=None, path=None, enabled=True, l10n_translate=False, value=None)
Sensor(key='6100_00543100', name='metering_current_consumption', unit='W', factor=None, path=None, enabled=False, l10n_translate=False, value=None)
Sensor(key='6400_00543A00', name='metering_total_consumption', unit='kWh', factor=1000, path=None, enabled=False, l10n_translate=False, value=None)

```

[image]

rklomp, 2021-12-12:
Are you able to view the data in the SMA GUI? Do you know if the key is correct?

You can find the key by right clicking the value in the SMA GUI and selecting inspect.
You will find something like this:
```html
<td class="col-xs-5 col-sm-4 ng-scope" style="text-align: right">
<span id="v6100_00464800" class="ng-binding">233.22 V</span>
</td>
```

The id of the `span` element is the key we need to query the data. In this case 6100_00464800.

flopp999, 2021-12-12:
Thanks, I can see Total consumption o Sunnyportal and there I checked with Inspect. I didn't think of GUI.
I do not find Total consumption in GUI so I guess Sunnyportal is using some kind of math to show to me.

closed 2021-12-12 by flopp999