kellerza/pysma #91: Example for one key

flopp999, 2021-12-09:
Anyone that have an example how to get data for one or more keys?
I don't want to ask for all keys.
I tried the example in READ.ME but it doesn't work.

rklomp, 2021-12-09:
Can you share the code you tried? As looking at the readme I think this is still valid.

flopp999, 2021-12-09:
It was not Readme, I was remembering wrong.
It was this I followed

https://github.com/kellerza/pysma/blob/master/docs/getting_started.rst

rklomp, 2021-12-10:
Yes that document looks correct. As said, if you can share your code so far, I might be able to point you in the correct direction.

flopp999, 2021-12-10:
This is my py code

```
import pysma
import aiohttp

session = aiohttp.ClientSession(connector=aiohttp.TCPConnector(ssl=False))
url = "https://192.168.100.137"
password = "xxxxxxxxxx"

sma = pysma.SMA(session, url, password)
sma_sensors = sma.get_sensors()
sma_sensors = Sensors()
my_sensor = Sensor("6300_12345678_0", "dummy_sensor") # This key won't work!
sma_sensors.add(my_sensor)
sma_sensors.add(pysma.definitions.pv_power_a)

sma.read(sma_sensors)

for sma_sensor in sma_sensors:
    print(f"{sma_sensor.name}: {sma_sensor.value}")
```

this is my error

```
sudo python3 SMAGithub.py
/home/debian/temp/SMAGithub.py:4: DeprecationWarning: The object should be created within an async function
  session = aiohttp.ClientSession(connector=aiohttp.TCPConnector(ssl=False))
Traceback (most recent call last):
  File "/home/debian/temp/SMAGithub.py", line 10, in <module>
    sma_sensors = Sensors()
NameError: name 'Sensors' is not defined
sys:1: RuntimeWarning: coroutine 'SMA.get_sensors' was never awaited
Unclosed client session
client_session: <aiohttp.client.ClientSession object at 0x7f24a47058b0>
```

rklomp, 2021-12-10:
Please read about asyncio: https://docs.python.org/3/library/asyncio.html
Have a look at example.py how to use it properly. 

> sma_sensors = sma.get_sensors()
This will load all sensors for your device. Something you don't wat as I understand 

> sma_sensors = Sensors()
> my_sensor = Sensor("6300_12345678_0", "dummy_sensor") # This key won't work!
> sma_sensors.add(my_sensor)

You should define a working key here, or use a predefined sensor as you do on the next line.

> sma_sensors.add(pysma.definitions.pv_power_a)

It's three different ways of loading sensors.

flopp999, 2021-12-10:
Finally I made it :)
```
import asyncio
import signal
import aiohttp
import pysma

VAR = {}

async def main_loop(loop):
    async with aiohttp.ClientSession(
        loop=loop, connector=aiohttp.TCPConnector(ssl=False)
    ) as session:
        VAR["sma"] = pysma.SMA(session, "https://192.168.100.137", "password")

        try:
            VAR["running"] = True
            sensors = pysma.Sensors()
            sensors.add(pysma.definitions.daily_yield)
            while VAR.get("running"):
                await VAR["sma"].read(sensors)
                for sensor in sensors:
                    print("{:>25}{:>15} {}".format(sensor.name, str(sensor.value), sensor.unit))
                break
                await asyncio.sleep(2)

        finally:
            await VAR["sma"].close_session()

def main():
    loop = asyncio.get_event_loop()

    def _shutdown(*_):
        VAR["running"] = False

    signal.signal(signal.SIGINT, _shutdown)
    loop.run_until_complete(
        main_loop(loop)
    )

if __name__ == "__main__":
    main()
```

closed 2021-12-10 by flopp999

rklomp, 2021-12-11:
I have simplified the example. Might make things easier for you. See #92.

Based on that your code can be simplified to look like this:

```python
import asyncio
import signal
import aiohttp
import pysma

VAR = {}

async def main():
    def _shutdown(*_):
        VAR["running"] = False

    signal.signal(signal.SIGINT, _shutdown)

    async with aiohttp.ClientSession(
        connector=aiohttp.TCPConnector(ssl=False)
    ) as session:
        VAR["sma"] = pysma.SMA(session, "https://192.168.100.137", "password")

        try:
            VAR["running"] = True
            sensors = pysma.Sensors()
            sensors.add(pysma.definitions.daily_yield)
            while VAR.get("running"):
                await VAR["sma"].read(sensors)
                for sensor in sensors:
                    print("{:>25}{:>15} {}".format(sensor.name, str(sensor.value), sensor.unit))
                break
                await asyncio.sleep(2)

        finally:
            await VAR["sma"].close_session()

if __name__ == "__main__":
    asyncio.run(main())
```