kellerza/pysma #100: Support for SMA Data Manager M / Lite

danielduerr, 2022-05-17:
Hi Johann,

I'm using your integration in Home Assistant with a Sunny Boy 5.0 inverter that runs my secondary solar array and it works well!  But, I have an older Sunny Boy 8000-US inverter on my primary solar array which is based on the older RS-485 comm protocol and does not have WebConnect in it.  So I am trying to find a way to get that inverter "retrofitted" so that your integration can talk to it.  I currently have the SMA WebBox connected to it, which I believe has an API inside of it, but which has been discontinued by SMA so there would be no incentive for you to add support for it.  So, they have referred me to the SMA Data Manager M / Lite to solve for this -- see https://www.sma-america.com/products/monitoring-control/data-manager-m-lite.html for more info.  Would you have any interest in adding support for this device to your library?  It would provide a way for older and newer inverters to be integrated into Home Assistant.

Thanks,
Daniel

kellerza, 2022-05-20:
Hi @danielduerr, can the data manager M lite then interface with your SB8000 and expose the data in a newer format?

Since I created pysma I moved house and recently installed a Sunsynk hybrid inverter. @rklomp has been doing an excellent job in maintaining the library over the last couple of years.

It's almost impossible to add support for a device if you don't own it, especially if you don't have the API specifications. There is countless iterations in testing and hardening a library. 

You are welcome to try add support. To be honest I'm not even sure if this data manager uses any form of WebConnect, in which case it should be fairly easy. It mentions SMA Data2+

danielduerr, 2022-05-20:
@kellerza thanks for the reply, and the background. And it is nice to meet you @rklomp.  I completely agree about the challenges of adding support like this w/o access to the device, and of course with SMA those devices are never cheap. :)

Yes, after speaking with the Apps Engineering team at SMA, it looks like this Data Manager M / Lite product can interoperate with inverters using the variety of communication protocols.  It is also designed to "roll up" multiple inverters into a single, unified monitoring experience.  It appears to offer a UI similar to WebConnect, but I have not seen anything confirming it is in fact WebConnect.  What this probably means to us here is that it would likely push against some of the design constraints of this library since (a) you'd be gathering data for multiple inverter instances, not just one, and (b) the UI would surely be different enough that you'd have to scrape the data a little differently.

I don't personally have the time to work on this effort.  And given the >$600 US price tag of the device, plus installation effort, it's not my first choice.  At this point I'm most likely to just install some Shelly EMs on the inverters and call it a day.  But if you are interested in adding support for this product, I would be willing to consider purchasing it, installing it, and then offering it as a test case where we could work together on the buildout.

closed 2022-08-22 by rklomp