kellerza/pysma #104: Add Insulation Resistance Sensor

peteraclausen, 2022-07-26:
I have a SMA inverter that sometimes shuts down due to the Insulation resistance dropping below 500 kOhm. Can this sensor be added to the integration please?

The new entry in the definitions.py would look something like this:

`
#: PV Isolation Resistance
pv_isolation_resistance = Sensor(“6102_00254F00_0”, “pv_isolation_resistance”, unit=”kOhm”, factor=100, enabled=False)`

Thanks!

kellerza, 2022-08-05:
Hi @peteraclausen can you please create a PR on definitions.py? for an example see #103

peteraclausen, 2022-08-05:
Hi @kellerza 

I'm not actually a developer, but I did know to look at the html code for the Insulation Resistance value from the SMA web interface:

[https://i.imgur.com/OXf2CPr.jpg](url)

And from looking at other entries in definitions.py, I surmised that adding the following entry would expose the Insulation Resistance value:

`#: PV Isolation Resistance
pv_isolation_resistance = Sensor(
     “6102_00254F00_0”, “pv_isolation_resistance”, unit=”kOhm”, factor=100, enabled=False
)
`

I have never created a PR before and I'm not sure I would feel comfortable doing so.

Thanks

rklomp, 2022-08-12:
@peteraclausen could you checkout my branch https://github.com/rklomp/pysma/tree/isolation_resistance and test with example.py if this works?

peteraclausen, 2022-08-15:
@rklomp thank you so much for creating a branch with the isolation_resistance sensor added.  I see where you made the update to definitions.py, but I'm unsure how exactly to drop that into my install.  I asked around but I haven't gotten anything back yet.

So where do I drop the updated definitions.py (renamed as example.py) and how to I point my existing SMA integration to it?

Thanks and sorry for being such a noob when it comes to python.

rklomp, 2022-08-15:
You can use the following commands on any linux host:

```bash
# git clone -b isolation_resistance https://github.com/rklomp/pysma.git
# cd pysma
# python3 example.py
```

The example.py script needs some command line parameters. 
Then if that runs successfully, see if the isolation resistance value is shown in the output.

peteraclausen, 2022-08-16:
@rklomp   Thanks, that did the trick.  I tested it on all 8 of my inverters and they all returned the proper value.

Here's the output from one of them:

`      status             Ok
               pv_power_a            381 W
               pv_power_b            580 W
               pv_power_c            579 W
             pv_voltage_a         240.77 V
             pv_voltage_b         356.31 V
             pv_voltage_c         358.57 V
             pv_current_a          1.585 A
             pv_current_b          1.629 A
             pv_current_c          1.617 A
  pv_isolation_resistance       17202.77 kOhm
               grid_power           1511 W
                frequency          60.02 Hz
               current_l1          6.078 A
               current_l2          6.078 A
               current_l3
               voltage_l1          124.8 V
               voltage_l2         125.36 V
               voltage_l3
                 power_l1            754 W
                 power_l2            757 W
                 power_l3              0 W
              sps_voltage
              sps_current
                sps_power              0 W
              total_yield      38010.486 kWh
              daily_yield
             pv_gen_meter      38010.486 kWh
INFO:__main__:Closing Session...
`

kellerza, 2022-08-17:
Nice! Is this the expected value? Or should it be 1700kOhm only?

peteraclausen, 2022-08-17:
That is the expected value and matches what I see in the SMA web GUI.  By default, there is a safety cutoff at 500 kOhm, meaning that if the value drops below that, the inverter won't come online.  When I deployed my inverters, I had one that would not come online initially because the value was around 400 kOhm, so I reduced the threshold to 350 kOhm to bring it up.

A few weeks later, I had another inverter that did not come online, and it had a value of 380 kOhm, so I dropped the trigger down to 300 kOhm.  I'm finding this issue only seem to occur in early mornings when its very humid and once the sun comes out and dries everything up, all my inverters report values above 1,000 kOhm.

The measured value for this attribute is from around 100 kOhm to 3,000 kOhm.

Being able to track this attribute in Grafana for all 8 of my inverters will allow me to set the proper threshold on each inverter to ensure I don't dip below it.

rklomp, 2022-08-17:
So the value now shown (17,202) is a factor 10 to high?

peteraclausen, 2022-08-17:
Yes.  I should have caught that.

[image]

Actual reading:

[image]

peteraclausen, 2022-08-17:
Looked at the values on all 8 inverters to compare them:

`
SB 7.7 16208.2  kOhm		1,620.82 kOhm
SB 7.7 20196.44 kOhm		2,019.64 kOhm
SB 7.7 10946.39 kOhm		1,094.64 kOhm
SB 6.0  7568.06 kOhm		  756.81 kOhm
SB 6.0  6796.38 kOhm		  679.64 kOhm
SB 6.0 21207.73 kOhm		2,120.77 kOhm
SB 7.0 21623.24 kOhm		2,162.32 kOhm
SB 7.0 20897.13 kOhm		2,089.71 kOhm
`
Left is pysma, right is SMA web gui.

rklomp, 2022-08-17:
If you do a `git pull` it will update and should have the correct factor. Can you test?

peteraclausen, 2022-08-17:
Maybe I'm doing it wrong?

`peter@sandbox:~/pysma$ sudo git pull https://github.com/rklomp/pysma.git
From https://github.com/rklomp/pysma
 * branch            HEAD       -> FETCH_HEAD
Already up to date.
peter@sandbox:~/pysma$
`

peteraclausen, 2022-08-17:
Yep it now has the correct factor.

Example:

`pv_isolation_resistance       2162.324 kOhm`

closed 2022-08-17 by rklomp

rklomp, 2022-08-17:
See also https://github.com/home-assistant/core/pull/76937

I expect it to be available in the next major release of HA

PR #106 (2022-08-17): Add pv_isolation_resistance
Fixes #104