kellerza/pysma #105: New "Sunny Tripower Smart Energy"

Toooias, 2022-08-04:
The new SMA Sunny Tripower Smart Energy do not seem to be supported yet.
Is this the case and  can the devices be added? 

Thanks a lot!

FindusK, 2022-08-16:
Hi there,

I also have the new SMA Sunny Tripower SE.
I'm currently working on additions to the library.

There seem to be multiple differences to the other devices:
- the URL to get the sensordata (`URL_VALUES = "/dyn/getValues.json"`)  only delivers a few values like IP-Address or the SW-Version, the values for the sensors are available over `/dyn/getAllOnlValues.json`
- there seems to be some kind of special character in `/data/l10n/en-US.json`, that crashes the Logging library
- the return value to check the device_class (function `get_devclass`) changed, I modified the code, so it detects the Tripower SE as a classic Inverter, with this modification I can read some of the values
- all values seem to be packet into a subarray (I'm not sure if it's done like this in other devices)
- the Tripower SE also delivers values for an energymeter and a battery

I think the best option would be to add a new device-class for the Tripower SE and to switch the `URL_VALUES` after detecting this new device class.<br>I hope this can help to implement the new inverter.

I attach the two json-documents to make the discussion about the new inverter easier <br>(we don't have a battery yet, so these values are zero):
[getAllOnlValues.json.txt](https://github.com/kellerza/pysma/files/9351174/getAllOnlValues.json.txt)
[getValues.json.txt](https://github.com/kellerza/pysma/files/9351176/getValues.json.txt)

FindusK, 2022-08-16:
I'm a few steps further into the additions and have to revise my statement about the changed URL:

I used a browser to get the json-documents and in the browser the website uses getAllOnlValues.json for all values and getValues.json is nearly empty. The values in getValues.json change depending on the website you visit.
Pysma initializes the website correctly and can gather all values over getValues.json (different behavior compared to Firefox)

I have now added new JMESPATHs for the inverter and can receive nearly all values (Inverter and energy-meter via Inverter), when I change the device class to battery and can also receive nearly all values there.

kellerza, 2022-08-17:
Thanks @FindusK ! It would be great if you can do a PR with the suggested changes...

I don't have a SMA inverter anymore, but there are several unit tests and @rklomp can check that these changes do not affect his inverter

FindusK, 2022-08-17:
Sure, I'm currently testing the additions in my home-assistant instance, the unit-tests all work.

I also fixed the async-unit-tests, that didn't run correctly in the github actions.

I think I will have the PR ready tomorrow.

rklomp, 2022-08-17:
Thanks @FindusK ! Looking forward to it.

Sonusss, 2022-09-25:
@FindusK  I have a SUNNY TRIPOWER 10.0 SE and like to test your code.

I changed files in python libs of my HA 2022.9.6 but now when trying to add SMA integration I got the following message:
_This integration does not support configuration via the UI. If you followed this link from the Home Assistant website, make sure you run the latest version of Home Assistant._

How can I get your code to work ?

rklomp, 2022-09-25:
@Sonusss,

I think the easiest way is to just install the custom lib. Below command should install the branch containing the changes by @FindusK and my additional changes as per #109. The command should be run inside the home assistant docker container or venv.
`pip install git+https://github.com/rklomp/pysma.git@dynamic-sensors`

_Note: I have not tested this command_

Sonusss, 2022-09-25:
@rklomp Your command worked without error in my docker container but it looks like the code is not active.
Adding SMA integration works but without any sensor...

rklomp, 2022-09-25:
Did you restart HA after running the pip command?

Sonusss, 2022-09-25:
Indeed

Sonusss, 2022-09-25:
I got a lot of data in the HA log file for SMA and pysma. Looks like the connection is well established but no sensors reported to HA.

Logging settings :
```
logger:
  default: info
  logs:
    homeassistant.components.sma: debug
    pysma: debug
```

rklomp, 2022-09-25:
Can you share the logging in a file?

Sonusss, 2022-09-25:
@rklomp 
Here is the log
[pysma.zip](https://github.com/kellerza/pysma/files/9640991/pysma.zip)

rklomp, 2022-09-25:
Looks like it is still running the old pysma code

rklomp, 2022-09-25:
Try this command instead:
`pip install -e git+https://github.com/rklomp/pysma.git@dynamic-sensors#egg=pysma`

Sonusss, 2022-09-25:
@rklomp  It did it ;-)

I have 1 device and 46 entities...

You're a Master !

rklomp, 2022-09-25:
Good to hear it is working! Can you share the logging again?

Sonusss, 2022-09-25:
@rklomp 

Here you are...
[pysma.zip](https://github.com/kellerza/pysma/files/9641045/pysma.zip)

rklomp, 2022-09-25:
Thanks! I added your device data in the [test data](https://github.com/kellerza/pysma/pull/109/commits/7477b1cc6f5e0dc58905221aac2c1cde2dacb34b)

phdelodder, 2022-10-03:
When will the this version/fix be included in HA?

rklomp, 2022-10-04:
> When will the this version/fix be included in HA?

I hope to get this included in HA 2022.11

phdelodder, 2022-10-10:
How is the progress going? Do you need any help with testing?

lucaberti, 2022-10-17:
> Try this command instead: `pip install -e git+https://github.com/rklomp/pysma.git@dynamic-sensors#egg=pysma`

I have the same problem. I'm currently running HA in HA OS. Is there any canche to implement this fix on HA OS or should switch my congiguration, running HA on a container?

closed 2022-10-19 by rklomp

ReinoutW, 2022-10-21:
I have tried the updated pysma as well on my home-assistant core instance, to see if everything works on my smart energy invertor.

Steps:
1. Running the command to install the updated pysma
`pip install -e git+https://github.com/kellerza/pysma.git#egg=pysma`
2. Confirm version 0.7.1 is installed
`pip show pysma`
3. Restart / don't restart home-assistant (I tried both)
4. Add the SMA Solar integration in home-assistant
5. This takes a bit of time, so I went back to the console in the docker container, I requested the version of pysma once again as I thought it was just reverting this. And indeed, it was reverted to 0.6.12
6. Upgrade pysma again did not change the behaviour of the SMA integration in home-assistant.

In the logs, I can see 0.6.12 is used and not 0.7.1, no matter what I have tried.

`
2022-10-21 14:27:14.949 DEBUG (MainThread) [pysma.sensor] Sensor serial_number: Will be decoded with "9"[0].val from {'9': [{'low': 0, 'high': 4294967294, 'val': 3012392310}]}
2022-10-21 14:27:14.949 DEBUG (MainThread) [pysma.sensor] Sensor device_name: Will be decoded with "9"[0].val from {'9': [{'val': 'SN: 3012392310'}]}
2022-10-21 14:27:14.949 DEBUG (MainThread) [pysma.sensor] Sensor device_type: Will be decoded with "9"[0].val[0].tag from {'9': [{'validVals': [19048, 19049, 19050, 19051], 'val': [{'tag': 19051}]}]}
2022-10-21 14:27:14.950 DEBUG (MainThread) [pysma.sensor] Sensor device_manufacturer: Will be decoded with "9"[0].val[0].tag from {'9': [{'validVals': [461], 'val': [{'tag': 461}]}]}
`

rklomp, 2022-10-21:
This will be included in HA 2022.11. Beta version will be release the 26th of october.