kellerza/pysma #110: Include Current Power

phdelodder, 2022-10-15:
Could it be possible to include PV Power id "v6100_0046C200", according to my inverter (Sunny Tripower 6.0 SE) it is the "Current Power" generated by the solar panels. It's totally different from the pv_power_a and pv_power_b values. 

[image]

rklomp, 2022-10-19:
So This would be the sum of  pv_power_a and pv_power_b? Can you confirm?

Can you confirm it is a different value from grid_power 6100_40263F00?

phdelodder, 2022-10-19:
@rklomp pv_power != to the sum of pv_power_a and pv_power_b as they are from the DC Side, pv_power is from the AC Side. 
[image]

grid power has also a different value then pv_power.

rklomp, 2022-10-19:
So why is there a difference between grid_power and pv_power? As both are from the AC side? Does it have something to do with your battery?

I you want you can add it to [definitions.py](https://github.com/kellerza/pysma/blob/master/pysma/definitions.py) and create a pull request to have it included :)

phdelodder, 2022-10-19:
@rklomp created the PR

phdelodder, 2022-10-19:
I have no idea why there is a difference between grid_power and pv_power, if I would take the sum of pv_power and battery_discharge I would get grid power of subtract battery_charge from pv_power to get grid_power. How ever I prefer to expose it as an entity then to calculate it through templates
[image]

rklomp, 2022-10-19:
Yes this makes sense. As I have a sunnyboy with no battery, grid_power and pv_power will always be the same. But in the case of your device the barrery also provides part to the power

closed 2022-10-19 by rklomp