kellerza/pysma #116: Energy meter (Home manager 2.0) is not detected anymore

TuokkoEsa, 2022-12-19:
Home Manager 2.0 is not detected anymore correctly. 
It was working earlier. Probably HA core update to 2022.12.5 (or 2022.12.6) broke it.

Looks like pysma scans for energy meter serial no from 6800_008AA300 but inverter is reporting it in 6800_008AA200.

Inverter: STP6.0-3AV-40 410
Firmware: 3.11.11.R

Home Assistant 2022.12.7
Supervisor 2022.11.2
Operating System 9.4
Frontend 20221213.1 - latest

[home-assistant_sma_2022-12-19T12-48-50.023Z.log](https://github.com/kellerza/pysma/files/10259677/home-assistant_sma_2022-12-19T12-48-50.023Z.log)

rklomp, 2022-12-19:
There was no change in pysma in this part. Did your device update to a different version?

TuokkoEsa, 2022-12-19:
I don't think so.

I updated Inverter firmware after HA stopped working. It was before 3.10.16.R.

TuokkoEsa, 2022-12-19:
Anyway for some reason 6800_008AA200 does not contain serial number of Home Manager 2.0. It is now contained in 6800_008AA300. And therefore 

You can also see from logs, that /dyn/getAllOnlValues.json returns fields     
ENERGY_METER_VIA_INVERTER: [
        metering_power_supplied,
        metering_power_absorbed,
        metering_frequency,
        metering_total_yield,
        metering_total_absorbed,
        metering_current_l1,
        metering_current_l2,
        metering_current_l3,
        metering_voltage_l1,
        metering_voltage_l2,
        metering_voltage_l3,
        metering_active_power_feed_l1,
        metering_active_power_feed_l2,
        metering_active_power_feed_l3,
        metering_active_power_draw_l1,
        metering_active_power_draw_l2,
        metering_active_power_draw_l3,
        metering_current_consumption,
        metering_total_consumption,
    ],
except two last.

I tested with Postman that inverter also returns those fields (including correct values) with /dyn/getValues.json after fields are inserted to keys list.

Home Manager measurements are also shown correctly in inverter web page and in sunnyportal.com.

rklomp, 2022-12-20:
@Sonusss
I see the same in [the data I got from your device](https://github.com/kellerza/pysma/pull/109/commits/7477b1cc6f5e0dc58905221aac2c1cde2dacb34b):
```
                    "6800_008AA200": {
                        "9": [{"low": 0, "high": None, "val": 1901444353}]
                    },
                    "6800_008AA300": {"9": [{"low": 0, "high": None, "val": None}]},
```

Do you have a extra energy meter device? As all the metering sensor values in your data are None.

rklomp, 2022-12-20:
@TuokkoEsa 
Looking at the GUI it seems 6800_008AA300 will contain data if the meter is usable. Can you check what your device displays here?
[image]

TuokkoEsa, 2022-12-20:
[image]

[image]

rklomp, 2022-12-20:
Can you find anything in your device log why it is not "usable" anymore?

TuokkoEsa, 2022-12-20:
Not really. There are some "Communication fault with meter unit", but those has been there for ages.

One thing what could cause this, is that solar panels are now covered with snow, and inverter has not produced any power for 3-4 weeks.

Instant values / AC Side / Measured values page is displaying correct values real time.

[image]

Is it possible to add similar test to 6800_008AA200 when 6800_008AA300 fails?

```
        _LOGGER.debug("Checking if Energy Meter is present...")
        # Detect and add Energy Meter sensors
        em_sensor = copy.copy(definitions.energy_meter)
```

TuokkoEsa, 2022-12-21:
@rklomp 
Mystery solved, it was because of snow.

Now snow has melted from solar panels, and now inverter reports as it was reporting earlier.
`'6800_008AA200': {'1': [{'low': 0, 'high': None, 'val': 3002849626}]}, 
'6800_008AA300': {'1': [{'low': 0, 'high': None, 'val': 3002849626}]}`

Theory why that happened?
- Inverter was waiting too long (days/weeks) solar power and went to some kind of waiting mode
--> Inverter stopped to send 6800_008AA300, but continues to send measurements
--> Everything works in HA
14.12:
- HA update was installed
--> SMA integration was restarted and availability of energy_meter was re-scanned
--> Energy meter was not detected from 6800_008AA300
--> Integration started to poll only GENERIC_SENSORS
21.12:
- Snow melted
--> Inverter started to send again 6800_008AA300
--> HA still polls only GENERIC_SENSORS
- HA was resetted
--> SMA integration was restarted and availability of energy_meter was re-scanned
--> Energy meter was detected from 6800_008AA300
--> Integration started to poll GENERIC_SENSORS and ENERGY_METER_VIA_INVERTER
--> Everything works in HA

Proposal for changes:
```
energy_meter2 = Sensor("6800_008AA200", "energy_meter")
----
        _LOGGER.debug("Checking if Energy Meter is present...")
        # Detect and add Energy Meter sensors
        em_sensor = copy.copy(definitions.energy_meter)
        em_sensor.extract_value(all_sensors)

        em2_sensor = copy.copy(definitions.energy_meter2)
        em2_sensor.extract_value(all_sensors)

        if em_sensor.value or em2_sensor.value:

```

sweetpi, 2023-02-03:
I had the same problem with a newly installed SUNNY TRIPOWER 8.0 SE which acts as an energy meter too. Important sensors like for example pv_gen_meter where marked as not available. After a lot of debugging i was giving up. A month later after restarting home assistant it just works and sensor values are there. I think the problem is the newly introduces sensor detection in https://github.com/kellerza/pysma/pull/109