kellerza/pysma #119: New Inverter Generation "Tripower X (STP XX-50)" not working

Kurisutian, 2023-04-01:
After setting up a new SMA inverter (SMA Tripower X STP 25-50) I wanted to integrate it with Home Assistant and the current SMA integration.
I have checked with all possible combinations of the settings but unfortunately the connection seems to be impossible. Although I have the installer password, a static IP address and did disable the ssl verification I always get an authentication error (Invalid Authentication) as a result.
So currently the SMA integration does not seem to work with the new SMA inverter generation at all.
I also found a comment in the HA Forum where somebody mentioned that pysma does not support the new generation at all and after opening an issue at the HA side (https://github.com/home-assistant/core/issues/90313) I was asked to open one here as it seems to be rather related to pysma.

rklomp, 2023-04-01:
I need debug logging to troubleshoot this as I dont own such a device.

Kurisutian, 2023-04-01:
@rklomp as I just literally turned it on last week and it's my first inverter ever would you mind pointing me to the direction where to get the debug logs? Or are you talking about the support log? I'm not quite sure which data you'd need and how I can get them. Sorry for that noobish start of mine. 🫣

rklomp, 2023-04-01:
The best option for testing would be to clone this repository and run the [example.py](https://github.com/kellerza/pysma/blob/master/example.py) script.

Kurisutian, 2023-04-01:
Here's what I did receive doing this (password is of course removed definitely the right one was used):

```
WARNING:pysma:Password should not exceed 12 characters
DEBUG:pysma:Sending POST request to /dyn/login.json: {'data': '{"right": "istl", "pass": "<REMOVED>"}', 'headers': {'content-type': 'application/json'}}
WARNING:pysma:Request to /dyn/login.json did not return a valid json.
ERROR:pysma:Could not start session: Session ID expected [result.sid]
WARNING:__main__:Authentication failed!
```

I don't know about previous inverters but this one does allow me to log in with a chosen username and not like the energy meter I also have to pick either "user" or "installer". In the initial setup it asks for a username and the role. Perhaps this was different in the previous generation? Can't tell because I do not have some "older" model which I could access so it's just a random guessing.

rklomp, 2023-04-01:
Could you log traffic in your browser while logging in to the webinterface of your inverter and share that info? 
https://support.zendesk.com/hc/en-us/articles/4408828867098-Generating-a-HAR-file-for-troubleshooting

You can send it in a pm if you prefer that.

Kurisutian, 2023-04-01:
> Could you log traffic in your browser while logging in to the webinterface of your inverter and share that info? https://support.zendesk.com/hc/en-us/articles/4408828867098-Generating-a-HAR-file-for-troubleshooting
> 
> You can send it in a pm if you prefer that.

Don't really know how to send a PM over GitHub. I was under the impression there is no such option... do you have a place where I can send the file to?

rklomp, 2023-04-01:
Ah yes you are right. I thought my discord name was on my profile. Do you have discord? My username is rklomp#3849

Kurisutian, 2023-04-01:
Thanks, I did sent you a request (Kurisutian#0542)... I'm not that often in there but I did create an account at one point.

rklomp, 2023-04-01:
Thanks for the har file. I looked at the communication with the inverter. These new models use a new api to communicate with the device which is completely different to the current implementation for `pysma`. I don't think we can include it into this library, so a new library should be developed to support this.

littleyoda, 2023-04-02:
Just as a remark: I have written a small proof of concept script for sending data from the Tripower X series via MQTT to the Home Assistant.
https://github.com/littleyoda/Home-Assistant-Tripower-X-MQTT

A general question: Do you insist on a new library or would you accept a PR with an extension?
I think this is because it is difficult for users to know which library to use.

rklomp, 2023-04-02:
If we can integrate it nicely into one library that would be great. Just make a proposal and I will have a look at it :)

mrestorff, 2023-04-06:
Implementing the new Tripower X series is going to involve a bit more work, as the webconnect API has been changed up completely. 

There's now a bearer token required for authentication and proper user management (no more password-only login). All the endpoints have changed, as well as the JSON responses and naming scheme of the measurements and parameters.

AnotherDaniel, 2023-04-07:
I've just today started trying to integrate my tripower unit with homeassistant, running into the same issue,leading me here. Will clone the repo and look at the code later today, is there any information/logs/etc needed to help this along?

Kurisutian, 2023-04-21:
Has there been any new insights on this? Just asking out of curiosity not out of impatience (I have an energy meter in place so I can get the most basic information needed from there). Just wanted to know if there has been any breakthrough from anyone or anything I could help with by testing it or anything. ;)

studiotijn, 2023-06-10:
Hello everyone, I found this thread since I am owner of a Tripower X STP 15-50. Really looking forward to the alfa-reseales and final result. I am unexperienced in writing code and troubleshooting connection-issues and figuring out how to write a code that implements the required API etc.
If I can be of any help (for testing/validation) please let me know.
Keep up the good work.

steffenknott, 2023-06-20:
Hi all, I got here because I tried adding a Tripower X STP 25-50 to the SMA integration in Home Assistant. If I can help testing, let me know.
BR
Steffen

SteffenViperKoch, 2023-09-26:
> Hi all, I got here because I tried adding a Tripower X STP 25-50 to the SMA integration in Home Assistant. If I can help testing, let me know. BR Steffen

Same here. Willing to test out any new code ;-)

miquelbeltran, 2023-09-27:
FYI, this model can now be integrated using the ModBus protocol and the SunSpec integration in Home Assistant.

There is some info here: https://community.home-assistant.io/t/sma-tripower-x-stp-xx-50/536572/21

redfusion83, 2024-02-11:
I am also available for testing. It would be great if the X inverters would also work with the official integration. 

By the way, these inverters use a different operating system called ennexOS. This operating system will be used by SMA in the future. I don't know whether this is relevant for this integration, but it might be useful information.

MeMichael151, 2024-03-14:
I am in the same situation. SMA Tripower X 12KW to be integrated in HA. 
Is there a way for me to support the integration? I'm not a dev, but perhaps I can help otherwise? Testing, Documentation etc.

littleyoda, 2024-03-26:
Can someone run the example.py from this lib on a supported (Tripower SE?) inverter?
I need the information to see how I can adapt the Lib to the Tripower x.

https://github.com/kellerza/pysma/blob/master/example.py

littleyoda, 2024-03-27:
To avoid redundant work:
I have started adding support of Tripower X to pysma.  

https://github.com/littleyoda/pysma/tree/tripowerX
You can already test the example.py to view the measured values.
`python3 example.py https://<ip> <user> <pwd> ennexos`

The next steps for integration in Home Assistant will be a lot of work. 
Don't expect any successes in the next few weeks, I only have limited resources available for this project and I will also be dependent on help from @kellerza and/or @rklomp

littleyoda, 2024-04-01:
For the german readers:
https://www.photovoltaikforum.com/thread/225270-home-assistant-und-sma-ger%C3%A4te/?postID=3682369#post3682369

I have finished a prototyp.

For testing reason I created a custom integration first.

Now supported:
-    (as before) Webconnect devices
-    EnnexOS-based devices (currently only Tripower X)
-    Energy Meter (EMETER-10 + 20) and Sunny Home Manager 2

The current version is for people who are familiar with Home Assistant and HACS.

The version will certainly contain errors. 
At the latest when my changes are merged in pysma, you will probably have to recreate the devices in HA and thus also lose the history.

Procedure:
- HACS must be installed
- You must add my repository in HACS.
   https://github.com/littleyoda/ha-pysmaplus
- Then add the following integration via HACS
-   SMA Devices Plus (not for production use)
-  Restart HA
- Add your SMA devices based on the SMA Device Plus integration.
  (Settings / Devices & Services / Integrations)
-  Not all entities are activated automatically. You may have to do this manually.
  (I have initially used the default values of the original SMA integration, which are not optimal for the new devices)

Kurisutian, 2024-04-01:
> For the german readers: https://www.photovoltaikforum.com/thread/225270-home-assistant-und-sma-ger%C3%A4te/?postID=3682369#post3682369
>
> I have finished a prototyp.
>
> For testing reason I created a [...]

Just tested with an Energy Meter 2 and a Tripower X-25, both seem to work flawless for now. Thanks for your efforts. I will keep monitoring them for stability. Hopefully it will be more reliable than the sunspec protocoll which sometimes just not work.

SteffenViperKoch, 2024-04-02:
> For the german readers: https://www.photovoltaikforum.com/thread/225270-home-assistant-und-sma-ger%C3%A4te/?postID=3682369#post3682369
>
> I have finished a prototyp.
>
> For testing reason I created a [...]

Just installed on a test platform of HA. Works fine with my Sunny Tripower X 12 and my Sunny Boy Storage SBS5.0-10.
What values are you able to get from the integration (EnnexOS)? Everything the webpage shows?

One question though: How is the detection of the Sunny Home Manager 2.0 working? When I try to add it to the integration HA hangs for around 10 minutes (seems to go into some kind of timeout then). No messages shown.
Can you make it possible to enter the IP of the SHM2.0?

kr
Steffen

littleyoda, 2024-04-02:
> Just installed on a test platform of HA. Works fine with my Sunny Tripower X 12 and my Sunny Boy Storage SBS5.0-10. 
> What values are you able to get from the integration (EnnexOS)? Everything the webpage shows?

No everything, but a lot of it.

see:
https://github.com/littleyoda/pysma/blob/485aeec1be2238c78f1bf98c38678eb7df7a9d3d/pysma/smaennexos.py#L24
every entry  with a "Identifier" in it  is transferred to HA. 

If you need further measured values, please let me know.

> One question though: How is the detection of the Sunny Home Manager 2.0 working? When I try to add it to the integration HA hangs for around 10 minutes (seems to go into some kind of timeout then). No messages shown. Can you make it possible to enter the IP of the SHM2.0?

This is definitely a bug. A timeout is probably missing there.
The IP address of the SHM2 does not help.  The SHM2 transmits the packets via multicast (normally one per second). It does not offer an API or simlar. So I have to wait for the packets to arrive at the Home Assistant instance. In your case, it may be that the multicast packets are not arriving on your Home Assistant system for whatever reason.

MeMichael151, 2024-04-02:
@littleyoda 
This is great, thank you. Will this ever be part of the "normal" integrations? Or will it always be in HACS?

littleyoda, 2024-04-02:
> @littleyoda This is great, thank you. Will this ever be part of the "normal" integrations? Or will it always be in HACS?

I definitely want to integrate my changes into pysma at some point. After that, the way is open for HA core.

But that will be a long way off. For this I will also need support from the library authors, with whom I am already in contact, but who also only have very limited time. 

As I also only have limited resources, I'm not in a hurry.

Fokko, 2026-08-30:
@KimCarlsen I'm able to connect my Tripower STP 8.0 SE from Home Assistant, so I think it should work.

@littleyoda Appreciate all the work, let me know if there is anything that the community can help out with