kellerza/pysma #120: Feature request: control active power

gigatexel, 2023-04-27:
As dynamic electricity pricing is getting more and more available, the need to be able to controle the inverter output power is also rising.
For instance, it might not be feasible to generate more power than your own consumption during negative prices (or maybe no generation at all).
Simply switching off the convertor on the AC or DC side is offcourse possible, but not optimal.

The SMA web interface enables you to define the maximum active power, as a value in W and as a percentage.
Would it be possible to control this using PySMA and later on using Home Assistant?

[image]

thestonewell, 2024-05-26:
There's an additional configuration parameter that would be interesting: to force charge/force discharge the battery. A good example is to charge the battery over night - especially in winter - when electricity prices are low.

My SUNNY TRIPOWER 10.0 SE provides the necessary Battery Installer Mode (search for `Installation mode` or `Bat.IstlMod`):

[image]

While this does not relate to active power it relates because of the same use case of dynamic energy pricing.

I did look at the code too. It will be a more difficult change as
- I don't know how to really get the parameter id (something like `6380_40251E00_0` but not this value
- unlike all things before we no longer talk about a sensor that just reports the current value
  - needs to turn into a configurable parameter
  - needs to adhere to the possible values that can be set
  - needs to be integrated in HA then as either a https://developers.home-assistant.io/docs/core/entity/select or as an https://developers.home-assistant.io/docs/core/entity/number with respective min/max values
  
@littleyoda from https://github.com/littleyoda/pysma/tree/tripowerX any ideas?

littleyoda, 2024-05-26:
Oh dear. The next person to ask about this function.

The effort involved should not be underestimated.
As you may know, my fork supports several interfaces. (Besides Webconnect also Speedwire and EnnoxOS-based devices) Each interface would have to be implemented and tested separately.

For Speedwire, nobody has figured out how to set values yet.
I already have an approach for ennexOS.
For Webconnect I would need someone who is willing to support me.
This person must be able to install the library and at least run the example.py. As a first step, I would need an HDR file (see below) with  "Open the relevant web page" and "change a value". Then I could check if something like this can be realized with reasonable costs.  

The topic of HA comes at the very end for me.
For a first version, I would use a service call. 

https://support.servicenow.com/kb?id=kb_article_view&sysparm_article=KB0647719

thestonewell, 2024-05-26:
Alright. I will get to this (but slowly). I already have the example.py
running in the VS Code debugger and been stepping through. I can do the
Webconnect, np. Will get the HDR file. I am sure I can figure some myself.
Happy to collaborate.

- Thomas

littleyoda, 2024-05-26:
sounds like 'challenge accepted'

You can find my E-Mail here:
https://littleyoda.github.io/

gigatexel, 2024-05-27:
> sounds like 'challenge accepted'
> 
> You can find my E-Mail here: https://littleyoda.github.io/

Same for me. I've been running this feature modbus-based for now.