kellerza/pysma #122: Pv_isolation_resistance Unavailable

edautz, 2023-07-27:
In HA 2023 core-2023-7.1 

After enabling the sensor.Pv_isolation_resistance on the SMA integration. It gives the state Unavailable.

[image]

When running example.py on my RPI4 of your pysma repository:

python3 example.py https://192.168.X.YYY installer XXXXXX | grep pv_isolation_resistance
DEBUG:pysma:Adding sensor pv_isolation_resistance (6102_00254F00_0)
DEBUG:pysma.sensor:Sensor pv_isolation_resistance: Will be decoded with * | [0][0].val from {'1': [{'val': 2923413}]}
  pv_isolation_resistance       2923.413 kOhm
  pv_isolation_resistance       2923.413 kOhm
  pv_isolation_resistance       2923.413 kOhm
  pv_isolation_resistance       2923.413 kOhm
  pv_isolation_resistance       2923.413 kOhm

I see the correct value. 

   current_l1          1.147 A
   current_l2
   current_l3
   current_total          1.147 A
   daily_yield            289 Wh
   frequency          50.02 Hz
   grid_apparent_power            264 VA
   grid_apparent_power_l1            264 VA
   grid_apparent_power_l2
   grid_apparent_power_l3
               grid_power            181 W
        grid_power_factor          0.874
grid_power_factor_excitation    Overexcited None
      grid_reactive_power            100 var
   grid_reactive_power_l1            100 var
   grid_reactive_power_l2
   grid_reactive_power_l3
        grid_relay_status         Closed None
insulation_residual_current             10 mA
       inverter_condition             Ok None
     inverter_power_limit           5000 W
     inverter_system_init            Yes None
 operating_status_general      Activated None
                 power_l1            181 W
                 power_l2              0 W
                 power_l3              0 W
             pv_current_a           0.51 A
             pv_current_b          0.515 A
             pv_gen_meter      10075.194 kWh
  pv_isolation_resistance       2923.413 kOhm
                 pv_power            181 W
               pv_power_a            134 W
               pv_power_b            131 W
             pv_voltage_a         263.38 V
             pv_voltage_b         256.16 V
                   status             Ok None
              total_yield      10075.194 kWh
               voltage_l1         230.44 V
               voltage_l2
               voltage_l3

I see also     insulation_residual_current             10 mA

What could be wrong?

edautz, 2023-07-28:
I see in the file definitions.py line 90 an other format and an extra comma.

enabled=False,

could that be the problem?

edautz, 2023-07-29:
Made modificaties in my test HA container for the definitions.py file. After starting the patched container both sensors could be enabled and produce the correct output. 

It would be nice to correct this file in this repo for future HA versions. 

No experience with PR until now, but I will give it a try when I find some spare time.

rklomp, 2023-07-30:
Can you provide debug logging while reproducing the issue?
Turn on debug in the configuration.yaml and than restart HA.
```yaml
logger:
  default: info
  logs:
    homeassistant.components.sma: debug
    pysma: debug
```

edautz, 2023-07-31:
Added the pysma debug log from one of my unpatched HA production systems.

[pysma_debuglog.zip](https://github.com/kellerza/pysma/files/12213340/pysma_debuglog.zip)

rklomp, 2023-07-31:
I don't see `6102_40254E00` in the reply from `/dyn/getAllParamValues.json` or `/dyn/getAllOnlValues.json`. That's why the sensor is not loaded. See also https://github.com/kellerza/pysma/blob/master/pysma/__init__.py#L408

I don's see why your change in #123 would fix that.
Could you send the same debug log in an instance that is working correctly?

edautz, 2023-07-31:
I saw I made a mistake in my comment.

Should be 6102_40254E00_0 instead of 6102_40254E00 (and corrected the comment).

So I changed 6102_40254E0 to 6102_40254E00_0 and the formatting in the definitions.py file. Implementated on one of my production system. After the restart with the patched container I had also to trick the integration, by fooling it as i try to installed it again, to activate the changes.

[image]

I will provide you with a new debug log on the patched production system.

rklomp, 2023-07-31:
Changing `6102_40254E00` to `6102_40254E00_0` should not make any difference. As `_0` is default.

edautz, 2023-07-31:
[pysma_debuglog_patched.zip](https://github.com/kellerza/pysma/files/12214222/pysma_debuglog_patched.zip)

The debug logging of the patched correct working system.

I changed in definitions.py the pv_isolation_resistance and insulation_residual_current in a onliner: 

pv_isolation_resistance = Sensor("6102_00254F00_0", "pv_isolation_resistance", unit="kOhm", factor=1000, enabled=False)

insulation_residual_current = Sensor("6102_40254E00_0", "insulation_residual_current", unit="mA", enabled=False)

and that worked. It is maybe related to my invertor type and firmware version?

rklomp, 2023-07-31:
In both logs there is a difference between the data returned by `/dyn/getAllOnlValues.json`.

In the "working" situation you are using the installer account and have more permissions. Thus having access to the pv_isolation_resistance sensor.

edautz, 2023-07-31:
Allright. 

My mistake.

Sorry for the trouble.